SensorsIot/IOTstack #29: Yaml template merging

Slyke, 2020-04-21:
Copying this from: https://github.com/gcgarner/IOTstack/issues/128

Found some example code that will achieve this: https://stackoverflow.com/questions/47424865/merge-two-yaml-files-in-python

`my-custom.yml` example:
```
services:
  pihole:
    restart: always
    extra_hosts:
      - router: 192.168.1.1
      - iotstack 192.168.1.5
```

The current `service.yml` for pihole:
```
pihole:
    container_name: pihole
    image: pihole/pihole:latest
    ports:
      - "53:53/tcp"
      - "53:53/udp"
      - "67:67/udp"
      - "8089:80/tcp"
      #- "443:443/tcp"
    env_file:
      - ./services/pihole/pihole.env
    volumes:
       - ./volumes/pihole/etc-pihole/:/etc/pihole/
       - ./volumes/pihole/etc-dnsmasq.d/:/etc/dnsmasq.d/
    dns:
      - 127.0.0.1
      - 1.1.1.1
    # Recommended but not required (DHCP needs NET_ADMIN)
    #   https://github.com/pi-hole/docker-pi-hole#note-on-capabilities
    cap_add:
      - NET_ADMIN
    restart: unless-stopped
```

The output to `docker-compose.yml`:
```
  pihole:
    extra_hosts:
      - router: 192.168.1.1
      - iotstack 192.168.1.5
    container_name: pihole
    image: pihole/pihole:latest
    ports:
      - "53:53/tcp"
      - "53:53/udp"
      - "67:67/udp"
      - "8089:80/tcp"
      #- "443:443/tcp"
    env_file:
      - ./services/pihole/pihole.env
    volumes:
       - ./volumes/pihole/etc-pihole/:/etc/pihole/
       - ./volumes/pihole/etc-dnsmasq.d/:/etc/dnsmasq.d/
    dns:
      - 127.0.0.1
      - 1.1.1.1
    # Recommended but not required (DHCP needs NET_ADMIN)
    #   https://github.com/pi-hole/docker-pi-hole#note-on-capabilities
    cap_add:
      - NET_ADMIN
    restart: always
```

You can see that `restart` is overridden by the value in `my-custom.yml` and that `extra_hosts` with it's properties is added in.

Will likely add this in as a Python script. That means Python will have to be installed on the host system. Could also just use a docker container with this script in it to generate. Will do some research.

robertcsakany, 2020-04-21:
As I know for docker-compose python is required.

robertcsakany, 2020-04-21:
Another suggestion: When a container is configured sometimes it effects another container's configuration. - for example port for a service. I think when the merging is done have to be able to reuse structures for merging.

For example:
   variables.yaml
```
variables:
   - mqtt-port: 1883
   - base-domain: 192.168.16.251.xip.io
```

   custom.yaml

```
mosquitto:
    ports:
      - ${mqtt-port}:1883

openhab:
    environment:
    - VIRTUAL_HOST=openhab.${base-domain}
```

With the merge and the minimal template feature I think we can add all of required dynamics for the final compose file generation. What do you think?

Slyke, 2020-04-22:
> As I know for docker-compose python is required.

That makes it super useful. I didn't know docker-compose installs Python on the host system. Guess it makes sense since it's written in Python!

For the settings based off which containers are installed, yep. I intend to make some time of advanced menu for these - where these options are shown. If the user doesn't want to change or see these settings, then they can just skip it and install it how we think is best.

Slyke, 2020-04-26:
This feature is in https://github.com/SensorsIot/IOTstack/pull/38. Closing issue (feel free to reopen if issues though).

closed 2020-04-26 by Slyke