SensorsIot/IOTstack #33: Rewrite some or all of menu.sh

Slyke, 2020-04-21:
Rewrite some or all of `menu.sh` to allow for more advanced configurations, and automated deployments.

Consider using python or node.
Consider having some of the deployment done inside docker containers so that node and python does not need to be installed on the host system.

877dev, 2020-04-21:
**> Rewrite some or all of `menu.sh` to allow for more advanced configurations, and automated deployments.**

I think the appeal of this project is that it's simple for most people to install and setup, we must be careful not to make it too complex and niche.

Just my opinion.

877dev, 2020-04-21:
I think the appeal of this project is that it's simple for most people to install and setup, we must be careful not to make it too complex and niche.

Just my opinion.

Paraphraser, 2020-04-21:
+1 on 887dev’s comment

Slyke, 2020-04-22:
Yep, totally agree. I don't mean to say I'm going to write a full application around it. Maybe just break it up a bit so that it is easier to manage the features being requested. I think it will make more sense to have a main menu file, but then have the different menu items broken into separate files.

Install menu would be it's own file that is imported at run time so that it's seamless for example. Any containers that have advanced settings, like PiHole, NodeRed etc would also have their own file too. We currently have 2 bittorrent clients requested to be added, and adding lots of containers to the menu would make it long and a pain to find any particular thing, so could break it up into categories for example.

Another example is Yaml merging. It would be difficult at best in bash, but in python is relatively easy. Issue is that it requires python on the base system, but it's a super useful feature for templating the docker-compose file, and all the containers.

Still thinking about what's best, and gives the most control for those who want it, but is still easy for those who just want to see a menu and do 1 click installs type of thing. The code has to continue to have a low barrier of entry so that it is easy for those new to the community who want to to expand on the functionality and submit PRs of their own.

SmilyBorg, 2020-10-06:
Something that occurs to me would be to have split the menu.sh into a component that just installs docker, then have the rest broken out into a stand-alone docker container. This way, each time the menu is called, it can pick up the latest menu container(if desired), and all dependencies for the menu are contained in the container so do not pollute the host OS.

Please forgive me if this ideas has already been suggested as I only found the project today.

ukkopahis, 2022-02-18:
A simple Python virtualenv prevents polluting of the host OS with python dependencies.

pierreh, 2023-08-08:
Menu is currently not working with python 3.11 so I think it would be great to simply run the menu inside a docker container.