SensorsIot/IOTstack #35: Python3 Fails to Run

lampwriter, 2020-04-23:
We re-cloned from this forked project and went through the setup again. Got the same error as before.

root@wanda:~/IOTstack# docker-compose up -d
Building python
Step 1/5 : FROM python:3
 ---> ac7e76fc5dfc
Step 2/5 : WORKDIR /usr/src/app
 ---> Using cache
 ---> 3482ade3362a
Step 3/5 : COPY requirements.txt ./
ERROR: Service 'python' failed to build: COPY failed: stat /var/lib/docker/tmp/docker-builder249099125/requirements.txt: no such file or directory
root@wanda:~/IOTstack#

Slyke, 2020-04-24:
Will check it out shortly.

Slyke, 2020-04-25:
Hey @lampwriter looks like no one put the `requirements.txt` file in that the Dockerfile is expecting. I'm currently working on a PR now to fix this, but for now, just navigate to the IOTstack directory, and run
```
touch .templates/python/requirements.txt
```

You should be able to run Python without errors then.
[image]

Slyke, 2020-04-25:
PR for fix: https://github.com/SensorsIot/IOTstack/pull/37

closed 2020-04-25 by Slyke