SensorsIot/IOTstack #41: when i am modefing the docker-compose.yml with some foren containers it fails to auto start on reboot

DevSaumyadip, 2020-05-02:
whenever the container is updated, it does not auto start on system restart and also pushes with a warning to remove the foreign container, can u suggest me a fix. Thanks in advance.

Slyke, 2020-05-02:
Can you paste your docker-compose.yml file?

DevSaumyadip, 2020-05-03:
version: '3.6'
services:

  portainer:
    container_name: portainer
    image: portainer/portainer
    restart: unless-stopped
    ports:
      - 9000:9000
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
      - ./volumes/portainer/data:/data

  mongodb:
    image: mongo:4.2.3-bionic
    container_name: mongodb
    ports:
      - 27017:27017
    environment:
      - MONGO_INITDB_DATABASE=admin
      - MONGO_INITDB_ROOT_USERNAME=admin
      - MONGO_INITDB_ROOT_PASSWORD=admin
    volumes:
      # seeding scripts
      - ./volumes/mongodb/mongo-entrypoint:/docker-entrypoint-initdb.d
      # named volumes
      - ./volumes/mongodb/mongodb:/data/db
      - ./volumes/mongodb/mongoconfig:/data/configdb

Slyke, 2020-05-03:
Hey @DevSaumyadip there's 2 issues here. The first one, the issue that the "container" doesn't start (I'm assuming you mean the `mongodb` container based off your `docker-compose.yml`) is hopefully an easy fix. Just put this code right below the line `container_name: mongodb`
```
    restart: unless-stopped
```
Make sure the indentation is correct or it will error.

The second issue is harder to diagnose. Can you copy and paste the exact error message you see, along with the commands you entered in to produce it?

DevSaumyadip, 2020-05-07:
no after adding restart: unless-stopped it does not work also, I think that's a bug with Docker version 19.03.8

https://github.com/docker/for-linux/issues/652

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke