SensorsIot/IOTstack #47: Individual container configuration menus

DIYtechie, 2020-05-07:
As mentioned in the broader issue on rewriting the menu #33 and as discussed more specifically in PR #24 a configuration option / screen for each container would be very useful and a much-appreciated feature. (This is my way of including the discussion surrounding this feature in an issue rather than in a pull request)

Until the compose-override feature #38 is implemented, I'm not sure how to merge configurations (amendments to the service.yml files in the services folder) into docker-compose without running through the build menu a second time, which is how I do it now. Is there a command that will simply rerun the docker-compose merging that happens when using the menu (without having to run through the menu again)? Or is there a way to add a configuration menu (like Node-RED) in the current version of the menu.sh?

Slyke, 2020-05-08:
Override feature should be in master soon. I've asked Andreas to remove the branch protection on master. I initially asked him to put it there because I thought there was going to be more active mods ^.^; He said he'll do it shortly.

Slyke, 2020-05-10:
I have most of the new menu working.
[image]
The "options" functionality has been implemented, but I haven't actually written any of the options. I've written a generic port checker and placed it into `blynk_server`, and you can see from the screenshot that it detects a port conflict with `diyhue`.

Next step is to add in the build logic, which should be fairly easy since all the selected services are already in memory. I do need to add in saving of selected items (and options) so it can be automatically loaded next build, and also build hook functions though, in case any of the services need to run code before the docker file is written to disk (nginx would be one example of this).

I've pushed the code for this to the `new-menu` branch too.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke