SensorsIot/IOTstack #50: python failed to build

glenn3451, 2020-05-11:
Step 3/5 : COPY requirements.txt ./
ERROR: Service 'python' failed to build: COPY failed: stat /var/lib/docker/tmp/docker-builder007370358/requirements.txt: no such file or directory

i thought that maybe this error was due to me using an external hdd to run everything from so i did a clean install on a sd without the ssd and still had this error. not sure if others are also getting it...

Slyke, 2020-05-12:
Can you let me know the exact steps you're doing to produce this? We had a similar bug previously, but I thought I had fixed it.

glenn3451, 2020-05-12:
sure, but i think you will be disappointed...
basically i just copy and paste through vnc so hopefully no problems there.
1. sudo apt-get install git -y (already latest version so no change)
2. git clone https://github.com/SensorsIot/IOTstack.git ~/IOTstack (everything seems to work fine, no errors and all files appear where expected)
3. cd ~/IOTstack && bash ./menu.sh (do the regular, install docker, reboot, build stack)
4. docker-compose up -d (then i end up at the error i put up in the first comment). 

like i said, i thought maybe something happened with me running root on an ssd so i tried from scratch on a clean image with just an sd card and ran into the same problem. i dont know if it effects anything, but im using a full version of raspbian. when i get a minute i will try again just to be sure...

Slyke, 2020-05-13:
I just realised that I had fixed this, but it has not been merged in yet (I don't have permission to merge my own pull requests). It is fixed in https://github.com/SensorsIot/IOTstack/pull/37 .

chrisog505, 2020-06-29:
I am new to Docker etc., but I downloaded and ran via ./menu.sh  I'm installing on a Raspberry Pi 4 with 4GB on a new Raspberry Pi OS build with personal security updates (deleting default Pi user, adding my own username, adding a root password).  I get the following error.

Building python
Step 1/5 : FROM python:3
 ---> dcccf302a1db
Step 2/5 : WORKDIR /usr/src/app
 ---> Using cache
 ---> d6576f54fdf2
Step 3/5 : COPY requirements.txt ./
ERROR: Service 'python' failed to build: COPY failed: stat /var/lib/docker/tmp/docker-builder340159858/requirements.txt: no such file or directory

I then ran the directoryfix.sh in the python services folder.  This didn't help me either.  I noticed in this .sh file it has, what I think, is a specified "pi" user.  Given I deleted the pi user and now am running under my own username, I updated the file, ran it and still didn't seem to work.  Any help is appreciated.
Thanks.

Paraphraser, 2020-06-29:
I did the very same thing on an RPi 3 the first time I ran up IOTstack last year but I had all manner of odd problems. I don't remember the actual error messages I got at the time but I don't think it was the particular ones you have reported.

Your messages sound like they might be coming from the Dockerfile for python:

```
$ cat .templates/python/Dockerfile 
FROM python:3
WORKDIR /usr/src/app
COPY requirements.txt ./
RUN pip install --no-cache-dir -r requirements.txt
CMD [ "python", "./app.py" ]
```

I don't install the python container but my understanding is that everything in .templates/python is copied to ./services/python before the Dockerile is run so my guess is that that copying operation failed and "requirements.txt" wasn't where it was expected.

In my own case I struggled along for a while trying to make it work but, in the end, I gave up and accepted a few basic facts of IOTstack life:

* "~/IOTstack" is baked into IOTstack;
* "IOTstack" had better be spelled exactly thus;
* "~" had better expand to "/home/pi"; and
* "pi" had better be user ID 1000.

Once I started over with a clean Raspbian image, resisted the very deep-seated urge to tinker, and just installed IOTstack "as is", it all just worked. Doing the same is the best advice I can offer. I hope that helps.

Of course, if you go to all the time and trouble of starting over from a clean image and it still doesn't work then you have my sincerest apologies-in-advance for leading you up the garden path.

Paraphraser, 2020-06-30:
Just for the heck of it I ran the menu and installed python:

```
$ docker-compose up -d
Building python
Step 1/5 : FROM python:3
3: Pulling from library/python
cff0731da2e4: Pull complete
3bbeb0ded58d: Pull complete
d6159fbed770: Pull complete
7ba732bafeb6: Pull complete
26d84f63917d: Pull complete
38a9df87fb47: Pull complete
8d7eaf2c104c: Pull complete
2ce45b603fcf: Pull complete
1d85a343b967: Pull complete
Digest: sha256:dd6cd8191ccbced2a6af5d0ddb51e6057c1444df14e14bcfd5c7b3ef78738050
Status: Downloaded newer image for python:3
 ---> dcccf302a1db
Step 2/5 : WORKDIR /usr/src/app
 ---> Running in 52581d01c204
Removing intermediate container 52581d01c204
 ---> 822bcb98db68
Step 3/5 : COPY requirements.txt ./
 ---> ed4706f3ddfa
Step 4/5 : RUN pip install --no-cache-dir -r requirements.txt
 ---> Running in 884e9916985f
Removing intermediate container 884e9916985f
 ---> 245cdda3fddf
Step 5/5 : CMD [ "python", "./app.py" ]
 ---> Running in 66f0bd77923e
Removing intermediate container 66f0bd77923e
 ---> 7d9694d4051b

Successfully built 7d9694d4051b
Successfully tagged iotstack_python:latest
WARNING: Image for service python was built because it did not already exist. To rebuild this image you must use `docker-compose build` or `docker-compose up --build`.
postgres is up-to-date
Recreating mosquitto ... 
grafana is up-to-date
Creating python      ... 
portainer is up-to-date
nodered is up-to-date
influxdb is up-to-date
Recreating mosquitto ... done
Creating python      ... done
$ 
```

A few observations. First, I don't understand why Mosquitto was recreated. The Mosquitto version number didn't change. There is no obvious reason why it was recreated. I hate things like that.

Second, the python container seems to have installed OK. It is in a restart loop displaying "hello world" on each restart but I believe that's the expected behaviour. So, an "out of the box" install of python on a system where the basic facts of IOTstack life hold, python appears to work.

Third, what I said before about .templates/python being copied to services/python before Dockerfile is run appears to be true. Either a permissions issue prevented .templates/python/requirements.txt from being copied to services/python/requirements.txt or the file was copied but the permissions on the copy are preventing it from being read by the Dockerfile process.

chrisog505, 2020-06-30:
Thanks for trying it.  I continue to play with it, but haven't figured it out.  As you suggest above, I'm tempted to reinstall Raspberry Pi OS again and try IOTstack without any modifications to see if it works, ...perhaps getting rid of the pi user account etc. has caused permission issues I'm not aware of.  I'll post if I learn anything further.  Thanks for your suggestions, much appreciated!

chrisog505, 2020-06-30:
I put a new Raspberry Pi OS image on my SD card, did not make any updates to the security of the image (i.e. kept pi user, didn't add root password, didn't install firewall etc. etc.) and the IOTstack install worked successfully.  Although a good experiment which demonstrates it does work for a standard build Raspberry Pi 4 w/ 4GB machine, I will be making key security updates as I want a more secure environment.

closed 2020-09-30 by glenn3451