SensorsIot/IOTstack #51: NextCloud with MariaDB/NextCloud_db seems to not work

dolphikcode, 2020-05-12:
The whole day today I'm trying to install NextCloud. Tried already several times on clean and "refurbished" systems, with setting up passwords before running docker-compose.

First I thought that maybe some ports from nextcloud_db are not ok as I had also separate MariaDB instance, but after clean start without installing MariaDB situation is the same.

When I try to activate NextCloud with SQLite everything is ok, but when doing the same with MariaDB/nextcloud_db no visible error occurs but page is loading last two hours and nothing happens like script is stuck somewhere...

Slyke, 2020-05-12:
Can you paste your `docker-compose.yml` here? Also any configs you have for MariaDB? I haven't tested this image in a while, but from memory I also remember I had difficulties setting it up the first time. I think I had to make the databases' hostname something in particular. Either way, I will check it out.

dolphikcode, 2020-05-14:
Looks like probably my SD card is exceptionally slow... I left the registration page running while watching movie and NextCloud finally processed it and created admin user.
`docker-compose.yml` is standard from template just with passwords changed.
Only one more thing which I observed is that when NextCloud container was downloaded showed message "newer version downloaded"

closed 2020-05-14 by dolphikcode