SensorsIot/IOTstack #52: Home Assistant installation reports "Not Supported"?

Baldorian44, 2020-05-12:
Hello

Has something happened with the Home Assistant Installation via the IoTStack? I've now have everything running but HA. I just got a RPI4 to use a SSD and wanted to add HA (as my SD burned out, surprise :)). And when i try to install HA it reports "Not supported" and doesnt continue.
This happens with both RPI4 32 and 64 bit.

[image]

Thank you in advance

Slyke, 2020-05-12:
We did change the installer. I'll check this out. I'm currently rewriting the menu anyway, and this will be one of the things I test.

Baldorian44, 2020-05-13:
Thanks for the feedback @Slyke. Is there any workaround until this is fixed you can suggest? So I can get the HA installation running with the Core HA + supervisor setup?

glenn3451, 2020-05-13:
if you just add something like...

homeassistant:
    container_name: home-assistant
    image: homeassistant/home-assistant:stable
    volumes:
      - /home/pi/IOTstack/services/home-assistant:/config
    environment:
      - TZ=America/New_York
    restart: always
    network_mode: host

to the docker-compose.yml file it should download and install home assistant...

Baldorian44, 2020-05-13:
Thanks for feed back Glenn. I ended up getting help with just ignoring the not supported statement by forking out. That seemed to do the trick

DIYtechie, 2020-05-13:
> Thanks for feed back Glenn. I ended up getting help with just ignoring the not supported statement by forking out. That seemed to do the trick

Did you use this fork?: https://github.com/dykandDK/supervised-installer

If so, SensorsIOT should perhaps use this or make an identical fork, so we have a working installer again?

Baldorian44, 2020-05-15:
@DIYtechie yes i did, it works :) (I am rather new to Git, so didnt know if I were allowed to link other forks here :))

DIYtechie, 2020-08-09:
This installer is now supported again. I just tested and it works without issues. This issue can be closed now.

closed 2021-12-29 by Baldorian44