SensorsIot/IOTstack #61: deCONZ template is not being copied to docker-compose.yml

DIYtechie, 2020-05-29:
Just tried to build a new stack now, and unfortunately, deCONZ is not added to the stack, despite the fact that the config menu is shown. For some reason, the selected service.yml (e.g. .templates/deconz/service_conbee_II.yml) is not copied to the docker-compose.yml.

This is a new issue, since I used the latest version of SensorsIot/IOTstack to successfully install deCONZ 5 days ago.

Slyke, 2020-05-29:
Last night I merged the experimental branch into master. That is without much of a doubt the reason that this bug came up. Let me check it out, and I'll see if I can apply a hotfix.

Slyke, 2020-05-29:
Just did a fresh clone and can confirm that this is a bug.
[image]

Slyke, 2020-05-29:
Okay, I think I have figured it out. Need to modify the deconz `build.sh` file to echo it out to the temporary `./.tmp/docker-compose.tmp.yml` file. The actual `docker-compose,yml` file is override by the yaml merging code at build time (after all the `build.sh` have ran).

Slyke, 2020-05-29:
Created a PR for the fix: https://github.com/SensorsIot/IOTstack/pull/62/files
@DIYtechie Do you think you could test this new PR on your setup? I tested it locally and it seems fine, but want to ensure it also works with an actual deCONZ setup.

DIYtechie, 2020-05-30:
just tried it (updating iotstack and replacing the build.sh file with the one referenced above) and I'm happy to say that it works. Thanks

Slyke, 2020-05-31:
Awesome! merged it in.

closed 2020-05-31 by Slyke