SensorsIot/IOTstack #66: Prometheus?

pelted, 2020-06-12:
Any reason this isn't included as an option? 
If not, I believe I have it working and can do a PR.

Paraphraser, 2020-06-12:
I don't speak for the maintainers but (a) I have never heard of Prometheus, (b) if you mean [this](https://www.wonderware-benelux.com/en/hmi-scada/prometheus/) then I read the page, was impressed by the sheer quality of the technobabble while being left none-the-wiser as to what it actually does, and (c) say "go for it" in answer to your PR question. It has been my observation that the maintainers like nothing more than a nice clean PR that just works.

pelted, 2020-06-13:
@Paraphraser ahah [Prometheus.io](https://prometheus.io) a very popular data source for Grafana.

Paraphraser, 2020-06-13:
Double ahah! That makes a lot more sense. So, double "go for it" on the PR.

In the continuing spirit of asking dumb questions hopefully leading to illumination, given that both Influx and Prometheus claim to be time-series optimised, what particular benefits do you see that the latter has over the former? I'm an Influx newbie (six months, give or take) coming from a classic-SQL background stretching back to the 1990s. Influx worked out of the box and seems reliable when it comes to getting data in via Node-Red and visualising it in Grafana. The speed with which a Grafana can display a query for a couple of years of 10-second-interval data is astonishing, given that the engine is just an RPi4 albeit with an SSD. I'm seriously impressed.

On the flip side, I find Influx is a bit of a pest when it comes to doing things like arranging old (SQLite) data in its line-format for import, or doing anything like rearranging a series to, say, change a field to a tag. Even CLI queries seem unnecessarily hard when it comes to selecting a time range. I'd dearly love to stumble across a way to say, "look, son, I want you to assume `tz('Australia/Sydney')` on every query because I never, ever, want to have to type that boring string again."

There are also some behaviours that I find odd, such as if I run:

```
docker exec -it influxdb influx
``` 

on my live RPi4, it takes 12 seconds before I get a prompt. If I do it on my test RPi4, the prompt is practically instantaneous. The hardware and builds are identical. The only real differences are:

* the live machine is receiving a constant feed (at least one insert every 10 seconds) whereas the test machine only ever receives data if I explicitly point a downstream device at it; and
* the influx database(s) on the live machine have grown over time with no rebuilding whereas I routinely populate the test machine with backups from the live machine, so those influx databases are invariably fresh loads.

I don't like differences with no obvious explanation. It makes me worry about corruption. That feeling is not helped when I load a backup on the test machine and find it peppered with lines like this:

```
2020/06/13 00:48:56 Meta info not found for shard 706 on database _internal. Skipping shard file 20200518T013813Z.s706.tar.gz
```

When I do an obvious check like `select count(*) ...` on my various series, any differences are easily explained by the time difference between when the backup was taken on the live machine and when it was restored on the test machine. No data of mine ever seems to have gone walkabout. The error message appears to mean what it says but that just begs the question of why its "_internal" database is wonky?

I'm probably moaning about not much but, if there's a better way and Prometheus is its name...

pelted, 2020-06-17:
I've gone ahead and added a PR #78 for this.

@Paraphraser as far as the questions you brought up, Prometheus can be configured for longer retention times, but I believe the default is 15 days. My understanding is that longer term retention is one of the use cases for InfluxDB. I haven't spent any time with Influx yet, so I can't answer those questions. My goal is to use Influx for things I want to keep longer terms stats on around and Prometheus with shorter retention for everything else.

Paraphraser, 2020-06-18:
I finally got around to "fixing" the \<long delay for the CLI to get started\> on my live machine. A dump and reload plus a pull of a more recent image seemed to do the trick. Only time (ahah) will tell if this is a permanent fix or if it is a byproduct of something else I'm doing.

Interesting about retention periods. Out of the box, InfluxDB keeps everything forever and I'm here to tell you that I would not have considered Influx for one moment if it chucked data away by default. I've been in the IT game for 45 years and it wasn't until the last 10-15 years that storage became sufficiently cheap that there wasn't always some brainless bean-counting bozo somewhere encouraging people like me to do more with less.

The numbers of times I've been in the position where I could have answered the question save for being forced to delete the data months or years before...

So, with my own data, I can live the dream - it's retained!

alxlo, 2020-08-14:
I strongly support adding prometheus.io. It is a commonly used, mature and valuable tool for gathering metrics. It complements this stack very well and integrates seamlessly with influxdb and grafana.

somervda, 2020-11-09:
One more vote from me for adding Prometheus as well, it seems it would be a good complement for monitoring the overall IOTstack services. I would also find it useful for setting up quick experimental installations via IOTStack/Pi . Some info on Prometheus PI installation can be found here http://www.d3noob.org/2020/02/installing-prometheus-and-grafana-on.html and https://hub.docker.com/r/prom/prometheus/

pelted, 2020-11-09:
@somervda I believe it's in there. I added it in #78 and it was merged into master back in September.

closed 2020-11-09 by pelted

somervda, 2020-11-09:
Perfect thanks, I just updated my version of IOTstack to use this repository and can see it now. Thanks for the quick response.