SensorsIot/IOTstack #67: Is port 9001 needed for Mosquitto?

Paraphraser, 2020-06-14:
# Is port 9001 needed for Mosquitto?

Summary:

* Out-of-the box, the eclipse-mosquitto image does not listen to port 9001.
* The evidence suggests that the docker-compose port mapping for 9001 is redundant.
* Does anyone know any different?
* If there are no objections, I'm proposing removal.

## Chain of evidence

What eclipse-mosquitto image is on my RPi4?

```
$ docker images eclipse-mosquitto
REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
eclipse-mosquitto   latest              a2b55301913b        11 days ago         5.86MB
``` 

What does Docker know about this image?

```
$ docker inspect eclipse-mosquitto
[
    {
        … [snip] …
        "RepoTags": [
            "eclipse-mosquitto:latest"
        ],
        … [snip] …
        "Created": "2020-06-02T21:51:19.217586203Z",
        … [snip] …
        "ContainerConfig": {
            … [snip] …
            "ExposedPorts": {
                "1883/tcp": {}
            },
            … [snip] …
            "Env": [
                … [snip] …
                "VERSION=1.6.10",
                … [snip] …
            ],
        },
    }
]
```

Observation: *as far as `docker inspect` is concerned, only port 1883 is exposed.*

Assume IOTstack is not running.

What does `netstat` have to say about the baseline situation on the RPi4:

```
netstat -lntu
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 0.0.0.0:5900            0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN     
tcp6       0      0 :::5900                 :::*                    LISTEN     
tcp6       0      0 :::22                   :::*                    LISTEN     
udp        0      0 0.0.0.0:68              0.0.0.0:*                          
udp        0      0 0.0.0.0:68              0.0.0.0:*                          
udp        0      0 0.0.0.0:5353            0.0.0.0:*                          
udp        0      0 0.0.0.0:41959           0.0.0.0:*                          
udp6       0      0 :::5353                 :::*                               
udp6       0      0 :::37708                :::*                               
```

[... 239 lines cut ...]
listener 9001
```
*Both lines are needed - if you only add 9001 then 1883 stops working.*

That definitely causes Mosquitto to listen on the port:

```
$ docker exec mosquitto netstat -lntu
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:9001            0.0.0.0:*               LISTEN      
tcp        0      0 127.0.0.11:40117        0.0.0.0:*               LISTEN      
tcp        0      0 :::1883                 :::*                    LISTEN      
tcp        0      0 :::9001                 :::*                    LISTEN      
udp        0      0 127.0.0.11:55876        0.0.0.0:*                           
```

but it's an open question as to what you can actually do with it.....

## Bottom line

My view is that port 9001 should be removed from the Mosquitto template, with the two-step how-to of enabling port 9001 put in the documentation for anyone who subsequently discovers a use-case.  

If nobody objects to that plan, I'll prepare a pull request.

Willem-Dekker, 2020-06-15:
I think its a good thing to remove it. since it's not needed for 99% of the usecases. I think you can make the pr.

Paraphraser, 2020-06-16:
I think I will hold off until PR #69 has been applied. Just removing 9001 from the template probably wouldn't cause a conflict but I also intended to document the how-to of adding it back and that probably would create a conflict somewhere. Granted, I might be being a bit over-cautious in my estimation of what would and would not be likely to cause a conflict...

Slyke, 2020-06-17:
I just merged https://github.com/SensorsIot/IOTstack/pull/69, feel free!

thelostinthought, 2020-07-02:
you have to add a 

listener 9001
protocol websockets

to mosquitto.conf for mosquitto to use it.  This would allow web browsers to publish/subscibe to topics.

closed 2021-06-01 by Paraphraser