SensorsIot/IOTstack #71: Node-red can't connect to InfluxDB "Error: connect ECONNREFUSED 127.0.0.1:8086"

e-abdi, 2020-06-15:
Hi all,
I have a strange problem which I did not find anyone else having. So, I'm sorry if it's a very newbie issue...
I went through the instructions from the video and successfully installed everything. I have a sensor connected to COM port and when I try to store the value from 'Serial In' node to InfluxDB I get the error: "Error: connect ECONNREFUSED 127.0.0.1:8086" and "no host available". I have a user "pi" in influx which I inputted it's name and password and created a database there to receive the values. Just can't figure out why it can't connect to it.
Thanks in advance!

Willem-Dekker, 2020-06-15:
try `influxdb:8086` thats the address i am using in my node-red install

Paraphraser, 2020-06-16:
Each container is a virtual computer so it implements its own loopback address 127.0.0.1. If you "ping 127.0.0.1" on the RPi when you login, that's the hardware's IP stack responding. If you open a shell into the nodered container and do the ping there, that's the virtual hardware of the container responding. When you use "127.0.0.1:8086" you're telling Node-Red (the process) running inside the nodered container to communicate with whatever process Is listening to port 8086 in the **same** container. Nothing is listening so the communication fails.

All the containers are connected to an internal "bridged" network. You can get an idea of what is going on if you do a `traceroute` from both the Pi and from inside a container. You'll see that the container has an extra hop and that tells you that the internal bridged (Layer 2) network interconnecting the containers is routed (Layer 3) to the Pi's IP stack.

A good way to think of it is that every container has a "hosts" file that maps between each container's name (nodered, influxdb, mosquitto, etc) and the IP address that Docker allocates dynamically when the container is brought up. Those addresses start with 172.16.x.x. Don't make the mistake of thinking you can use a Docker-allocated IP address instead of the loopback address because the addresses change all the time. You can figure out the names of containers via either "docker ps" or "docker-compose ps".

So, as Willem-Dekker says, just use `influxdb:8086`. That causes a lookup to get the IP address currently associated with the influxdb container, and then the communication is sent to port 8086 on that other container, which is where InfluxDB (the process) is listening.

You might also find it helpful to [read this gist](https://gist.github.com/Paraphraser/c9db25d131dd4c09848ffb353b69038f).

e-abdi, 2020-06-16:
Thanks for the informative messages guys!

closed 2020-06-16 by e-abdi