SensorsIot/IOTstack #76: InfluxDB - most environment keys in the template do not work

Paraphraser, 2020-06-17:
The `~/IOTStack/.template/influxdb/influxdb.env` contains the following:

```
#INFLUXDB_DB=mydb
INFLUXDB_DATA_ENGINE=tsm1
INFLUXDB_REPORTING_DISABLED=false
#INFLUXDB_HTTP_AUTH_ENABLED=true
INFLUXDB_ADMIN_ENABLED=true
#INFLUXDB_ADMIN_USER=myadminuser
#INFLUXDB_ADMIN_PASSWORD=myadminpassword
INFLUXDB_USER=nodered
INFLUXDB_USER_PASSWORD=nodered
#INFLUXDB_READ_USER=myreaduser
#INFLUXDB_READ_USER_PASSWORD=myreadpassword
#INFLUXDB_WRITE_USER=mywriteuser
#INFLUXDB_WRITE_USER_PASSWORD=mywritepassword
```

and, because [a question about the following key](https://github.com/gcgarner/IOTstack/issues/141) has come up before, I have added it to the list:

```
#INFLUXDB_HTTP_FLUX_ENABLED=true
```

## Test 1 – search the documentation page

With the [Configure InfluxDB OSS](https://docs.influxdata.com/influxdb/latest/administration/config/) page open, use the browser's built-in search facility to look for the exact keys:

Key                             | Result
--------------------------------|---------------
INFLUXDB\_ADMIN\_ENABLED        | not documented
INFLUXDB\_ADMIN\_PASSWORD       | not documented
INFLUXDB\_ADMIN\_USER           | not documented
INFLUXDB\_DATA\_ENGINE          | not documented
INFLUXDB\_DB                    | not documented
INFLUXDB\_HTTP\_AUTH\_ENABLED   | documented
INFLUXDB\_HTTP\_FLUX\_ENABLED   | documented
INFLUXDB\_READ\_USER            | not documented
INFLUXDB\_READ\_USER\_PASSWORD  | not documented
INFLUXDB\_REPORTING\_DISABLED   | documented
INFLUXDB\_USER                  | not documented
INFLUXDB\_USER\_PASSWORD        | not documented
INFLUXDB\_WRITE\_USER           | not documented
INFLUXDB\_WRITE\_USER\_PASSWORD | not documented

## Test 2 - use Google to search the site

Of the keys that are **not** documented, use Google to perform a restricted search using this syntax:

```
"key" site:.docs.influxdata.com
```

Key                             | Hits
--------------------------------|---------------
INFLUXDB\_ADMIN\_ENABLED        | 0
INFLUXDB\_ADMIN\_PASSWORD       | 0
INFLUXDB\_ADMIN\_USER           | 0
INFLUXDB\_DATA\_ENGINE          | 0
INFLUXDB\_DB                    | 0
INFLUXDB\_READ\_USER            | 0
INFLUXDB\_READ\_USER\_PASSWORD  | 0
INFLUXDB\_USER                  | 1
INFLUXDB\_USER\_PASSWORD        | 0
INFLUXDB\_WRITE\_USER           | 0
INFLUXDB\_WRITE\_USER\_PASSWORD | 0

The hit for "INFLUXDB\_USER" is at a page about [Creating InfluxDB and Kapacitor connections](https://docs.influxdata.com/chronograf/v1.8/administration/creating-connections/#environment-variables-in-src-files) and the context is that of a substitution key, not an environment key.

## Test 3 - use the site's search capability 

Sticking to the keys that were **not** documented in test 1, use the "Search the docs…" field at the top of the [Configure InfluxDB OSS](https://docs.influxdata.com/influxdb/latest/administration/config/) page:

Key                             | Hits
--------------------------------|---------------
INFLUXDB\_ADMIN\_ENABLED        | 0
INFLUXDB\_ADMIN\_PASSWORD       | 0
INFLUXDB\_ADMIN\_USER           | 0
INFLUXDB\_DATA\_ENGINE          | 0
INFLUXDB\_DB                    | [a]
INFLUXDB\_READ\_USER            | 0
INFLUXDB\_READ\_USER\_PASSWORD  | 0
INFLUXDB\_USER                  | [b]
INFLUXDB\_USER\_PASSWORD        | 0
INFLUXDB\_WRITE\_USER           | 0
INFLUXDB\_WRITE\_USER\_PASSWORD | 0

* [a] appears to be matching on "InfluxDB" and "DB" but finds nothing relevant
* [b] appears to be matching on "InfluxDB" and "user" but finds nothing relevant

## Test 4 - look for partial matches

Using the same approach as test 1, look for partial matches as a way of (hopefully) picking up typos and/or renamed keys:

Prefix            | Hits
------------------|-----:
INFLUXDB\_        | 142
INFLUXDB\_ADMIN\_ | 0
INFLUXDB\_DATA    | 18
INFLUXDB\_DATA\_  | 0
INFLUXDB\_DATA\_E | 0
INFLUXDB\_D       | 18
INFLUXDB\_DB      | 0
INFLUXDB\_READ    | 0
INFLUXDB\_U       | 9
INFLUXDB\_US      | 0
INFLUXDB\_W       | 0 

## The bottom line for environment keys

I think test 1 was fairly conclusive all by itself. Tests 2 and 3 were aimed at discovering whether environment keys might be documented in places other than [Configure InfluxDB OSS](https://docs.influxdata.com/influxdb/latest/administration/config/). The answer to that question seems to be "no". Test 4 was aimed at eliminating typos or renames. That was fruitless as well.

That leaves us with these keys:

```
INFLUXDB_HTTP_AUTH_ENABLED
INFLUXDB_HTTP_FLUX_ENABLED
INFLUXDB_REPORTING_DISABLED
```

The convention in [the documentation](https://docs.influxdata.com/influxdb/latest/administration/config/) is to show the default value of the key, so the IOTstack `.template` file should probably become:

```
INFLUXDB_HTTP_AUTH_ENABLED=false
INFLUXDB_HTTP_FLUX_ENABLED=false
INFLUXDB_REPORTING_DISABLED=false
```

There seems little point in commenting-out keys **and** setting them false. All that achieves is to require **two** edits to activate a key.

The alternative representation would be:

```
#INFLUXDB_HTTP_AUTH_ENABLED=true
#INFLUXDB_HTTP_FLUX_ENABLED=true
#INFLUXDB_REPORTING_DISABLED=true
```

On balance, I'm more inclined to go with the first approach and adopt the standard established in the documentation.

## IOTstack documentation for InfluxDB

The [IOTstack InfluxDB documentation](https://sensorsiot.github.io/IOTstack/InfluxDB/) says:

> The credentials and default database name for influxdb are stored in the file called influxdb/influx.env. The default username and password is set to "nodered" for both. It is HIGHLY recommended that you change them. The environment file contains several commented out options allowing you to set several access options such as default admin user credentials as well as the default database name. Any change to the environment file will require a restart of the service.

That's somewhere between *misleading* and *wrong*. The relevant documentation page is [Authentication and authorization in InfluxDB](https://docs.influxdata.com/influxdb/latest/administration/authentication_and_authorization/) and there is mention of these two environment keys:

```
INFLUX_USERNAME
INFLUX_PASSWORD
```

but the context is by way of providing defaults for the CLI rather than **setting** a username and password combination that will then be enforced by InfluxDB for all comers.

For the record, there was no `INFLUX_DB` (or similar) environment key described at [Authentication and authorization in InfluxDB](https://docs.influxdata.com/influxdb/latest/administration/authentication_and_authorization/). I tried adding it to see if it had any effect: it did not. The [influx - InfluxDB command line interface](https://docs.influxdata.com/influxdb/latest/tools/influx-cli/) documentation is silent on this question.

All handling of credentials is internal to the database with commands like:

```
> CREATE USER testadmin WITH PASSWORD 'easyToGuess' WITH ALL PRIVILEGES
> CREATE USER testuser WITH PASSWORD 'easierToGuess'

> SHOW USERS
user      admin
----      -----
testadmin true
testuser  false

> CREATE DATABASE test
> USE test
> GRANT ALL ON test TO testadmin
> GRANT READ ON test TO testuser

> SHOW GRANTS FOR testadmin
database privilege
-------- ---------
test     ALL PRIVILEGES

> SHOW GRANTS FOR testuser
database privilege
-------- ---------
test     READ

> DROP USER testadmin
> DROP USER testuser
> DROP DATABASE test
```

My view is that the whole field of database access control is beyond the scope of IOTstack documentation. Giving the impression username and password combinations can ever be "set and forget" is quite misleading. Covering the field properly would involve duplicating [Authentication and authorization in InfluxDB](https://docs.influxdata.com/influxdb/latest/administration/authentication_and_authorization/). The IOTstack documentation should point to that page and leave the rest to the user.

## EOF

I'll wait a day or two for comments, and then prepare a pull request to:

1. Change the template `influxdb.env` to:

	```
	INFLUXDB_HTTP_AUTH_ENABLED=false
	INFLUXDB_HTTP_FLUX_ENABLED=false
	INFLUXDB_REPORTING_DISABLED=false
	```

2. Change the documentation to:

	* a discussion of the environment keys provided in the template
	* a pointer to the InfluxDB documentation as the source for more environment keys
	* how to activate new or changed environment key 
	* a pointer to the Influx documentation on security.

Slyke, 2020-06-17:
If you like, you can create a PR and have this merged into the `experimental` branch. Anyone on it can test it out then. I think we need to update the documentation as you said too. It's been a while since I've looked at it, so will need to redo.

Tasm-Devil, 2020-10-10:
What is the state of this issue now? I feel, that I still have no users defined.
```
user@host:~/IOTstack/services/influxdb$ cat influxdb.env 
#INFLUXDB_DB=mydb
INFLUXDB_DATA_ENGINE=tsm1
INFLUXDB_REPORTING_DISABLED=false
#INFLUXDB_HTTP_AUTH_ENABLED=true
INFLUXDB_ADMIN_ENABLED=true
#INFLUXDB_ADMIN_USER=myadminuser
#INFLUXDB_ADMIN_PASSWORD=myadminpassword
INFLUXDB_USER=nodered
INFLUXDB_USER_PASSWORD=nodered
#INFLUXDB_READ_USER=myreaduser
#INFLUXDB_READ_USER_PASSWORD=myreadpassword
#INFLUXDB_WRITE_USER=mywriteuser
#INFLUXDB_WRITE_USER_PASSWORD=mywritepassword
user@host:~/IOTstack/services/influxdb$ docker-compose up -d
portainer is up-to-date
grafana is up-to-date
influxdb is up-to-date
mosquitto is up-to-date
nodered is up-to-date
telegraf is up-to-date
user@host:~/IOTstack/services/influxdb$ docker exec -u root -it influxdb bash
root@702a9aa15fe0:/# set | grep INFLUXDB
INFLUXDB_ADMIN_ENABLED=true
INFLUXDB_DATA_ENGINE=tsm1
INFLUXDB_REPORTING_DISABLED=false
INFLUXDB_USER=nodered
INFLUXDB_USER_PASSWORD=nodered
INFLUXDB_VERSION=1.8.3
root@702a9aa15fe0:/# influx
Connected to http://localhost:8086 version 1.8.3
InfluxDB shell version: 1.8.3
> show users
user admin
---- -----
> 
```

Paraphraser, 2020-10-11:
Hi @Tasm-Devil - see if [Enabling authentication in InfluxDB](https://gist.github.com/Paraphraser/9f3aa21fddd5d4e8141a7ab03ffb4dd7) answers your question. If you really want to set up authentication for InfluxDB then it should get you started.

I have never been able to find any solid InfluxDB documentation supporting the existence of many of the keys in `influxdb.env`. The only ones I have **proven** to work are:

```
INFLUXDB_HTTP_AUTH_ENABLED=
INFLUXDB_HTTP_FLUX_ENABLED=
INFLUXDB_REPORTING_DISABLED=
INFLUX_USERNAME=
INFLUX_PASSWORD=
```

and, when you read the above gist, you will find that `INFLUX_USERNAME` and `INFLUX_PASSWORD` are **only** for passing default administrator credentials to `influx` (the command line interface), once authentication is enabled by setting `INFLUXDB_HTTP_AUTH_ENABLED` true.

I assume that any environment key actually documented at [configuration settings](https://docs.influxdata.com/influxdb/v1.8/administration/config/#configuration-settings) will work as per [environment variables](https://docs.influxdata.com/influxdb/v1.8/administration/config/#environment-variables).

While I don't know this for a **fact**, I suspect that all the others in `influxdb.env` were either a triumph of hope over actual testing, or hooks for an imagined future where some of this authentication stuff might have been automated. For example, a first-run script pushed into the influxdb container by a Dockerfile which picked up initial credentials from these environment keys and called influx to do the work.

I'm thinking a Pull Request to reduce `influxdb.env` to the known-good set might be a good idea.

closed 2020-10-11 by Paraphraser

Paraphraser, 2020-10-11:
@Tasm-Devil see [PR 148](https://github.com/SensorsIot/IOTstack/pull/148). The only keys left in the template now are:

```
INFLUXDB_HTTP_FLUX_ENABLED=false
INFLUXDB_HTTP_AUTH_ENABLED=false
INFLUX_USERNAME=dba
INFLUX_PASSWORD=supremo
INFLUXDB_UDP_ENABLED=false
INFLUXDB_UDP_BIND_ADDRESS=0.0.0.0:8086
INFLUXDB_UDP_DATABASE=udp
```

I've also added a lot of inline documentation to the template environment file. You can see what I'm proposing by drilling down into the Pull Request.

Tasm-Devil, 2020-10-17:
Yea. Nice. That makes it easier for the next guy.
Thank you.