SensorsIot/IOTstack #82: Mosquitto authorization

BorisHruby, 2020-06-22:
Hi guys,

I've just found out there is a wrong path in mosquitto.conf file for enabling passwords.
There stands:

password_file /mosquitto/pwfile/pwfile

As this didn't work for me I looked inside into the container and saw the pwfile sitting in the config folder not pwfile folder. I don't know why is the pwfile folder there and mapped to local pwfile folder. I suppose the pwfile should sit in this folder.

If I changed the path to:

password_file /mosquitto/config/pwfile

everything runs smoothly again.

Paraphraser, 2020-06-23:
Please see issue #68. The problem I found is that, while the `/mosquitto/config/pwfile` works, it does not persist because the file itself is not mapped to anywhere outside the container. Other **files** in `/mosquitto/config` are mapped but the **folder** is not mapped. As soon as the container is rebuilt, the internal pwfile disappears and has to be reconstructed. It's actually a silent failure. The user may **think** he/she has set up a password but it just disappears and everything keeps truckin'.

Now, sure, you could add the file mapping in the same way as `mosquitto.conf` and `filter.acl` but during my testing I found you can get a chicken-and-egg situation where mosquitto gets into a restart loop if the external file disappears. Sticking it into a different folder avoids that problem. But, you do need to either do a "pull full service from template" or mimic the important bits by adding the mapping:

```
- ./volumes/mosquitto/pwfile:/mosquitto/pwfile
```

and then running `.templates/mosquitto/directoryfix.sh` (which creates the `./volumes/mosquitto/pwfile/` directory and gives it 1883:1883 owner and group) before telling Docker to bring up Mosquitto.

Thereafter, any pwfile you create (by following the updated doco page - albeit not the shell script until PR #73 is processed) will appear in that folder, will persist across container rebuilds, and will get backed-up. Also, you gain the ability to turn passwords on and off by doing reasonably logical things like renaming pwfile to pwfile.off or just deleting pwfile or setting its contents null, all without causing Docker or Mosquitto to grizzle.

It isn't clear to me why the base mapping mapped files (mosquitto.conf and filter.acl) rather than exposing the whole /mosquitto/config folder but it might be another chicken-and-egg problem where those files are in "services" rather than "volumes". I did a small amount of experimentation to try to figure out if I could keep the pwfile in config and expose it and have it persist but I kept running into restart loops if everything wasn't "just right". A fair bit of Mosquitto's sensitivity to its environment has to do with its default behaviour of running as user 1883 (also covered in the new doco). A separate folder seemed the best way to solve all those problems. Does that make sense?

Paraphraser, 2020-06-23:
And you also need to create the pwfile in its proper location inside the container via `mosquitto_passwd -c /mosquitto/pwfile/pwfile «MYUSER»` as per the new doco.

BorisHruby, 2020-06-23:
Thanks a lot for the answer, Phill. I also don't understand why the whole config folder is not mapped outside the container. Yes, the main issue is probably that the pwfile is created in config folder instead of pwfile folder (like the shell tells you). So we have to change this path.

Paraphraser, 2020-06-23:
I don't know this for a fact (and haven't tested it) but I suspect that mosquitto.conf has to be mapped into services so it is "there" the first time the container starts but before stuff in volumes gets populated on that first run. Previous experience also suggests that the implied recursion you get if you map both a folder and a file within that folder is something that the layered file system doesn't really support.

I think (again without knowing) that mosquitto.conf also has to be in services so that changes act as a trigger for docker-compose to rebuild the container.

I might be full of it, of course. It's possible that mapping the folder into volumes and hard linking the config file into services would achieve the same result.

Paraphraser, 2022-06-19:
Fixed as part of Mosquitto self repair.

closed 2022-09-11 by Slyke