SensorsIot/IOTstack #84: Add RTL_433 to mqtt

LeeSpa, 2020-06-28:
When I try to to add "RTL_433 to mqtt" by using container selection I get following message after docker-compose:
Step 7/8 : RUN apt-get update && apt-get install -y  git libtool libusb-1.0.0-dev librtlsdr-dev rtl-sdr cmake automake && 	git clone https://github.com/merbanan/rtl_433.git /tmp/rtl_433 && 	cd /tmp/rtl_433/ && 	mkdir build && 	cd build && 	cmake ../ && 	make && 	make install
 ---> Running in 33366bfc61e4
Err:1 http://security.debian.org/debian-security buster/updates InRelease
  Temporary failure resolving 'security.debian.org'
Err:2 http://deb.debian.org/debian buster InRelease
  Temporary failure resolving 'deb.debian.org'
Err:3 http://deb.debian.org/debian buster-updates InRelease
  Temporary failure resolving 'deb.debian.org'
Reading package lists...
W: Failed to fetch http://deb.debian.org/debian/dists/buster/InRelease  Temporary failure resolving 'deb.debian.org'
W: Failed to fetch http://security.debian.org/debian-security/dists/buster/updates/InRelease  Temporary failure resolving 'security.debian.org'
W: Failed to fetch http://deb.debian.org/debian/dists/buster-updates/InRelease  Temporary failure resolving 'deb.debian.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package git
E: Unable to locate package libtool
E: Unable to locate package libusb-1.0.0-dev
E: Couldn't find any package by glob 'libusb-1.0.0-dev'
E: Couldn't find any package by regex 'libusb-1.0.0-dev'
E: Unable to locate package librtlsdr-dev
E: Unable to locate package rtl-sdr
E: Unable to locate package cmake
E: Unable to locate package automake
ERROR: Service 'rtl_433' failed to build: The command '/bin/sh -c apt-get update && apt-get install -y  git libtool libusb-1.0.0-dev librtlsdr-dev rtl-sdr cmake automake && 	git clone https://github.com/merbanan/rtl_433.git /tmp/rtl_433 && 	cd /tmp/rtl_433/ && 	mkdir build && 	cd build && cmake ../ && 	make && 	make install' returned a non-zero code: 100

Slyke, 2020-06-28:
Either debian.org is down... or there's a problem with your internet connection. What happens if you try to go to security.debian.org and the other domains that it attempts to?

LeeSpa, 2020-06-28:
Thanks for quick response. 
All these addresses work in chromium.

dannutu, 2020-07-07:
I had a similar issue after the 1st IOTstack installation. If I stopped the IOTstack containers with docker-compose then DNS resolution would work again, if I restarted them then DNS would stop working. I noticed the wlan interface (the only connection to the net) lost its IP address. I reinstalled raspbian and I configured the wifi interface using the /boot/wpa_supplicant file and then I also set a static IP for it before reinstalling IOTstack. This fixed the issue for me.

HTH,
Dan

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke