SensorsIot/IOTstack #86: Disk writes

dannutu, 2020-07-07:
Hi & thank you for putting IOTstack together!

I noticed that right after installing a small set of IOTstack containers (i.e. portainer, node-red, influxdb, grafana, eclipse-mosquitto, pi-hole, rtl_433 to mqtt) on a fresh raspbian OS there are quite frequent writes to the disk on an otherwise completely idle pi. Over 24 hours these writes amounted to some 5 GB, which go towards shortening the life of the SD card or SSD. I was wondering if you could offer any hints to limit these writes? Are they really necessary considering the system is completely idle otherwise?

I've tried to identify the "hot file(s)" but I got nowhere so far (I could not get sysdig to work due to some compilation errors when building dkms and I didn't have time to troubleshoot it further; sysdig docker container also throws some errors so no luck with that either).

Many thanks!
Dan

dannutu, 2020-07-11:
Any idea how to trace these disk writes? I left the rpi idle for 6 days and iostat is showing 28 GB written to sda although df is not showing a decrease in free disk space, so I suppose some file(s) is/are written over and over. There is no swap configured (no swap partition and no swap file).

Slyke, 2020-07-11:
Hey @dannutu , try turning  off swap space. It's in the menu, but you can also run these 2 commands:
```
.scripts/disable_swap.sh disableswap
.scripts/disable_swap.sh uninstallswap
```

If you checkout the `.scripts/disable_swap.sh` file, you can see how to do manually.

dannutu, 2020-07-12:
I did mention above that there is no swap configured :)

Paraphraser, 2020-07-12:
Have you tried this?

```
$ sudo apt install iotop
$ sudo iotop
a
```

The "a" puts iotop into accumulation mode. The biggest writers tend to bubble to the top (at least they do on my system). I get lots of instances of influxd and dockerd which is what I expected, although I hadn't fully appreciated how many separate instances of those were being spawned all the time.

Most of my IoT devices go MQTT->NodeRed->Influx. Some use SQLite but those are sporadic so it isn't surprising that they don't show up in iotop. Probably the biggest surprise in my output was PiHole which seems quite chatty (logs, presumably). My RPi4 is running on SSD so I'm not particularly concerned about writes - I'd be a lot more worried about minimising writes on an SD.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke