SensorsIot/IOTstack #89: Node-RED update to newest

BorisHruby, 2020-07-14:
Hi guys! Does anybody here have an idea why I can't update the Node-RED to the newest version? I still have 1.0.6 and there is already 1.1.2 out. Any suggestion will be highly appreciated. Thanks in advance.

Paraphraser, 2020-07-14:
You did not say what commands you were using but this is what I do:

```
$ cd ~/IOTstack
$ docker-compose down
$ docker rmi iotstack_nodered:latest nodered/node-red:latest
$ docker-compose pull
$ docker-compose up --build -d
```

I just ran those commands and now my stack is running v1.1.2. There are probably other ways of achieving the same result but the above has always worked for me. The reason why nodered needs special handling is because of how it is built - by pulling down nodered/node-red:latest then running a Dockerfile to build iotstack_nodered:latest locally which then becomes the running image.

My **guess** is that any container built from a Dockerfile will need the same treatment in order to force its update. I don't run any other containers that employ Dockerfile so I don't know this for certain.

Please let me know if this solves your problem.

BorisHruby, 2020-07-15:
Thank you Paraphraser! You're the best! ;-)
It indeed worked also for me. But I tried to stop just the nodered container. In the IOTstack menu there is an option to update all the containers, but it didn't work for me. At least for the Node-RED.
Thanks again. Have a nice day!

Paraphraser, 2020-07-15:
I have a memory of Graham Garner saying something like "known issue - will need to be fixed sometime" when I struck this very same problem.

There are quite a few things wrong with how NodeRed is built. Only a few days ago somebody pointed out that every command in the Dockerfile creates a new layer and that we would be better off creating a pseudo package (my words) where what the Dockerfile now installs became package dependencies. That strategy means a single new layer, apparently. In my own builds, I have adopted the approach of "install via the menu if it's in the menu, otherwise install via the Node-Red GUI" so I wind up with a mixture where some things follow the image (and need an image rebuild to update) while other just live in volumes and can be updated via the GUI plus a restart via the CLI.

Because a full rebuild is so damn slow (particularly if SQLite is selected - which I need), I'm thinking it might be better to only offer through the menu those things that can ONLY be installed at image build time, leaving everything else for GUI installation.

I don't quite get why SQLite needs to be compiled from scratch. Before IOTstack I was running NodeRed on macOS via "homebrew" and I just installed the SQLite support nodes via the GUI. Why doesn't that work on Raspbian+Docker?

I need to find time to get to the bottom of all these mysteries...

I also have half a mind to try raising a PR against the base image to see if they'll agree to add the eudev-dev dependency needed for Bluetooth support.

Slyke, 2020-07-15:
I think I can modify the "update all images" command in the menu to upgrade NodeRed. There is a way to have all the dependencies installed with a single command too, I might modify the script and see how well this fares. SQLite will still need its own special command. The way that the developers made it to be able to be installed via npm requires it to be compiled. If anyone finds an alternative though, do let me know! I also want to remove this requirement.

Paraphraser, 2022-06-19:
Combination of pilot error and no further response. Should probably be considered stale.

closed 2022-09-11 by Slyke