SensorsIot/IOTstack #91: Error message from Transmission when trying to compose up

ApricotFarm, 2020-07-26:
After running menu.sh setup, I immediately ran `docker-compose up -d` as suggested in the instructions. I received the following error message:

```
pi@raspberrypi:~/IOTstack $ docker-compose up -d
ERROR: Named volume "transm_config_volume:/config:rw" is used in service "transmission" but no declaration was found in the volumes section.
```

I'm too much of a Docker noob to know how to get in and tinker with this, so I'm afraid I can't provide much insight as to why this is happening. I couldn't find any documentation to fall back to; it looks like the Transmission container is a recent experimental addition to the stack.

Paraphraser, 2020-07-27:
On a Docker-noob scale of 1..10, I might be at 1.01. I also don't use `transmission` so I'm stuck at zero on that noob scale. However, I'll try to give you some pointers based on what I would try if I struck a problem like this.

You can also help by going into `~/IOTstack/docker-compose.yml` and finding the chunks that look like this:

##### *Chunk 1:*

```
  transmission:
    image: linuxserver/transmission
    container_name: transmission
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=Europe/Rome
    volumes:
      - transm_config_volume:/config
      - transm_download_volume:/downloads
      - transm_watch_volume:/watch
    ports:
      - "9091:9091"
      - "51413:51413"
      - "51413:51413/udp"
    restart: unless-stopped
```

##### *Chunk 2:*

```
  transm_download_volume:
    driver: local
    driver_opts:
       o: bind
       type: none
       device: .volumes/transmission/downloads

  transm_watch_volume:
    driver: local
    driver_opts:
       o: bind
       type: none
       device: .volumes/transmission/watch

  transm_config_volume:
    driver: local
    driver_opts:
       o: bind
       type: none
       device: .volumes/transmission/config
```

### Check 1

If what is in your `docker-compose.yml` looks different then please append the relevant chunks of your `docker-compose.yml` to this issue.

### Check 2

If you see both chunks then I suspect that the problem is that each "device" line in chunk 2 is mal-formed. The first one looks like this:

```
       device: .volumes/transmission/downloads
```

but should probably look like this (see "More discussion" below for an explanation):

```
       device: ./volumes/transmission/downloads
```

For all three `device:` lines I'd insert a "/" after the "." then save `docker-compose.yml` and try running `docker-compose up -d` again.

[... 124 lines cut ...]
```
    volumes:
      - transm_config_volume:/config
      - transm_download_volume:/downloads
      - transm_watch_volume:/watch
```

would become:

```
    volumes:
      - ./volumes/transmission/config:/config
      - ./volumes/transmission/downloads:/downloads
      - ./volumes/transmission/watch:/watch
```

I'd also double-check that:

```
~/IOTstack/.templates/transmission/build.sh
```

had been run. It creates the three folders inside `.volumes/transmission` but I think those need to be there before `docker-compose` is run.

I hope something in this helps....

ApricotFarm, 2020-08-05:
@Paraphraser: THANK YOU so much for the detailed and insightful answer. I would say that you're easily a full 2 on the Docker-noob scale... ;-)  At any rate, I very much appreciate the time you've taken to get back to me on this.

Your hunches were all spot-on. **Check 1** and **Check 2** were exactly as you described them, and the fix you suggested (adding a slash after the dot) fixed the problem right up. The `.templates/transmission/volumes.yml` file is being copied accurately into the `docker-compose.yml` file, but the `volumes.yml` file has the same error in it: no slash between the dot and the filename. I suspect this is something that will need to be fixed.

As the whole point of the stack and menu system was to make the installation simple and mindless, I had been hoping not to have to delve into Dockers "compose" mechanism this far. Now that I have, however (and this was, in fact, not difficult), I definitely have a much better sense of how these stack/compose-up files work, and I'm glad I do. Thanks again for your help, and hopefully this can be addressed to in a future release.

Paraphraser, 2020-08-05:
You are welcome. Thanks so much for confirming that the missing slashes are indeed the problem. I have created [Pull Request 102](https://github.com/SensorsIot/IOTstack/pull/102) to change the three lines in the template. With any luck it will get picked up in a future release.

Paulzie, 2020-08-10:
Brand new download and install of Raspberry OS on to a new usb SSD
Install IOTMenu using instructions
Install Transmission plus others through menu

`docker-compose up -d
ERROR: Named volume "transm_config_volume:/config:rw" is used in service "transmission" but no declaration was found in the volumes section.`

Chunk 1 - exists
Chunk 2 - does not exist

Carry out Check 3 to add contents of transmission to docker-compose.yml

`~/IOTstack $ docker-compose up -d
ERROR: The Compose file './docker-compose.yml' is invalid because:
Unsupported config option for services.transm_config_volume: 'driver_opts'
Unsupported config option for services.transm_download_volume: 'driver_opts'
Unsupported config option for services.transm_watch_volume: 'driver_opts'`

`~/IOTstack $ cd ./volumes
~/IOTstack/volumes $ ls
mosquitto`

So default directories do not exist

After creating the directories

`pi@IOTStack:~/IOTstack $ docker-compose up -d
ERROR: The Compose file './docker-compose.yml' is invalid because:
Unsupported config option for services.transm_config_volume: 'driver_opts'
Unsupported config option for services.transm_download_volume: 'driver_opts'
Unsupported config option for services.transm_watch_volume: 'driver_opts'
`

So removed the items in Chunk 2

changed docker-compose.yaml to
`  transmission:
    image: linuxserver/transmission
    container_name: transmission
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=Europe/Rome
    volumes:
      - ./volumes/transmission/config
      - ./volumes/transmission/downloads
      - ./volumes/transmission/watch
    ports:
      - 9091:9091
      - 51413:51413
      - 51413:51413/udp
    restart: unless-stopped`

Docker now starting.  

Files are downloading but no idea where they're actually being placed.  Presume they're in the docker container somewhere.

I now notice that docker has at some point created folders within IOTStack folder named .volume with the transmission folders as they should have been

Paraphraser, 2020-08-11:
Thanks for that comment @Paulzie - it led me to something I had missed when preparing [Pull Request 102](https://github.com/SensorsIot/IOTstack/pull/102), which I have just fixed.

The extra problem was in `.templates/transmission/build.sh`:

```
#!/bin/bash

# create directories for named volumes
TRANSMISSION_BASEDIR=.volumes/transmission
mkdir -p $TRANSMISSION_BASEDIR/downloads
mkdir -p $TRANSMISSION_BASEDIR/watch
mkdir -p $TRANSMISSION_BASEDIR/config
```

The 4<sup>th</sup> line should have been:

```
TRANSMISSION_BASEDIR=./volumes/transmission
```

That explains why you have a `.volumes` directory in your `IOTstack` directory. My guess is that your `.volumes` only contains:

```
transmission/
  downloads
  watch
  config
```

and that each of those three sub-directories below `transmission` is empty. I think it's probably safe to delete `.volumes`.

If you want to test PR102 in advance of it being implemented officially, I think you can do it like this:

```
$ cd ~/IOTstack
$ git remote add -f Paraphraser https://github.com/Paraphraser/IOTstack.git
$ git checkout -b 20200805-transmission-volumes
$ git merge remotes/Paraphraser/20200805-transmission-volumes
```

Then run the menu and do a "full service from template" style of installation for transmission.

The `mkdir -p` commands are non-destructive and will not do anything if those folders are already present in `~/IOTstack/volumes/transmission`.

When it comes time to undo that:

```
$ cd ~/IOTstack
$ git checkout master
$ git branch -D 20200805-transmission-volumes
$ git remote remove Paraphraser
```

Git sometimes offers multiple ways to achieve the same thing. The steps above are what I use when I want to test something from someone else's fork but if you know of a better way then (a) do that and (b) please tell me too.

Paraphraser, 2022-06-19:
Can probably be regarded as fixed.

closed 2022-09-11 by Slyke