SensorsIot/IOTstack #96: Add Unbound DNS to pihole

amamel, 2020-08-01:
Add Unbound support to pihole container

Slyke, 2020-08-01:
Hey @amamel do you have any more details on this?

amamel, 2020-08-01:
Setting up Pihole with Unbound as a recursive DNS server solution. This is advantageous in that no server can fully log the exact paths you're going. Pihole will also verify DNSSEC signatures and discard bogus domains. I've found two sources that give a detailed explanation on how to set this up, but again would be awesome if it was rolled into the IoT install.

https://docs.pi-hole.net/guides/unbound/
https://github.com/notasausage/pi-hole-unbound-wireguard

Slyke, 2020-08-16:
This needs to integrate with https://github.com/SensorsIot/IOTstack/issues/83

ukkopahis, 2022-06-23:
DNSSEC is easily enabled in the admin interface:
[image]

As Unbound (Recursive DNS lookup) requires a bit of installation and setup, and would probably be better implement as a separate "pihole-unbound" service. It would include a Dockerfile to do this setup, extending on the base pihole container. Or maybe Unbound could be installed regardless if it's used or not, not requiring a different service.

Regardless, recursive DNS is a pretty advanced feature, and I'm not convinced it's a good fit into the project. IOTstack doesn't aim for extreme privacy and security. Its target audience are beginners and hackers trying things out for the first time.

If someone were to implement an Unbound enabled Pi-Hole container, I'd be happy to include into documentation how to override the default container with it.

Going to suggest this issue to be closed until an actual docker container implementation is available.

closed 2022-09-11 by Slyke