SensorsIot/IOTstack #97: Add Wireguard support

amamel, 2020-08-01:
Add Wireguard installation template

Slyke, 2020-08-01:
I'll consider this one. It'll be this or openVPN. It'll auto configure. I do have openVPN working with PiHole and optionally allowing forwarding to LAN, but I only managed to get that part running on x86 architecture. ARM kept failing. If Wireguard just "works" then I'll probably use it instead. Wireguard also looks a lot more promising.

SaJaToGu, 2020-08-01:
Hi,
linuxserver/wireguard is a Good Container!
But I have problems to configure the port mapping in my Fritzbox router.
Can you add a documentation for that?

Slyke, 2020-08-16:
@SaJaToGu Not sure about the Fritzbox specifically, but can add some documentation on port forwarding in general. Each router has a different UI, it would be almost impossible (and create a HUUGE amount of work) to add a how-to-port-forward for every router.

closed 2022-09-11 by Slyke