SensorsIot/IOTstack #98: Feature request: Add Ubiquiti UnFi controller support

amamel, 2020-08-01:


Slyke, 2020-08-01:
Hey @amamel do you have more details on this one, too?

amamel, 2020-08-01:
Ubiquiti networking home and enterprise equipment (https://unifi-network.ui.com/) requires the use of a network controller to support monitoring and visualization of network traffic. There is already an existing image that works for Raspberry Pi (https://hub.docker.com/r/ryansch/unifi-rpi) that would be awesome if it could be rolled into the installer as a template and part of the IoT stack.

Slyke, 2020-08-16:
There maybe issues running this with the 64bit version of RaspberryPi OS, it seems they haven't got that image on their dockerhub, but should be fine if the 32bit is installed.

S474N, 2020-08-16:
And will be available support from this script?

Slyke, 2020-08-16:
It will, but it won't work with all available versions of the Raspberry Pi OS. Ubiquiti needs to update their docker image to support the latest Raspberry Pi OS.

S474N, 2020-08-17:
And now is some way to add Controller to Docker? Need them :(

lichtteil, 2020-12-22:
You can try like this: in your IOTstack folder create a `compose-override.yml` and with the content from here https://gist.github.com/lichtteil/83b9f4a733e4ca17e2b25789b3dda3f0