SensorsIot/IOTstack #100: Version in "./docker-compose.yml" is unsupported

kentahikaru, 2020-08-02:
Hello,
i'm trying to follow instructions from README.md (or youtube video) but aftter docker-compose up -d (or via menu.sh) i get following error

pi@amalthea:~/IOTstack $ docker-compose up -d
ERROR: Version in "./docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a version of "2" (or "2.0") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/
pi@amalthea:~/IOTstack $ ./menu.sh 
~/IOTstack ~/IOTstack
checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
An update is available for the project
checking docker version
Docker version >= 18.2.0. You are good to go.
ERROR: Version in "./docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a version of "2" (or "2.0") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/

I am running Raspbian on Raspberry pi 3 B+
pi@amalthea:~/IOTstack $ uname -a
Linux amalthea 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

pi@amalthea:~/IOTstack $ docker -v
Docker version 19.03.12, build 48a6621

my docker-compose.yml (created by menu.sh)
version: '3.6'
services:

  portainer:
    container_name: portainer
    image: portainer/portainer
    restart: unless-stopped
    ports:
      - 9000:9000
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
      - ./volumes/portainer/data:/data

  nodered:
    container_name: nodered
    build: ./services/nodered/.
    restart: unless-stopped
    user: "0"
    privileged: true
    env_file: ./services/nodered/nodered.env
    ports:
      - 1880:1880
    volumes:
      - ./volumes/nodered/data:/data

  influxdb:
    container_name: influxdb
    image: "influxdb:latest"
    restart: unless-stopped
    ports:
      - 8086:8086
      - 8083:8083
      - 2003:2003
    env_file:
      - ./services/influxdb/influxdb.env
    volumes:
      - ./volumes/influxdb/data:/var/lib/influxdb
      - ./backups/influxdb/db:/var/lib/influxdb/backup

  grafana:
    container_name: grafana
    image: grafana/grafana
    restart: unless-stopped
    user: "0"
    ports:
      - 3000:3000
    env_file:
      - ./services/grafana/grafana.env
    volumes:
      - ./volumes/grafana/data:/var/lib/grafana
      - ./volumes/grafana/log:/var/log/grafana

  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto
    restart: unless-stopped
    user: "1883"
    ports:
      - "1883:1883"
    volumes:
      - ./volumes/mosquitto/data:/mosquitto/data
      - ./volumes/mosquitto/log:/mosquitto/log
      - ./volumes/mosquitto/pwfile:/mosquitto/pwfile
      - ./services/mosquitto/mosquitto.conf:/mosquitto/config/mosquitto.conf
      - ./services/mosquitto/filter.acl:/mosquitto/config/filter.acl

I am not familiar with docker.
Please advise. Thank you.

Paraphraser, 2020-08-02:
On my RPi 4

```
$ uname -a
Linux sec-dev 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux
$ docker -v
Docker version 19.03.12, build 48a6621
$ head -1 docker-compose.yml
version: '3.6'
```

Can't think why a Raspbian update would fix anything but perhaps try `sudo apt update; sudo apt upgrade`.

Also, how did you install docker? I've noticed some strange behaviours that were ultimately traced to docker being installed "by hand" using `apt` rather than via the menu. If you didn't install it via the menu, perhaps try `sudo apt remove docker docker-compose` then run the menu and let it install docker. The menu does it like this:

```
curl -fsSL https://get.docker.com | sh
sudo usermod -G docker -a $USER
sudo usermod -G bluetooth -a $USER
sudo apt install -y docker-compose
```

If neither of those work, I'm at a loss to explain what you're seeing.

Paulf007, 2020-08-03:
Get the same error from a fresh install. Did the docker install via the menu as well as the upgrade was also done via the menu.

Paraphraser, 2020-08-03:
OK. I'll try to replicate your problem on a real RPi3B+.

> Spoiler alert: it works for me. I'm hoping that being able to compare/contrast what I did with your process may lead to the "ahah" moment.

#### Starting point:

* Fresh SD card imaged from [Raspberry Pi OS (32-bit) with desktop](https://www.raspberrypi.org/downloads/raspberry-pi-os/):

	```
	2020-05-27-raspios-buster-armhf.zip
	```

If the reason for any particular step in the following seems unclear, you'll probably find the rationale [here](https://gist.github.com/Paraphraser/55643bd660db46716f6c81266905cae8).

> If you do read that gist, please keep in mind that the objective of the gist is to get an RPi4 running on an SSD. You'll have to ignore the bits that talk about cloning the SD to an SSD. The initial setup and the stuff that comes after Raspbian is running on the SSD are the same.

#### Support host

* Permit headless access via ssh and set up WiFi

```
$ touch /Volumes/boot/ssh
$ vi /Volumes/boot/wpa_supplicant.conf
```

* SD card installed in RPi
* Apply power

```
$ ping raspberrypi.local
$ ssh-keygen -R raspberrypi.local
$ ssh-keygen -R 192.168.203.10
$ ssh pi@raspberrypi.local
```

#### RPi 3B+

```
$ sudo raspi-config
```

Set options (including host name of `octopi.local`:

```
1	Change User Password
	⎣ «new password»
2	Network Options
	⎣	N1 Hostname
		⎣ «host name» eg "octopi"
4	Localisation Options
	⎣	I2 Change Timezone
		⎣ «set country and city»
	⎣	I4 Change WLAN Country
		⎣ «two-character upper-case country code» eg "AU"
5	Interfacing Options
	⎣	P3 VNC
		⎣ enable VNC server
6	Advanced Options
	⎣	A5 Resolution
		⎣ (DMT Mode 82 1920x1080 60Hz 16:9)
8	Update
Finish
```

```
$ sudo reboot
```

#### Support host

[... 383 lines cut ...]
Creating portainer ... done
Creating influxdb  ... done

$ docker-compose ps

  Name                 Command                  State                                       Ports                                 
----------------------------------------------------------------------------------------------------------------------------------
grafana     /run.sh                          Up             0.0.0.0:3000->3000/tcp                                                
influxdb    /entrypoint.sh influxd           Up             0.0.0.0:2003->2003/tcp, 0.0.0.0:8083->8083/tcp, 0.0.0.0:8086->8086/tcp
mosquitto   /docker-entrypoint.sh /usr ...   Up             0.0.0.0:1883->1883/tcp                                                
nodered     npm start --cache /data/.n ...   Up (healthy)   0.0.0.0:1880->1880/tcp                                                
portainer   /portainer                       Up             0.0.0.0:9000->9000/tcp                                                

$ uname -a

Linux octopi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux

$ docker -v

Docker version 19.03.12, build 48a6621

$ head -1 docker-compose.yml

version: '3.6'
```

kentahikaru, 2020-08-03:
> On my RPi 4
>
> ```
> $ uname -a
> Linux sec-dev 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux
> $ docker -v
> Docker version 19.03.12, build 48a6621
> $ head -1 docker-compose.yml
> version: [...]

I tried to uninstall docker and reinstall it from menu but no luck. Still the same error.
I tried to use --verbose to maybe get more info but there is nothing useful.
Is there some way how to debug this ?

kentahikaru, 2020-08-03:
Hm, i guess my docker-compose is old ?
pi@amalthea:~/IOTstack $ docker-compose -v
docker-compose version 1.8.0, build unknown

kentahikaru, 2020-08-03:
pi@amalthea:~/IOTstack $ ./menu.sh 
~/IOTstack ~/IOTstack
checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
An update is available for the project
docker not installed
Install Docker
# Executing docker install script, commit: 26ff363bcf3b3f5a00498ac43694bf1c7d9ce16c
+ sudo -E sh -c apt-get update -qq >/dev/null
+ sudo -E sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sudo -E sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
+ sudo -E sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian stretch stable" > /etc/apt/sources.list.d/docker.list
+ sudo -E sh -c apt-get update -qq >/dev/null
+ [ -n  ]
+ sudo -E sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
+ sudo -E sh -c docker version
Client: Docker Engine - Community
 Version:           19.03.12
 API version:       1.40
 Go version:        go1.13.10
 Git commit:        48a6621
 Built:             Mon Jun 22 15:53:00 2020
 OS/Arch:           linux/arm
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.12
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.10
  Git commit:       48a6621
  Built:            Mon Jun 22 15:47:00 2020
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.2.13
  GitCommit:        7ad184331fa3e55e52b890ea95e65ba581ae3429
 runc:
  Version:          1.0.0-rc10
  GitCommit:        dc9208a3303feef5b3839f4323d9beb36df0a9dd
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
If you would like to use Docker as a non-root user, you should now consider
adding your user to the "docker" group with something like:

  sudo usermod -aG docker pi

Remember that you will have to log out and back in for this to take effect!

WARNING: Adding a user to the "docker" group will grant the ability to run
         containers which can be used to obtain root privileges on the
         docker host.
         Refer to https://docs.docker.com/engine/security/security/#docker-daemon-attack-surface
         for more information.
Install docker-compose
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  coinor-libipopt1v5 libmumps-seq-4.10.0 libraw15
Use 'sudo apt autoremove' to remove them.
Recommended packages:
  docker.io
The following NEW packages will be installed:
  docker-compose
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/121 kB of archives.
After this operation, 403 kB of additional disk space will be used.
Selecting previously unselected package docker-compose.
(Reading database ... 144109 files and directories currently installed.)
Preparing to unpack .../docker-compose_1.8.0-2_all.deb ...
Unpacking docker-compose (1.8.0-2) ...
Setting up docker-compose (1.8.0-2) ...
Processing triggers for man-db (2.7.6.1-2) ...

kentahikaru, 2020-08-03:
Why is the raspbian installing old packages?
pi@amalthea:~/IOTstack $ sudo apt show docker-compose
Package: docker-compose
Version: 1.8.0-2
Priority: optional
Section: admin
Maintainer: Felipe Sateler <fsateler@debian.org>
Installed-Size: 403 kB
Depends: python-cached-property, python-docker (>= 1.9.0), python-dockerpty (>= 0.4.1), python-docopt, python-enum34 (>= 1.0.4), python-jsonschema, python-requests (>= 2.6.1), python-six (>= 1.3.0), python-texttable, python-websocket, python-yaml, python:any (<< 2.8), python:any (>= 2.7.5-5~)
Recommends: docker.io (>= 1.9.0)
Homepage: http://docs.docker.com/compose/
Download-Size: 121 kB
APT-Manual-Installed: yes
APT-Sources: http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
Description: Punctual, lightweight development environments using Docker
 docker-compose is a service management software built on top of docker. Define
 your services and their relationships in a simple YAML file, and let compose
 handle the rest.

pi@amalthea:~/IOTstack $ sudo apt show docker.io
Package: docker.io
Version: 1.8.3~ds1-2
Built-Using: glibc (= 2.19-22)
Priority: optional
Section: admin
Maintainer: Paul Tagliamonte <paultag@debian.org>
Installed-Size: 22.8 MB
Depends: adduser, iptables, init-system-helpers (>= 1.18~), perl, libapparmor1 (>= 2.6~devel), libc6 (>= 2.14), libdevmapper1.02.1 (>= 2:1.02.97), libsqlite3-0 (>= 3.5.9)
Recommends: ca-certificates, cgroupfs-mount | cgroup-lite, git, xz-utils
Suggests: aufs-tools, btrfs-tools, debootstrap, lxc, rinse, zfs-fuse | zfsutils
Breaks: docker (<< 1.5~)
Replaces: docker (<< 1.5~)
Homepage: https://dockerproject.com
Download-Size: 4,835 kB
APT-Sources: http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages

Paraphraser, 2020-08-04:
Well, for the record, on the RPi3B+ I used yesterday:

```
pi@octopi:~$ cat /proc/device-tree/model; echo ""
Raspberry Pi 3 Model B Plus Rev 1.3

pi@octopi:~$ docker -v
Docker version 19.03.12, build 48a6621

pi@octopi:~$ docker-compose -v
docker-compose version 1.21.0, build unknown

pi@octopi:~$ apt show docker.io 2>/dev/null | head -2 2>/dev/null
Package: docker.io
Version: 18.09.1+dfsg1-7.1+rpi1+rpt1
```

I don't know enough about `apt` to make sensible comments but, from stuff I have seen elsewhere, I gather that there are some "seed" lists somewhere inside the guts of Raspbian which are used by `apt update` to fetch the information that is used to decide what `apt upgrade` should do. The whole system breaks down if those seed lists are out-of-date but how one goes about fixing that problem is something I don't know. Once I strike a problem like that, I make the assumption that it is unlikely that only one thing is broken, so it's time to start with a clean installation of the most-recent image on a brand new SD card.

If you're running a clean install of IOTstack but are getting obsolete versions then it seems to me that the above scenario is a plausible explanation, provided that you did **not** start from a clean install of Raspbian.

If you **did** start from a clean install of 2020-05-27 Raspbian then I have absolutely no idea what might be going on. Sorry.

kentahikaru, 2020-08-04:
I put fresh raspberry pi os on my SD card and it's working.
docker-composer is 1.21 and it's downloading docker images.

Thank you all !

Paraphraser, 2020-08-04:
Excellent!

closed 2020-08-04 by kentahikaru