SensorsIot/IOTstack #103: Cannot install node-red-contrib-home-assistant-websocket

ronnyandre, 2020-08-06:
When building stack with Node RED and the Home Assistant websocket plugin, the HA plugin is not installed in Node RED. I also tried to execute the npm command directly in the Node RED Docker terminal, and it returns the following error:

```
bash-5.0# npm install node-red-contrib-home-assistant-websocket
npm ERR! code E404
npm ERR! 404 Not Found - GET http://registry.npmjs.org/@hapi%2fjoi - Not found
npm ERR! 404 
npm ERR! 404  '@hapi/joi@17.1.1' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it (or use the name yourself!)
npm ERR! 404 It was specified as a dependency of 'node-red-contrib-home-assistant-websocket'
npm ERR! 404 
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, http url, or git url.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-06T09_32_26_807Z-debug.log
```

The end of the debug log:

```
59 verbose stack Error: 404 Not Found - GET http://registry.npmjs.org/@hapi%2fjoi - Not found
59 verbose stack     at res.buffer.catch.then.body (/usr/local/lib/node_modules/npm/node_modules/npm-registry-fetch/check-response.js:104:15)
59 verbose stack     at process._tickCallback (internal/process/next_tick.js:68:7)
60 verbose statusCode 404
61 verbose pkgid @hapi/joi@17.1.1
62 verbose cwd /usr/src/node-red
63 verbose Linux 5.4.51-v7+
64 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-red-contrib-home-assistant-websocket"
65 verbose node v10.21.0
66 verbose npm  v6.14.4
67 error code E404
68 error 404 Not Found - GET http://registry.npmjs.org/@hapi%2fjoi - Not found
69 error 404
70 error 404 '@hapi/joi@17.1.1' is not in the npm registry.
71 error 404 You should bug the author to publish it (or use the name yourself!)
72 error 404 It was specified as a dependency of 'node-red-contrib-home-assistant-websocket'
73 error 404 Note that you can also install from a
74 error 404 tarball, folder, http url, or git url.
75 verbose exit [ 1, true ]
```

ronnyandre, 2020-08-06:
Found more information on NPM: https://www.npmjs.com/package/@hapi/joi

This package has been deprecated
Author message:
joi is leaving the @hapi organization and moving back to 'joi' (https://github.com/sideway/joi/issues/2411)

Paraphraser, 2020-08-06:
That "deprecated" message seems to be 5 months old but v0.24.0 of the package seems to have been released in the last 24 hours so it looks to me like this is a new bug. I've filed a bug report at [github.com/hassio-addons/addon-node-red/issues/646](https://github.com/hassio-addons/addon-node-red/issues/646) so let's see what happens.

Paraphraser, 2020-08-06:
The response at [hassio-addons/addon-node-red/issues/646](https://github.com/hassio-addons/addon-node-red/issues/646) was:

> This paletted is pre-installed and shipped with the add-on. Manual upgrades or installations are not supported.

Other than the inference that it *should* work, I don't claim to understand what that sentence actually means.

My starting point was the "Install" tab of "Manage Palette" where I could see the add-on but could not get it to install by clicking the `install` button:

[image]

What I tried next was:

```
$ docker exec -it nodered bash
# npm install node-red-contrib-home-assistant-websocket
# exit
$ cd ~/IOTstack
$ docker-compose restart nodered
```

which worked. It is installed inside the container:

```
$ docker exec -it nodered bash
# ls -ld node_modules/node-red-contrib-home-assistant-websocket
drwxr-xr-x    6 root     root          4096 Aug  6 21:20 node_modules/node-red-contrib-home-assistant-websocket
# exit
```

also confirmed by the "Nodes" tab of "Manage Palette":

[image]

So, yes, it can be installed but I'm not sure how persistent it will be. It really needs to be installed at build time. The node is in the build script as confirmed by this:

```
$ grep "node-red-contrib-home-assistant-websocket" ~/IOTstack/.templates/nodered/build.sh
	"node-red-contrib-home-assistant-websocket" " " "OFF" \
```

> the "OFF" meaning "not selected by default".

What you should do is to re-run the menu and, when you come to the list of nodered nodes, make sure to select it. Then, the most reliable way that I have found to rebuild nodered is:

```
$ cd ~/IOTstack
$ docker-compose down
$ docker rmi "iotstack_nodered" "nodered/node-red"
$ docker-compose pull
$ docker-compose up --build -d
```

Please let me know if this works.

ronnyandre, 2020-08-07:
Confirming that the installation through docker bash worked. Thanks @Paraphraser 👍

closed 2020-08-07 by ronnyandre