SensorsIot/IOTstack #104: Backup Error list.txt Permission denied

Cooilleeneyder, 2020-08-14:
Hi,

i get an error with the backup script.
The result is an empty backup on my dropbox :(

pi@intranet:~/IOTstack $ ~/IOTstack/scripts/docker_backup.sh
~/IOTstack ~/IOTstack
/home/pi/IOTstack/scripts/docker_backup.sh: line 8: list.txt: Permission denied
/home/pi/IOTstack/scripts/docker_backup.sh: line 9: list.txt: Permission denied
/home/pi/IOTstack/scripts/docker_backup.sh: line 10: list.txt: Permission denied
Moving old influxdb backups if they exist
backing up influxdb database
influxdb backup complete
/home/pi/IOTstack/scripts/docker_backup.sh: line 15: list.txt: Permission denied
compressing stack folders

Paraphraser, 2020-08-14:
The first thing I will say is that I can't quite marry-up the line numbers mentioned in your error messages with the line numbers in the [docker_backup.sh](https://github.com/SensorsIot/IOTstack/blob/master/scripts/docker_backup.sh) script on GitHub. By my count:

1. Line 8 is a comment.
2. Lines 9 thru 11 mention `list.txt` but only lines 9 and 10 appear in your error messages.
3. Lines 14 and 20 mention `list.txt` in the body of an `if` statement and my guess is that it is the one at line 20 that is turning up as line 15 in your error message.
4. Line 15 is a `fi` statement.

If all the error message line numbers had only been out-by-one then I would have shrugged my shoulders and moved on. But I don't think the explanation for the mismatches can be quite that simple. I think you should probably investigate to see if you can explain this.

That said and assuming that your docker_backup.sh is *roughly* similar to the version on SensorsIot/IOTstack, the relevant lines of the script are:

```
pushd ~/IOTstack
echo "./docker-compose.yml" >list.txt
echo "./services/" >>list.txt
echo "./volumes/" >>list.txt
        echo "./compose-override.yml" >>list.txt
        echo "./backups/influxdb/" >>list.txt
```

so the instances of `list.txt` mean `~/IOTstack/list.txt`. Agreed?

Not being able to create then extend a file in `~/IOTstack` would seem to imply that the permissions on the directory are wrong.

On my system, the directory is:

```
ls -ld ~/IOTstack/
drwxr-xr-x 13 pi pi 4096 Aug 12 14:24 /home/pi/IOTstack/
```

and its contents are:

```
$ ls -l ~/IOTstack/
total 144
drwxr-xr-x 3 pi   pi    4096 Jul 17 23:53 backups
-rw-r--r-- 1 pi   pi    2856 Aug  5 16:08 docker-compose.yml
drwxr-xr-x 2 pi   pi    4096 Aug  6 12:12 docs
drwxr-xr-x 2 pi   pi    4096 Feb  8  2020 duck
-rw-r--r-- 1 pi   pi   35149 Feb  8  2020 LICENSE
-rwxr-xr-x 1 pi   pi   19905 Aug  6 12:37 menu.sh
-rw-r--r-- 1 pi   pi     124 Apr 20 11:19 mkdocs.yml
-rw-r--r-- 1 pi   pi    1306 Jul 24 20:52 README.md
drwxr-xr-x 3 pi   pi    4096 Aug 14 23:41 scripts
drwxr-xr-x 9 pi   pi    4096 Aug  5 13:16 services
drwxr-xr-x 9 root root  4096 Aug  5 13:22 volumes
```

If you don't see a similar pattern but see "root" in more places than `volumes` then perhaps you did something like:

```
$ sudo git clone https://github.com/SensorsIot/IOTstack.git ~/IOTstack
```

when you did the base installation, instead of just:

```
$ git clone https://github.com/SensorsIot/IOTstack.git ~/IOTstack
```

If it wasn't too long ago, you might even be able to confirm that with:

```
$ history | grep "git clone"
```

Possible explanation? If that doesn't ring any bells then I can't think of any other logical reason why permissions would be stuffed up.

In general:

* Everything except `volumes` should have an owner and group of `pi`.
* Most things in `volumes` need an owner and group of `root` but you are better off looking through `services` to see where that gets overridden.

An example of the latter is where:

```
~/IOTstack/services/mosquitto/directoryfix.sh
```

sets:

```
~/IOTstack/volumes/mosquitto
```

and everything below that path to an owner and group of user ID 1883.

Caveat: I have never tried wholesale fuzting with permissions to fix a problem like this so I don't know if there are any gotchas.

You might be better advised to start over with a clean-ish installation. Something like:

```
$ cd ~/IOTstack
$ docker-compose down
$ cd
$ mv IOTstack IOTstack.old
$ git clone https://github.com/SensorsIot/IOTstack.git ~/IOTstack
$ cd IOTstack
$ ./menu.sh
```

Go through the menu and set things up how you want them, bring the stack up then take it down again (to get some basic initialisations), then hand-move the bits of IOTstack.old into IOTstack, fixing the permissions as you go.

Hope this helps.

Cooilleeneyder, 2020-08-14:
Thx a lot, yes, my contents belong to root...

pi@intranet:~ $ ls -l ~/IOTstack/
total 88
drwxr-xr-x 3 root root  4096 Aug 14 14:35 backups
-rw-r--r-- 1 root root  1974 Feb 12  2020 docker-compose.yml
drwxr-xr-x 2 root root  4096 Feb 12  2020 duck
-rw-r--r-- 1 root root 35149 Nov  3  2019 LICENSE
-rwxr-xr-x 1 root root 13188 Feb 12  2020 menu.sh
-rw-r--r-- 1 root root  8447 Feb 12  2020 README.md
drwxr-xr-x 2 root root  4096 Aug 14 16:12 scripts
drwxr-xr-x 8 root root  4096 Feb 12  2020 services
drwxr-xr-x 8 root root  4096 Feb 12  2020 volumes

Since i´m setting up a new system anyway and just wanted the influxdb data and the settings from the other containers, it seems that it is enough to create the list.txt and give it to the owner pi for one time 👍

Paraphraser, 2022-06-19:
Pilot error.

closed 2022-09-11 by Slyke