SensorsIot/IOTstack #105: NODE-RED not able to install

dasalazarle, 2020-08-15:
Hi, any idea why I am not able to install Node-RED? I get the following erro:
ERROR: Service 'nodered' failed to build: error creating aufs mount to /var/lib/docker/aufs/mnt/6fc4c2becaeb615527e0c4b88bfb259114b5f71a14f26395110d13e6fcc6c49d-init: mount target=/var/lib/docker/aufs/mnt/6fc4c2becaeb615527e0c4b88bfb259114b5f71a14f26395110d13e6fcc6c49d-init data=br:/var/lib/docker/aufs/diff/6fc4c2becaeb615527e0c4b88bfb259114b5f71a14f26395110d13e6fcc6c49d-init=rw:/var/lib/docker/aufs/diff/5da31066f8873d60b8d3d1bce5e0e2550b166b8efccda28fab9adaa7ca4a4969=ro+wh:/var/lib/docker/aufs/diff/2820ff712f3d99dabe38e3d3d727aa646af5dc4bedc530085d16e482405372f4=ro+wh:/var/lib/docker/aufs/diff/d3ac95d081d6cc3182673e6572e5b0452232560425ce0a1a5fe247d4d2647d27=ro+wh:/var/lib/docker/aufs/diff/d42c6013397cae57cc7e71419101e8b10234abbbbd8eccaa75968493b60bb62c=ro+wh:/var/lib/docker/aufs/diff/0772d6721a4a5b57a56447a079ea4c1bc4e38279eb972571c8379d00dd90ee6d=ro+wh:/var/lib/docker/aufs/diff/05ea7484c8147dadf5a5b97f25a89270f2eb7a065220ba139a841400586b29f9=ro+wh:/var/lib/docker/aufs/diff/81ae9e728993651e6a53cea7e36aa87f3cf4648fee5e318d19c6aa0394369c0b=ro+wh:/var/lib/docker/aufs/diff/2eb07fff6b64e783713e8241d6f991c7830d9f2e449beecab23755aafde48248=ro+wh:/var/lib/docker/aufs/diff/bc4dff8d2bff05f109928047588af6e94795711927259eff0943ebc10da57173=ro+wh:/var/lib/docker/aufs/diff/a7c3b9932c3885b2e4694fdd2c5bed640a32dd480f846b136e9ad9cec9d1d357=ro+wh:/var/lib/docker/aufs/diff/263c59e30d0324cb90475d48ba953a84e2db3662881ab045043ca48d6e6c4c3b=ro+wh,dio,xino=/dev/shm/aufs.xino: invalid argument

Paraphraser, 2020-08-15:
Absolutely no idea. Unless I'm misreading that, it seems to be talking about the path starting with `/var/lib/docker/aufs/mnt/' and I assume that's an external path on the RPi rather than something internal to the nodered container. My RPis have `/var/lib/docker` but no `aufs` directory so I don't know what to make of that. If it were me, I think I'd start by taking down the stack and rebooting

```
cd ~/IOTstack
docker-compose down
sudo reboot
```

Then, when the Pi came back, I'd see if

```
docker images
```

gave me sensible results. If it did, I'd iterate through the entire list doing

```
docker rmi IDFROMLIST
```

which either works or complains about a container, giving another ID which you can try getting rid of via

```
docker rm IDFROMMESSAGE
```

and keep going until there are no more entries in the images list. Then I'd try to bring the stack up again with:

```
docker-compose up --build -d
```

If that didn't work then I'd declare the whole thing to be hopelessly broken and start over from a clean image of Raspbian.

Slyke, 2020-08-16:
What does `df -Th` output? Seems like there's some issue with the file system.

Are you using the docker from the menu, or was it installed another way?

Paraphraser, 2022-06-19:
No further response. Should probably be considered stale.

closed 2022-09-11 by Slyke