SensorsIot/IOTstack #106: Nome Assistant not installing

dasalazarle, 2020-08-15:
Hi, do you have any idea why I am getting an error when installing hassocks.io?
Thank you, Daniel

ERROR: No NetworkManager support on host.

Paraphraser, 2020-08-15:
Based on your previous issue with nodered, I'm fast coming around to the view that you're best advised to start over with a clean install of Raspbian. This really is too weird.

dasalazarle, 2020-08-15:
> Based on your previous issue with nodered, I'm fast coming around to the view that you're best advised to start over with a clean install of Raspbian. This really is too weird.

thank you, but this is already the third time I am trying to install this, maybe there is another issue that I am not aware of

Paraphraser, 2020-08-15:
Try following [this gist](https://gist.github.com/Paraphraser/55643bd660db46716f6c81266905cae8). Skip the section on cloning to an SSD if you aren't trying to do that. I follow this gist every time and I get reproducible results on both RPi 3 (which I'm currently running from SD) and 4 (two of which I'm currently running from SSD). One caveat: I do install Node-Red but I don't install Home Assistant. I can't think why that would make any difference but I thought I'd mention it.

dasalazarle, 2020-08-15:
thanks, I had Raspberry running using Berryboot, I am installing again and booting from an SSD

Paraphraser, 2020-08-15:
Haven't tried BerryBoot. I watched Andreas' video on the topic but the process gave me the heebies. I decided to stick with the process documented in the gist until such time as I can (a) download a new Raspbian, (b) use BalenaEtcher to transfer that direct to the SSD, (c) plug the SSD into the RPi4, (d) hit the power, and have it all just work.

In other words, just what you can already do on an RPi3, albeit via USB2.

🛫

dasalazarle, 2020-08-15:
unfortunately, after restoring all, formatting sd card and installing Raspbian and moving all to SSD, get same error when trying to install hassocks.io
[Error] No NetworkManager support on host.
too bad.

Paraphraser, 2020-08-15:
First, did re-building the RPi from scratch cure your Node-Red problem?

Second, I don't install any of these things on my own systems so I have no practical experience to draw upon. But I am now wondering if I am guilty of making too many assumptions and not asking enough questions?

For example, the title of this issue is "Nome Assistant not installing." I assumed "Nome" was a typo and that you meant "Home Assistant". I also assumed you meant that you were launching the menu and choosing "Install Hass.io (Requires Docker)".

You have just written "hassocks.io" and I've realised you also wrote "hassocks.io" in your opening remarks. I think I may have "seen" Hass.io and assumed you were launching the menu and choosing "Install Hass.io (Requires Docker)". But, now that you've written "hassocks.io" twice, I'm starting to think we might be talking about two different things. I have just searched through the repo but I can't find any mention of "hassocks" anywhere so I'm really confused.

So that I can be 100% certain that I know what you are talking about, can you please describe the exact steps you are following and, if any of my earlier misunderstandings can be explained away as typos, please double-check to be sure that you are typing exactly what I will need to type if I try to follow along.

Tomorrow, I will try to follow your steps on my "test" RPi4 to see if I strike the same problem.

🛫

Slyke, 2020-08-16:
Hey @dasalazarle can you type the following command:
```
uname -a
```

And paste the output?

Also, which architecture are you selecting in the menu when asked?

dasalazarle, 2020-08-16:
> Hey @dasalazarle can you type the following command:
> 
> ```
> uname -a
> ```
> 
> And paste the output?
> 
> Also, which architecture are you selecting in the menu when asked?

Linux raspitv 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

dasalazarle, 2020-08-16:
> First, did re-building the RPi from scratch cure your Node-Red problem? Second, I don't install any of these things on my own systems so I have no practical experience to draw upon. But I am now [...]

Hi, I just followed the instructions given by Andreas and selected the option to install hass.io, that is all.

alexdelli, 2020-08-20:
I've the same error. Raspi 4 64 bit

dasalazarle, 2020-08-28:
> Same here:
> Linux darkstar 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
> running on an pi4/4GB with SSD boot
> 
> Selecting pi4 architecture when installing hassio from menu.sh
> 
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> avahi-daemon is already the newest version (0.7-4+b1).
> bash is already the newest version (5.0-4).
> curl is already the newest version (7.64.0-4+deb10u1).
> dbus is already the newest version (1.12.20-0+deb10u1).
> jq is already the newest version (1.5+dfsg-2+b1).
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> [Warning]
> [Warning] If you want more control over your own system, run
> [Warning] Home Assistant as a VM or run Home Assistant Core
> [Warning] via a Docker container.
> [Warning]
> [Error] No NetworkManager support on host.
> 
> After browsing thru the scripts the problem is on the hassio supervisor-installer side, not on IOTstack.
> After checking the installer script there, it looks like on my system nmcli and apparmor_parser are missing.
> sudo apt install network-manager and
> sudo apt install apparmor
> fixed the problem and hassio is running now.
> 
> After browsing thru the commits of the installer, 10 days ago, the missing of both commands where changed from warn to error what explains the broken installation.
> 
> Hope that helps
> 
> //br
> Stefan

thank you, this solved all the issues

kzgrey, 2020-09-15:
The Network-Manager issue needs to be resolved prior to installing Docker and containers.  If Docker is installed first, then the routing will be screwed up after Network-Manager is installed.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke