SensorsIot/IOTstack #108: Using Teslamate with IOTstack

ronnyandre, 2020-08-17:
I want to install Teslamate on my Pi with IOTstack already running lots of containers.

However, Teslamate installed with Docker, has its own set of Grafana, Mosquitto and Postgres DB. How can I install Teslamate without interferring with IOTstack containers?

Teslamate docs: https://docs.teslamate.org/docs/installation/docker

Slyke, 2020-08-18:
It's possible to do. We just use different service names, much like we do with Nextcloud (it has its own MariaDB server). I think Teslamate is a bit niche though for IOTstack, at least right now. I have some ideas on reorganising the menu, as we have a lot of similar services (Such as, Databases, Networking, 3rd party Integrations and so on).

I didn't know about Teslamate, I think I might set it up myself (I own a Model 3)!

ronnyandre, 2020-08-18:
Thanks for your reply. For once I actually did some research trying to implement it with IOTstack, with success. I didn't make a separate folder for Teslamate, but I used the `compose-override.yml` file method.

*TL;DR:*

1. Install Grafana and PostgreSQL.
2. Add content to `compose-override.yml` file.
3. Create the `./volumes/teslamate` folder.
4. Create Teslamate database credentials and make it superuser.
5. Add Teslamate data source in Grafana.
5. Install missing Grafana plugins.
6. Copy dashboard files to Grafana.

*Long version:*

This is the content of the override file:

```yaml
services:
  teslamate:
    container_name: teslamate
    image: teslamate/teslamate:latest
    restart: always
    environment:
      - DATABASE_USER=teslamate
      - DATABASE_PASS=secret
      - DATABASE_NAME=teslamate
      - DATABASE_HOST=postgres
      - MQTT_HOST=mosquitto
    ports:
      - 4000:4000
    volumes:
      - ./volumes/teslamate/import:/opt/app/import
    cap_drop:
      - all
```

I changed the configuration slightly from the official Teslamate documentation, which is the mounting point for the import volume. (Not sure if it worked, because the `./volumes/teslamate/import` folder was empty. I also updated the environment settings pointing to the PostgreSQL database installed via IOTstack. I also used Adminer for easy creation of the user and database. PS: The teslamate user has to be superuser in PostgreSQL.

After installing the Teslamate docker through IOTstack, I followed the [manual setup of Grafana!](https://docs.teslamate.org/docs/installation/debian#import-grafana-dashboards), by setting up the new data source for Teslamate and copy/paste of the dashboard files. After this I experienced some missing plugins in Grafana that I installed via `grafana-cli` (`grafana-cli plugins install PLUGIN_NAME`) through `docker exec`.

I haven't looked into the Mosquitto settings yet, as I had to put this away before I went to bed, but I will set up my Home Assistant as the MQTT broker.

closed 2020-12-15 by ronnyandre