SensorsIot/IOTstack #110: rclone filling up trash on gdrive

petinox, 2020-08-23:
I noticed that my backups had stopped syncing to Gdrive by chance. It turns out the rclone sync command fills up the trash folder and then fails silently in the backup script.

I added **--drive-use-trash=false** to the rclone sync line in docker-backup.sh as per the [documentation](https://rclone.org/drive/), that seems to fix the problem. Would it make sense to add that change to the repo?

Also, does anyone have suggestions on some kind of alarm if the backup is not successful? It's a bit disconcerting that this was a completely silent error.

Paraphraser, 2020-08-25:
The one time something similar happened to me, I noticed a "you have mail" when I logged-into the RPi via SSH. That alerted me to the problem.

You could try augmenting the backup script with a `mosquitto_pub` aimed at the local Mosquitto container (in this case, the loopback address 127.0.0.1 would work because the backup process is running **outside** Docker), with a suitable topic and payload indicating either success or failure. Node-Red could subscribe to that topic and then you could do what you liked (send yourself an email, a notification with PushSafer or similar).

Even just an "it worked" on success and silence on failure would get you a bit further in front. You'd eventually think, "hey, I wonder why I haven't seen any backup confirmations for a while?"

Does that help?

petinox, 2020-08-25:
Thank you for the suggestions. MQTT for the notification sounds like a good idea, I'll consider that. The trickier bit to me is to determine the success or failure of the backup script. I saw that you have a repo with a backup/restore solution for the IOTstack, I'll have a look there for inspiration.

The simplest/crudest approach for me right now would be to set up a file watcher in NodeRed that checks the Gdrive folder and alerts if new backups do't arrive on time, or if they deviate too much in size.

Paraphraser, 2022-06-19:
Probably qualifies as fixed.

closed 2022-09-11 by Slyke