SensorsIot/IOTstack #115: Error getting docker version

Roberto6969, 2020-09-18:
I am curious if this error could be fixed and if, also how:

Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            new-menu   -> FETCH_HEAD
Project is up to date
Python Version: 'Python 3.7.3'. Python is up to date.
PyYAML Version: 'pyyaml 5.3.1'. PyYAML is up to date.
Blessed Version: 'blessed 1.17.8'. Blessed is up to date.
**Error getting docker version. Error when running docker command. Check that docker is installed correctly.**
Docker version 19.03.12 >= 18.2.0. Docker is good to go.
Project dependencies up to date

Thank You!

Paraphraser, 2020-09-18:
That error message doesn't seem to be in menu.sh so my guess is that it is coming from the following command:

```
docker version -f "{{.Server.Version}}"
```

which is called by menu.sh. Can you try that and see what you get?

I have three RPis, two RPi4 plus one 3B+. One of the 4s is my "live" machine and has been running for about a year, just updated as and when updates become available. It started with gcgarner/IOTstack and migrated to SensorsIot/IOTstack. The other 4 is a test machine running against a fork of SensorsIot, about 6 months since the last rebuild, but otherwise updated as-and-when. The 3 is a fresh build two days ago starting from the most current Raspbian plus SensorsIot.

Across all that variety of lineage, all three RPis have the same response to the above command: `19.03.12`. No error messages.

If you get an error from that command then it probably means what it says. How long ago did you install? And did you do it by:

1. Starting with a Raspbian image to which you added any package you liked **except** docker and docker-compose.
2. cloning IOTstack into `~/IOTstack` with that exact spelling, where `~` expands to `/home/pi`, and where `pi’ has user ID 1000.
3. `cd ~/IOTstack`
4. `./menu.sh` or equivalent
5. choosing "Install Docker" from the first line of the menu.

If you think you may have departed from that sequence then that might be the explanation. I'm not saying departures are somehow **wrong**. But problems have often been traced to doing something other than the above so those are the first things I think of when people have issues I can't replicate.

I hope this helps.

Roberto6969, 2020-09-18:
Thank You - I have to do some further investigations, but here is just a quick answer on your first question:

$ docker version -f "{{.Server.Version}}"
**19.03.12**

Paraphraser, 2020-09-18:
Hmmm. If I ssh, go into IOTstack, run the menu, and cancel, this is all I see:

```
pi@octopi:~$ cd IOTstack/
pi@octopi:~/IOTstack$ ./menu.sh 
~/IOTstack ~/IOTstack
checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
checking docker version
Docker version >= 18.2.0. You are good to go.
pi@octopi:~/IOTstack$
```

This is on the 3B+ so very recent clean install. The obvious difference is the python checks. I can't see where those are coming from. I thought it might indicate the presence of a compose-override.yml but that didn't cause mine to mimic yours. What does a `git status` say? Any unexpected changes?

If it were my system, I think I'd start putting extra "echo" statements into menu.sh to see if I could nail down the line that was producing the message.

Roberto6969, 2020-09-21:
_git status_ returns:

$ git status
**On branch new-menu
Your branch is up to date with 'origin/new-menu'.

nothing to commit, working tree clean**

Back to your first answer: it is a Raspberry Pi 4, created at the end of last year. It runs IOTstack beside fail2ban. If I remeber well, I have installed docker before (and not through) the IOTstack. I have replaced SD card with SSD a few months ago, but I believe this has no impact on this error.

Idea with ECHO is fine, but I have no idea which part of menu.sh is under question to "echo-ing" it.

Paraphraser, 2020-09-21:
Installing docker outside of menu.sh is likely to be the source of the problem. This is the chunk of code in menu.sh that does the work:

[image]

That's what you have to mimic. You probably won't find the "bluetooth" line in your copy because that's something from a pending pull request. I think you might be able to get away with downing your stack, removing docker and docker-compose (using sudo apt remove), then running the menu, choosing the first option to let the script reinstall docker and docker-compose, then rebooting and bringing the stack up again.

Why does the menu do the installation the way it does? No idea. Why can't you install docker using sudo apt install? No idea. Just seems to be one of those things.

The trick with echo commands is just to start sprinkling them about, `echo "got to 1"` and so on, then see what you hit and drill down.

But,  now that you have explained that you didn't install docker via the menu, the original "Check that docker is installed correctly" error is probably saying "you didn't install docker via the menu - you need to uninstall and start again".

You **might** be able to get away with just doing this:

```
$ sudo usermod -G docker -a $USER
$ sudo reboot
```

but, on balance, I think the install step in the script of `curl -fsSL https://get.docker.com | sh` is probably pretty important.

Roberto6969, 2020-10-02:
We can close this issue. I agree the best way is installing by choosing "Install Docker" from the first line of the menu. But so far, my whole installation works well and the error doesn't impact to any function.
Thank You for your advices and ideas!

closed 2020-10-02 by Roberto6969