SensorsIot/IOTstack #117: Homeassistant doen not start ( standard_init_linux.go:211: exec user process caused "exec format error")

vladacr, 2020-09-19:
Thanks for the great IOTstack, first of all.

I got myself a new RPi4, installed Raspbian and IOTstack. From here I installed Docker, rebooted and then installed Hassio, followed with Portainer and couple of other containers. The problem is, HA won't start, when I try to start it from Portainer, only an error occurs:
`standard_init_linux.go:211: exec user process caused "exec format error"`
How can I fix this?

Thanks

vladacr, 2020-09-19:
Sorry for the mess, just wanted to add that all other containers (supervisor, cli, audio, etc) are running.

Paraphraser, 2020-09-20:
Apologies for the screen shot - I posted this solution on the IOTstack Discord channel.

[image]

I tested this on an RPi3B+ starting from the latest Raspbian image. I have not tried it on either of my RPi4 devices but I can't think why those should be any different.

If you are running "headless" and communicating with your Pi over SSH and/or VNC, I strongly recommend connecting the Pi to Ethernet and making sure your comms sessions are connecting to the Ethernet port **before** you start. The random WiFi MAC address changing behaviour that Network Manager introduces IN THE MIDDLE OF THE INSTALLATION gives me the screaming heebies.

Having said all that, I am **not** a Hass.io user. When I say this solution works, I mean "it doesn't throw up any errors during installation and the container seems to come up without error." Whether the service that is installed is fit for purpose - entirely different question.

vladacr, 2020-09-20:
All my devices are always on ethernet with fixed IP set, if possible, so is this one, I forgot to say, sorry. Also apparmor and network-manager have both been installed as well as system update & upgrade prior to IOTstack. Installing core HA via IOTstack's docker-compose works fine, but then there is no access to the bluetooth even if with `network=host` option, that's why I wanted to give Hassio a try in spite of preferring having full control of my systems.

I'm at my wit's end...

Paraphraser, 2020-09-20:
[This might help](https://github.com/SensorsIot/IOTstack/issues/64). It's about getting Bluetooth to work under Node-Red so it isn't directly applicable but the issues may be similar.

Paraphraser, 2022-06-19:
Stale issue.

closed 2022-09-11 by Slyke