SensorsIot/IOTstack #118: NodeRed can not install from menu.sh

IT-Berater, 2020-09-21:
Install from menu.sh with or without addons generated the Error:

`ERROR: Service 'nodered' failed to build: The command '/bin/sh -c for addonnodes in  ; do npm install ${addonnodes} ;done;' returned a non-zero code: 139`

And manuel
`
$ cd ~/IOTstack
$ docker-compose down
$ docker rmi "iotstack_nodered" "nodered/node-red"
$ docker-compose pull
$ docker-compose up --build -d
`
the same Error. Please help.

Paraphraser, 2020-09-21:
Can you please do a

```
cat ~/IOTstack/services/nodered/Dockerfile
```

And paste the results into this issue. Here is what I see:

```
FROM nodered/node-red:latest
USER root
RUN apk update
RUN apk upgrade
RUN apk add --no-cache eudev-dev
RUN for addonnodes in \
node-red-node-pi-gpiod \
node-red-dashboard \
node-red-contrib-influxdb \
node-red-contrib-boolean-logic \
node-red-node-rbe \
node-red-configurable-ping \
node-red-node-email \
node-red-contrib-generic-ble \
; do \
npm install ${addonnodes} ;\
done;
RUN npm install --unsafe-perm node-red-node-sqlite
```

There **will** be differences. Yours probably will have the `FROM`, omit the next 4 lines (which are part of a pending pull request), then have the `RUN for`, and then my guess is it will go straight to the `; do` line. If that's what you see then the basic problem will be that you did not choose at least one Node-Red node to be installed via menu.sh

IT-Berater, 2020-09-22:
I see:
<pre>
FROM nodered/node-red:latest
RUN for addonnodes in \
node-red-node-pi-gpiod \
node-red-dashboard \
node-red-contrib-influxdb \
node-red-contrib-boolean-logic \
node-red-node-rbe \
node-red-configurable-ping \
node-red-node-email \
; do \
npm install ${addonnodes} ;\
done;
</pre>

IT-Berater, 2020-09-22:
Next try. Select one node "node-red-dashboard". The same error:

<pre>
pi@pi-flug:~/IOTstack $ ./menu.sh
~/IOTstack ~/IOTstack
checking for project update
From https://github.com/IT-Berater/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
checking docker version
Docker version >= 18.2.0. You are good to go.
Adding portainer container
...portainer service not overwritten
Adding nodered container
...pulled nodered excluding env file
no override found, using docker-compose.yml
docker-compose successfully created
run 'docker-compose up -d' to start the stack
pi@pi-flug:~/IOTstack $ cat ~/IOTstack/services/nodered/Dockerfile
FROM nodered/node-red:latest
RUN for addonnodes in \
node-red-dashboard \
; do \
npm install ${addonnodes} ;\
done;
pi@pi-flug:~/IOTstack $ docker-compose up -d
Building nodered
Step 1/2 : FROM nodered/node-red:latest
 ---> fa3bc6f20464
Step 2/2 : RUN for addonnodes in node-red-dashboard ; do npm install ${addonnodes} ;done;
 ---> Running in cd9a8bbd6f98
ERROR: Service 'nodered' failed to build: The command '/bin/sh -c for addonnodes in node-red-dashboard ; do npm install ${addonnodes} ;done;' returned a non-zero code: 139
</pre>

Paraphraser, 2020-09-22:
Take the stack down:

```
$ cd IOTstack/
$ docker-compose down
Stopping nodered   ... done
Stopping postgres  ... done
Stopping portainer ... done
Stopping gitea     ... done
Stopping mosquitto ... done
Stopping pihole    ... done
Stopping influxdb  ... done
Stopping grafana   ... done
Removing nodered   ... done
Removing postgres  ... done
Removing portainer ... done
Removing gitea     ... done
Removing mosquitto ... done
Removing pihole    ... done
Removing influxdb  ... done
Removing grafana   ... done
Removing network iotstack_default
```

Make my Dockerfile look like yours:

```
$ cd services/nodered/
$ cp Dockerfile Dockerfile.bak
$ vi Dockerfile
$ cat Dockerfile
FROM nodered/node-red:latest
RUN for addonnodes in \
node-red-node-pi-gpiod \
node-red-dashboard \
node-red-contrib-influxdb \
node-red-contrib-boolean-logic \
node-red-node-rbe \
node-red-configurable-ping \
node-red-node-email \
; do \
npm install ${addonnodes} ;\
done;
```

Remove the base and built images:

```
$ cd ../..
$ docker rmi iotstack_nodered nodered/node-red
Untagged: iotstack_nodered:latest
Deleted: sha256:cef5ba13a7553945c96146e04b5862ac2cba5db65d8c3ef74ef0dd691afb47c5
Deleted: sha256:7b7f256e88fd95c466c1e8cd68752134a662a35c7f233ad64b070652bed718c7
Untagged: nodered/node-red:latest
Untagged: nodered/node-red@sha256:7903d6a8a0a96ac198c0349aa3f4aad3ad6568d1733b447403fdc67522b477fa
Deleted: sha256:fa3bc6f20464f4689676d2d70688d9192d2b4a1a70356594747037e91dc3d824
Deleted: sha256:667e4604d0292ff1984f38e92b1265f8f1d64659fd2d67e09d34335d5b47a350
Deleted: sha256:3018bfb3d851ab218c7511b0375b19cd3fe23a5f7c01620834f9ae8c2985a1eb
Deleted: sha256:be7d2190aa5569dfe22e1d78bccc719ce9e5c3637e8e2a3a419185058793457c
Deleted: sha256:e69b8e05fbd9e0ff6f6eb3b15ae915f22a93776cea9d54dd2f000d023fe1f266
Deleted: sha256:7cb5f9de12c874fa1c06e9f109385126b32e42fb828c1fa15454b89ca6d0a62a
Deleted: sha256:63205d168d3d82b1c954825e4865cf6726197b9b346c65278a9dded23793d3c6
Deleted: sha256:a1a8d46d00c34adb7e498d019dc23557f702ff48087b839e74bbaa64c0d3634f
Deleted: sha256:063848250438345ad5ed4fa0a6426d42054422f22cfad9ce945333834d00f908
Deleted: sha256:f44c0bc0f10074c35f22d2ff1be8a6c5c1859c4af4ba4725b8df4b6c80ab6ac5
Deleted: sha256:2c03c798c9e32fcb0809c309aa350e2017d99cea45c62b9314d89b88c1f6ba29
Deleted: sha256:093b0aa59d568b7b9472633c090a409cb2958fc337253680bc88ba591379d4f4
```

Build the stack. Notice "fa3bc6f20464" (20 lines down) which is the same as in your output so we are using the same base image.
[... 150 lines cut ...]
node-red-contrib-boolean-logic \
node-red-node-rbe \
node-red-configurable-ping \
node-red-node-email \
; do \
npm install ${addonnodes} ;\
done;
```

That has the effect of bringing the Alpine image up-to-date but, if that actually works, I won't be able to explain why. As far as I'm concerned, if I'm using your Dockerfile "as is" and we are getting the same checksum on the base image, we should be identical.

The nagging doubt in my mind is whether some part of a previous nodered build with the update/upgrade in place is lying in some obscure corner of Docker and it's "helping" me.

Another nagging doubt is this line from your output:

```
pulled nodered excluding env file
```

Maybe cat your environment file as well. Mine is a one-liner:

```
$ cat ~/IOTstack/services/nodered/nodered.env 
TZ=Australia/Sydney
```

IT-Berater, 2020-09-22:
Build stack fails:

<pre>

pi@pi-flug:~/IOTstack $ docker-compose down
Removing network iotstack_default
pi@pi-flug:~/IOTstack $ cat ~/IOTstack/services/nodered/Dockerfile
FROM nodered/node-red:latest
RUN for addonnodes in \
node-red-node-pi-gpiod \
node-red-dashboard \
node-red-contrib-influxdb \
node-red-contrib-boolean-logic \
node-red-node-rbe \
node-red-configurable-ping \
node-red-node-email \
; do \
npm install ${addonnodes} ;\
done;
pi@pi-flug:~/IOTstack $ docker rmi iotstack_nodered nodered/node-red
Error: No such image: iotstack_nodered
Error response from daemon: conflict: unable to remove repository reference "nodered/node-red" (must force) - container cd9a8bbd6f98 is using its referenced image fa3bc6f20464
pi@pi-flug:~/IOTstack $ docker rmi -f iotstack_nodered nodered/node-red
Untagged: nodered/node-red:latest
Untagged: nodered/node-red@sha256:7903d6a8a0a96ac198c0349aa3f4aad3ad6568d1733b447403fdc67522b477fa
Deleted: sha256:fa3bc6f20464f4689676d2d70688d9192d2b4a1a70356594747037e91dc3d824
Error: No such image: iotstack_nodered
pi@pi-flug:~/IOTstack $ docker images
REPOSITORY            TAG                 IMAGE ID            CREATED             SIZE
portainer/portainer   latest              dbf28ba50432        2 months ago        62.5MB
pi@pi-flug:~/IOTstack $ docker-compose up -d
Creating network "iotstack_default" with the default driver
Building nodered
Step 1/2 : FROM nodered/node-red:latest
latest: Pulling from nodered/node-red
3cfb62949d9d: Already exists
abb6df863fb2: Already exists
482c98ca0488: Already exists
6bf11bc811e0: Already exists
743c9169b9e3: Already exists
96c6c09f80bb: Already exists
55aedf8158fe: Already exists
d8f3d6fa90a6: Already exists
dfd8198df512: Already exists
75c529c9f436: Already exists
930adc80b9f5: Already exists
Digest: sha256:7903d6a8a0a96ac198c0349aa3f4aad3ad6568d1733b447403fdc67522b477fa
Status: Downloaded newer image for nodered/node-red:latest
 ---> fa3bc6f20464
Step 2/2 : RUN for addonnodes in node-red-node-pi-gpiod node-red-dashboard node-red-contrib-influxdb node-red-contrib-boolean-logic node-red-node-rbe node-red-configurable-ping node-red-node-email ; do npm install ${addonnodes} ;done;
 ---> Running in 0150a0440224
ERROR: Service 'nodered' failed to build: The command '/bin/sh -c for addonnodes in node-red-node-pi-gpiod node-red-dashboard node-red-contrib-influxdb node-red-contrib-boolean-logic node-red-node-rbe node-red-configurable-ping node-red-node-email ; do npm install ${addonnodes} ;done;' returned a non-zero code: 139

</pre>

IT-Berater, 2020-09-22:
Ok, last try. Clean all images

<pre>

pi@pi-flug:~/IOTstack $ docker-compose up -d
Building nodered
Step 1/2 : FROM nodered/node-red:latest
latest: Pulling from nodered/node-red
3cfb62949d9d: Pull complete
abb6df863fb2: Pull complete
482c98ca0488: Pull complete
6bf11bc811e0: Pull complete
743c9169b9e3: Pull complete
96c6c09f80bb: Pull complete
55aedf8158fe: Pull complete
d8f3d6fa90a6: Pull complete
dfd8198df512: Pull complete
75c529c9f436: Pull complete
930adc80b9f5: Pull complete
Digest: sha256:7903d6a8a0a96ac198c0349aa3f4aad3ad6568d1733b447403fdc67522b477fa
Status: Downloaded newer image for nodered/node-red:latest
 ---> fa3bc6f20464
Step 2/2 : RUN for addonnodes in node-red-node-pi-gpiod node-red-dashboard node-red-contrib-influxdb node-red-contrib-boolean-logic node-red-node-rbe node-red-configurable-ping node-red-node-email ; do npm install ${addonnodes} ;done;
 ---> Running in d62cfcb43ec7
ERROR: Service 'nodered' failed to build: The command '/bin/sh -c for addonnodes in node-red-node-pi-gpiod node-red-dashboard node-red-contrib-influxdb node-red-contrib-boolean-logic node-red-node-rbe node-red-configurable-ping node-red-node-email ; do npm install ${addonnodes} ;done;' returned a non-zero code: 139

</pre>

with your Dockerfile. It not works on Raspberry Pi Model B Rev 2. :-( 

Thank you very much for the quick and detailed help. It's a shame that it doesn't work. Would like to have the nodeRed in a docker container. If so, try another, newer device.

IT-Berater, 2020-09-22:
Raspberry Pi Model B Rev 2 is a ARM**v6**-compatible processor rev 7 (v6l). Currently there is a bug in Docker’s architecture detection that fails for arm32v6 - eg Raspberry Pi Zero or 1. For these devices you currently need to specify the full image tag, for example:

> docker run -it -p 1880:1880 --name twnodered nodered/node-red:1.1.0-10-arm32v6

Source: https://nodered.org/docs/getting-started/docker

Works for my :-)

IT-Berater, 2020-09-22:
Fix in ./services/nodered/Dockerfile, add the version number in line 1:

FROM nodered/node-red:**1.1.3-12-minimal-arm32v6**

and **IOTstack** menu.sh works with Rasberry Pi Model B Rev 2 and NodeRed. Thanks.

closed 2020-09-22 by IT-Berater

IT-Berater, 2020-09-22:
See http://blog.wenzlaff.de/?p=15332 (german)

Paraphraser, 2020-09-22:
Brilliant!