SensorsIot/IOTstack #119: Problem with docker-compose

lrzdeveloper, 2020-09-22:
Hi guys. I tried to install Portainer and Pi-Hole, but every time I used ./menu.sh doesn't work: when I digit 'cat docker-compose.yml' it only shows

version: '3.6'
services: 

Can someone help me with this problem? Because I can't use "docker-container up -d" if services cannot exist.
I tried on Debian 10 and also on Raspberry Pi3 with Raspian.

jimvarges, 2020-09-22:
I am facing the same problem on my RPi 3b+ running Ubuntu 18.04. Docker compose is 1.17.1 build. Could someone help?

jimvarges, 2020-09-23:
The comments by the gentleman in this thread seems to have helped. https://github.com/SensorsIot/IOTstack/issues/115

Start with a clean Raspbian install without Docker.

Paraphraser, 2020-09-23:
I can't speak for anything other than standard Raspbian on a 3B+

```
$ uname -a
Linux octopi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux.

$ docker -v
Docker version 19.03.13, build 4484c46

$ docker-compose -v
docker-compose version 1.21.0, build unknown

$ wc -l IOTstack/docker-compose.yml 
112 IOTstack/docker-compose.yml

$ DPS
NAMES               CREATED             STATUS
gitea               40 hours ago        Up 40 hours
influxdb            40 hours ago        Up 40 hours
portainer           40 hours ago        Up 40 hours
nodered             40 hours ago        Up 40 hours (healthy)
grafana             40 hours ago        Up 40 hours
pihole              40 hours ago        Up 40 hours (healthy)
mosquitto           40 hours ago        Up 40 hours
```

> DPS is an alias for `docker ps` with a filter for fewer columns - the intention here is to show containers running.

Paraphraser, 2020-09-23:
[This is the process I follow](https://gist.github.com/Paraphraser/55643bd660db46716f6c81266905cae8) for RPi3 and RPi4. I followed it for the RPi3 in the earlier comment less than a week ago, starting from [Raspberry Pi OS (32-bit) with desktop Release date:2020-08-20](https://www.raspberrypi.org/downloads/raspberry-pi-os/). If you only want to run from SD rather than SSD, just skip "Cloning the SD to an SSD".

If Hass.io is your goal, then **BEFORE** you start the "Install Docker & IOTstack" section, connect your Pi to Ethernet and connect your ssh session to the IP address associated with the Ethernet port, then do:

```
$ sudo apt install -y apparmor apparmor-profiles apparmor-utils
$ sudo apt install -y network-manager
```

After that, do the "Install Docker & IOTstack" steps. I'm not a Hass.io user (not in a pink fit!) but I got it to install on this same RPi3B+, albeit without the `apparmor-utils`. Andreas Spiess posted a note about the utils on Discord, which is why I've added it here.

> Unless you like a world of pain, don't ignore the advice about Ethernet. The network-manager install changes the MAC address of the WiFi interface **in the middle of the installation process**. That's why I say "not in a pink fit". If Hass.io depends on network-manager, and network-manager takes it into its tiny head to randomise WiFi MACs, then there's no way it's coming anywhere near any system of mine, and if that precludes Hass.io, so be it! 🤮🤮🤮☠️

lrzdeveloper, 2020-09-23:
Thank you so much for your help! I installed correctly pihole and portainer (both are in docker-compose.yml). Now I've got a problem with Home Assistant: no error shows up and in docker-compose.yml (homeassistant), but it doesn't appear and it clearly isn't working.

[image]

In this screenshoot I used Debian 10, I tried to install HomeAssistant with menu.sh and i chose raspberrypi4 and raspberrypi3

Paraphraser, 2020-09-23:
When you say it doesn't appear, what do you mean?

Can you see it in `docker ps` ?

If you have been trying that IP address (172.16.29.137) and getting no response, and that is what you mean by doesn't appear then try the IP address of your Raspberry Pi with the port 8123.

172.16 is used by the internal bridged network. In `docker ps` you will see things like `mosquitto           0.0.0.0:1883->1883/tcp` which, in the case of Mosquitto says that external port 1883 (on the RPi proper) is mapped to internal port 1883 on the bridged network (the 172.16 bit). The internal and external ports don't have to be the same which is why I suggest looking at `docker ps`. You might like to post the output you get into this case.

lrzdeveloper, 2020-09-23:
Yes, in `docker ps` I see, but in docker-compose.yml I expect to see the configuration about Home Assistant. 

[image]

Paraphraser, 2020-09-23:
I don't run Hass.io (I assume that's the same as "home assistant") so I don't know what "normal" looks like in `docker ps`, and I also don't know what to expect in docker-compose.yml. The only thing I see here that I might try next is port 4357 - what happens if you launch a browser and point it to that port on your RPi?

I have no idea why your earlier message burbled on about port 8123 when it doesn't appear here.

You can sometimes get some ideas by running `docker inspect imageid` where you get the imageid from `docker images`. The output is JSON.

Paraphraser, 2022-06-19:
I think this is either pilot error or a hass.io thing. Seeing as hass.io now does its own thing and is incompatible with IOTstack, this is probably a stale issue.

closed 2022-09-11 by Slyke