SensorsIot/IOTstack #120: Option max-size not working with docker journald logging

weissm, 2020-09-23:
When running the standard generated docker-compose file, the "max-size" option causes an error @ docker-compose when working with journald logging in docker. When removing this in the generated docker-compose everything works fine.

I found this defined in the templates and wondering if it shall be removed there or if there is a generic way to handle this?

Thanks in advance,
Matthias

Paraphraser, 2020-09-23:
I'm really confused

```
$ grep -iR "max-size" ~/IOTstack/* 2>/dev/null 
$
```

No hits in my installation so where are you seeing "max-size"?

weissm, 2020-09-23:
Thanks for quick feedback. I was refering to the experimental branch, e.g.
https://github.com/SensorsIot/IOTstack/blob/experimental/.templates/openhab/service.yml

logging:
      options:
        max-size: "5m"
        max-file: "3"

thx!

Paraphraser, 2020-09-23:
OK. For experimental branch issues, you need to wait for @Slyke to respond.

Slyke, 2020-09-28:
Hey @weissm what version of docker and docker-compose are you running? What is the exact error you're getting? I added in these options so that the log files don't grow indefinitely on Raspberry Pis. I can create a option to turn them off if it is causing errors.

weissm, 2020-09-29:
Thank you very much for getting back. Here is the versions I am running.
- docker-compose version 1.26.0, build unknown
- Docker version 19.03.13, build 4484c46
I guess the issue in my set up is the use journald logging. The solution you are proposing should solve it.

Slyke, 2020-10-06:
Awesome. Everything looks up to date. I'll add in an option to turn logging on and off, but it'll be after `experimental` has been merged into master, since it's a giant PR as it is.

weissm, 2020-10-06:
Understood – thanks!