SensorsIot/IOTstack #126: Addition of Wireguard causes a mess during a git pull

Paraphraser, 2020-09-24:
Just did this on straight clone from SensorsIot/IOTstack (ie no futzing around with forks):

```
$ cd ~/IOTstack
$ git pull
```

The result:

```
remote: Enumerating objects: 215, done.
remote: Counting objects: 100% (215/215), done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 356 (delta 155), reused 205 (delta 151), pack-reused 141
Receiving objects: 100% (356/356), 395.76 KiB | 619.00 KiB/s, done.
Resolving deltas: 100% (172/172), completed with 78 local objects.
From https://github.com/SensorsIot/IOTstack
   7a1fba9..6ca1b03  master       -> origin/master
   2f1c3cc..459f59a  experimental -> origin/experimental
 + fa60402...e760c66 gh-pages     -> origin/gh-pages  (forced update)
 * [new branch]      pr/102       -> origin/pr/102
   8b28a81..26d4d56  pr/152       -> origin/pr/152
 * [new branch]      pr/19        -> origin/pr/19
 * [new branch]      pr/93        -> origin/pr/93
Updating 7a1fba9..6ca1b03
fatal: cannot create directory at 'volumes/wireguard': Permission denied
$
```

The logical cause is:

[image]

As I understand it, the wireguard bits should be in `~/IOTstack/.template`, not in the main tree.

This **probably** works for an initial clone because `~/IOTstack./volumes` doesn't get changed to root ownership until the first "up" and, for most cases, stuff doesn't get touched later.

I haven't done a what-if to figure out the situation where IOTstack gets broken but it causes a right mess.

I don't know enough about git to recommend how best to undo the change in the repo. I hope you do! I think it needs fairly urgent attention too.

quadratrund, 2020-09-24:
"Permission denied" in this context usually means you really have no permission. You should get the same error, if you try to create the directory manually with mkdir. Did you try to take ownership for the whole repository dir?

Paraphraser, 2020-09-24:
Hi, no that isn't the problem.

The way IOTstack is meant to work is that the changes that made their way into `services/wireguard` and `volumes/wireguard` a few hours ago should actually be in `.templates/wireguard`.

When you run the IOTstack menu and select a container you want to install, that copies the necessary bits into place, typically:

* services/«container name» and 
* docker-compose.yml

The volumes definitions for the container in `docker-compose.yml` describe how `volumes/«container name»` should be set up, and docker-compose has the appropriate privileges to do that.

What's going on here is bypassing that mechanism. It really needs to be undone properly.

Paraphraser, 2020-09-24:
The way I found to recover from this problem **and** still benefit from the latest changes in the repo, is:

```
$ cd ~/IOTstack
$ git reset --hard
$ git status 
```

If your system behaves like mine, you will get a bunch of untracked files:

```
	.templates/domoticz/
	.templates/portainer_agent/
	.templates/prometheus/
	.templates/qbittorrent/
	.templates/transmission/build.sh.pr
	.templates/transmission/service.yml.pr
	.templates/transmission/volumes.yml.pr
	docs/Containers/
```

What you need to do is sort that list into two groups:

* things you recognise as having created yourself; and
* things you don't recognise.

In my case, the things I didn't recognise were:

```
	.templates/domoticz/
	.templates/portainer_agent/
	.templates/prometheus/
	.templates/qbittorrent/
	docs/Containers/
```

Those you should remove by hand. For example:

```
$ cd .templates
$ rm -rf domoticz portainer_agent prometheus qbittorrent
$ cd ..
$ rm -rf docs/Containers/
```

Take your stack down:

```
$ docker-compose down
```

Change the permissions on the volumes directory:

```
$ sudo chown pi:pi volumes
```

Now you can run the git update again:

```
$ git pull origin master
```

Then, put the permissions back the way they were:

```
$ sudo chown root:root volumes
```

and bring up the stack:

```
$ docker-compose up -d
```

This still leaves behind:

```
~/IOTstack/volumes/wireguard  
~/IOTstack/services/wireguard/
```

but both have pi:pi ownership and **should** disappear when the patch is applied to the repo.

> Your mileage may vary if you are actually running wireguard

Slyke, 2020-09-24:
Submitted a PR to fix this: https://github.com/SensorsIot/IOTstack/pull/127 Not able to test it on an RPi right now.

closed 2020-09-25 by Paraphraser