SensorsIot/IOTstack #132: Portainer/Portatiner is now deprecated

Airfoil73, 2020-09-29:
Portainer/Portatiner is now deprecated. Can you update it so it is using **portainer/portainer-ce**.  [https://hub.docker.com/r/portainer/portainer-ce](url)

Paraphraser, 2020-09-29:
@Slyke I volunteer to submit a PR for this.

Paraphraser, 2020-09-30:
@Slyke I'll start by declaring that Portainer is something I could easily live without. To be perfectly honest, I struggle to see the point of it. If I were to proceed strictly from my own limited perspective, I'd be totally gung-ho. I've implemented -ce on my RPis and it seems to work out-of-the-box so I'd have no trouble creating a PR to change the template. It's the classic one-line-change.

But...

The Portainer [release doco](https://github.com/portainer/portainer/releases/tag/2.0.0) has got more bold-faced notes and warnings than a kid with chickenpox. Just because **I** don't get the point of Portainer doesn't mean there aren't others who depend on it and whose IOTstack installations may break if -ce is foisted on them and it exposes a dependency on one of the said notes and warnings.

I realise that such a user would have to "pull full service from template" to fall into a hole but, on balance, I think a little more caution might be warranted.

Several possible approaches spring to mind:

1. Do nothing until a -ce release with fewer warnings. The basic problem with this approach is that the "extensions" issue is never going to go away:

	> Extensions have now been removed; there is now no ability to use RBAC, Registry Manager, or External Authentication extensions in CE 2.0 (Extension customers will be communicated directly with a free license for the upcoming Portainer Business Edition). **EXTENSION USERS, DO NOT UPGRADE TO PORTAINER CE 2.0**

2. Change the template `service.yml` so that the user can pick & choose for the price of a trip through a text editor:

	```
	  portainer:
	    container_name: portainer
	    # enable ONE of the following
	    image: portainer/portainer
	    #image: portainer/portainer-ce
	    restart: unless-stopped
	    ports:
	      - "9000:9000"
	    volumes:
	      - /var/run/docker.sock:/var/run/docker.sock
	      - ./volumes/portainer/data:/data
	```

	This actually works quite well, to the point of being able to switch back and forth in the space of a few seconds (once both images have been pulled). The two versions don't seem to care about sharing the same structures in `volumes` but that might have more to do with my unsophisticated use of the app. Someone with lots of add-ons may find things broken by a round-trip.
	
3. Duplicate "portainer" as "portainer-ce", with `docker-compose.yml` potentially looking something like:

	```
	  portainer:
	    container_name: portainer
	    image: portainer/portainer
	    restart: unless-stopped
	    ports:
	      - "9000:9000"
	    volumes:
	      - /var/run/docker.sock:/var/run/docker.sock
	      - ./volumes/portainer/data:/data
	
	  portainer-ce:
	    container_name: portainer-ce
	    image: portainer/portainer-ce
	    restart: unless-stopped
	    ports:
	      - "9002:9000"
	    volumes:
	      - /var/run/docker.sock:/var/run/docker.sock
	      - ./volumes/portainer-ce/data:/data
	```

	This also seems to work quite well, although I did get an "Unable to retrieve endpoints" error at one point, plus Safari stopped being able to repaint the 1.24.1 screen correctly. I could not replicate either in Chrome and both problems went away after I voided Safari's cache. I'm assuming Safari got confused by the way I kept moving the ground beneath its feet.
	
Options 2 & 3 give each user the choice. Option 2 keeps a single port and single data store at the price of a docker-compose.yml edit and an "up". Option 3 forces the user to deal with a second port (at least during the migration period) but does allow side-by-side comparison of old vs new for any sophisticated users.

On balance, I'm leaning towards #3. Thoughts?

Slyke, 2020-09-30:
Hmmm... I'd like to go with option 1, but worried about the warnings that come up. We'll give them a few releases and see how they are going.

Option 2 might work, but there might be breaking breaking changes (or non-backwards compatible changes) to Portainer-CE, and switching back to Portainer 1.24.x might cause it to fail and possibly destroy saved data (worst case scenario).

For now, I think the best choice is to do option 3. I will code both of them in. On the new menu I will add in a warning on the menu when selecting the deprecated Portainer that it is deprecated and could be removed from IOTstack at any time.

jmorris644, 2020-09-30:
I vote for option 3. I installed -ce on a test system and so far no issues. Having to use a different port is a very minor nuisance

877dev, 2020-10-01:
Number 3 here as well. 

I use Portainer all the time, it would be nice to have CE version so I can check it out without fully committing.

Paraphraser, 2020-10-03:
In testing, I noted the presence of port 8000 for Portainer-ce, which is described as:

> Port 8000 is a SSH tunnel server and is used to create a secure tunnel between the agent and the Portainer instance. 

The updated definition will need to be:

```
  portainer-ce:
    container_name: portainer-ce
    image: portainer/portainer-ce
    restart: unless-stopped
    ports:
      - "8000:8000"
      - "9002:9000"
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
      - ./volumes/portainer-ce/data:/data
```

Slyke, 2020-10-06:
Okay, I added deprecation warning on the new menu when selecting portainer, and also added the `portainer-ce` service directly below it.

Paraphraser, 2020-10-06:
For anyone running on the master branch, Portainer CE is now listed in the menu. You will find [instructions here](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Portainer-ce.md)

jmorris644, 2020-10-06:
> For anyone running on the master branch, Portainer CE is now listed in the menu. You will find [instructions here](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Portainer-ce.md)

When I ran menu.sh it stated that it updated but Portainer-ce is not in the Build list.

Paraphraser, 2020-10-06:
Possibly just try again. It might be a chicken and egg. Running the menu does a git pull but the menu probably isn't smart enough to realise it has changed and restart itself. To be absolutely sure you could do a `git status` and make sure it says "on branch master" (if not, `git checkout master`) then a `git pull`.

If it still doesn't work, please check `~IOTstack/.templates`. If you see `portainer-ce` then it indicates the template came down to your machine, so any problem will likely be in the menu.sh. If the `git status` suggests menu.sh has changed then the reason might need to be resolved first.

I have an RPI3B+ which I keep as a pristine straight clone from IOTstack. I tested on that and it just worked but, thinking back, I probably did a `git pull`, then checked .templates, then did a `grep "Portainer-ce" menu.sh`, all to satisfy myself that everything had come down - then I ran the menu and selected it.

jmorris644, 2020-10-06:
hmmm....

`Git status` reports origin/master
so I did a `git checkout master` and it stated I was already on master and that master was up to date.

I checked the .templates and no Portainer-ce present.

Paraphraser, 2020-10-06:
Weird.

[image]

The above rolls back to before the changes. The "grep" and "ls" confirming the CE bits are not there. Then a `git pull origin master` where you can see the stuff arriving in the patches, confirmed with another "grep" and "ls". No idea why it is not working for you.

AntumArk, 2020-10-06:
I have tried installing it into Ubuntu VM. 
[image]
[image]
Everything went fine. Followed instructions on readme.

Bettelmage, 2020-10-06:
I set up a new Pi last night. After i saw the news about the commits today, I tried to update as well.

I didn't work for me either. Project up to date. But I found out what's the problem: 
Wrong remote. Last night I just copied the git clone command from the getting started page without having a closer look.
It is referencing to the gcgarner stack. I got the wrong stack..... hence no update.

I guess the document should be fixed.

@jmorris644 maybe you are having the same issue.

jmorris644, 2020-10-06:
> I set up a new Pi last night. After i saw the news about the commits today, I tried to update as well.
>
> I didn't work for me either. Project up to date. But I found out what's the problem:
> Wrong [...]

YEs, that is my issue. So I tried to change the remote but am getting a conflict error for menu.sh. 

```
pi@raspberrypi:~/IOTstack $ git pull
Auto-merging menu.sh
CONFLICT (content): Merge conflict in menu.sh
Automatic merge failed; fix conflicts and then commit the result.
```

I really do not know git all that well. How do I resolve it?

877dev, 2020-10-06:
I think you can `git stash` to save your differences or `git reset —hard` to lose all changes. 

https://docs.gitlab.com/ee/topics/git/numerous_undo_possibilities_in_git/#unstaged-local-changes-before-you-commit

Slyke, 2020-10-06:
It's weird people are getting git issues. If you are experiencing them, can you run the following commands and place them here?

```
git status
git log
git stash list
git checkout master
git remote get-url origin
git pull
```
You can `git stash` if you have made your own changes.

jmorris644, 2020-10-06:
<pi@raspberrypi:~/IOTstack $ git diff
diff --cc menu.sh
index 4a481a4,3f7eeea..0000000
--- a/menu.sh
+++ b/menu.sh
@@@ -3,11 -3,23 +3,29 @@@
  #get path of menu correct
  pushd ~/IOTstack

++<<<<<<< HEAD
 +#This is the Display name for the menu
 +#structure : [CONTAINER]=MENU Display Text
 +#One entry per line to simplify PRs
++=======
+ CURRENT_BRANCH=${1:-$(git name-rev --name-only HEAD)}
+
+ # Consts/vars
+ TMP_DOCKER_COMPOSE_YML=./.tmp/docker-compose.tmp.yml
+ DOCKER_COMPOSE_YML=./docker-compose.yml
+ DOCKER_COMPOSE_OVERRIDE_YML=./compose-override.yml
+
+ # Minimum Software Versions
+ COMPOSE_VERSION="3.6"
+ REQ_DOCKER_VERSION=18.2.0
+ REQ_PYTHON_VERSION=3.6.9
+ REQ_PYYAML_VERSION=5.3.1
+
++>>>>>>> 7f3d143e97569b9a99fb935251f391bd4a4ee0c2
  declare -A cont_array=(
        [portainer]="Portainer"
+       [portainer-ce]="Portainer-ce"
+       [portainer_agent]="Portainer agent"
        [nodered]="Node-RED"
        [influxdb]="InfluxDB"
        [telegraf]="Telegraf (Requires InfluxDB and Mosquitto)"
@@@ -31,42 -47,51 +53,87 @@@
        [diyhue]="diyHue"
        [homebridge]="Homebridge"
        [python]="Python 3"
+       [gitea]="Gitea"
        [qbittorrent]="qbittorrent"
++<<<<<<< HEAD
 +      [zigbee2mqttassistant]="zigbee2mqttassistant"
 +)
 +
 +#The convension for CONTAINER is that it is the name of the .templates/CONTAINER directory and as the key below for the relevant arch
 +
 +# keys for CONTAINER
 +# One per line to simply PR
 +declare -a armhf_keys=(
 +      "portainer"
 +      "nodered"
 +      "influxdb"
 +      "grafana"
 +      "mosquitto"
 +      "telegraf"
 +      "mariadb"
 +      "postgres"
 +      "adminer"
 +      "openhab"
 +      "zigbee2mqtt"
 +      "pihole"
 +      "plex"
 +      "tasmoadmin"
 +      "rtl_433"
 +      "espruinohub"
 +      "motioneye"
 +      "webthings_gateway"
 +      "blynk_server"
 +      "nextcloud"
 +      "diyhue"
 +      "homebridge"
 +      "python"
 +      "zigbee2mqttassistant"
 +      "qbittorrent"
++=======
+       [domoticz]="Domoticz"
+       [dozzle]="Dozzle"
+       [wireguard]="Wireguard"
+       # add yours here
++>>>>>>> 7f3d143e97569b9a99fb935251f391bd4a4ee0c2
  )

+ declare -a armhf_keys=(
+       "portainer"
+       "portainer-ce"
+       "portainer_agent"
+       "nodered"
+       "influxdb"
+       "grafana"
+       "mosquitto"
+       "telegraf"
+       "prometheus"
+       "mariadb"
+       "postgres"
+       "timescaledb"
+       "transmission"
+       "adminer"
+       "openhab"
+       "zigbee2mqtt"
+       "deconz"
+       "pihole"
+       "plex"
+       "tasmoadmin"
+       "rtl_433"
+       "espruinohub"
+       "motioneye"
+       "webthings_gateway"
+       "blynk_server"
+       "nextcloud"
+       "diyhue"
+       "homebridge"
+       "python"
+       "gitea"
+       "qbittorrent"
+       "domoticz"
+       "dozzle"
+       "wireguard"
+       # add yours here
+ )
  sys_arch=$(uname -m)

  #timezones
>

This is after I changed the git .config from <gcgarner> to <SensorsIot>

jmorris644, 2020-10-06:
Also, the 

`git remote get-url origin`

`git reset --hard`

`git pull`

still gave me

```
Auto-merging menu.sh
CONFLICT (content): Merge conflict in menu.sh
Automatic merge failed; fix conflicts and then commit the result.
```

Slyke, 2020-10-06:
Did you merge anything into your local master?

Try this:
```
git checkout master
git fetch
git reset origin/master --hard
```
This should blow away everything on master and reset it to master as is on the remote branch. Though, I am a little concerned that the `git remote get-url origin` command gave you a merge conflict error. The `get-url` command is to just let me know which repo origin is set to.

jmorris644, 2020-10-06:
> Did you merge anything into your local master?
> 
> Try this:
> 
> ```
> git checkout master
> git fetch
> git reset origin/master --hard
> ```
> 
> This should blow away everything on local master and reset it to master as is on the remote branch. Though, I am a little concerned that the `git remote get-url origin` command gave you a merge conflict error. The `get-url` command is to just let me know which repo origin is set to.

The `git checkout master` errored out with the conflict error. Running the `git reset origin/master --hard` resolved the issue.

I think it was all caused by the original reference to `gcgarner` remote

thanks

Bettelmage, 2020-10-06:
I guess the problem with git pull is that the gcgarner stack is also some commits ahead, leading to merges.

git reset origin/master --hard solved it for me as well.

Should I open an issue for updating the documentation, preventing other people from accidentally cloning the wrong stack?

Paraphraser, 2020-10-06:
Initially I thought everyone was talking about the main readme and I polished my eyeballs several times trying to convince myself that it made no mention of gcgarner.

But a search of the docs folder finds some hits:

```
$ cd ~/IOTstack/docs
$ grep -iR "garner" * 
Backups.md:Make sure you are running the latest version of the project [link](https://github.com/gcgarner/IOTstack/wiki/Updating-the-Project).
Containers/EspruinoHub.md:please let me know about your success or issues [here](https://github.com/gcgarner/IOTstack/issues/84)
Getting-Started.md:git clone https://github.com/gcgarner/IOTstack.git ~/IOTstack
Getting-Started.md:Have a look a the wiki on how to share files between Node-RED and the Pi. [Wiki](https://github.com/gcgarner/IOTstack/wiki/Node-RED#sharing-files-between-node-red-and-the-host)
$
```

Is the first "Getting-Started.md" hit the one everyone is talking about? If there are other cases then I'd really appreciate some URLs pointing me to each one.

Bettelmage, 2020-10-07:
Correct, we are talking about the getting started document.

Are the wiki links you found still correct?

Paraphraser, 2020-10-07:
I've been working on doco updates. Watch this space... 🤓

AntumArk, 2020-10-07:
I used `git clone https://github.com/SensorsIot/IOTstack.git ~/IOTstack` and it was fine to me.

Paraphraser, 2020-10-07:
@Bettelmage [PR 140](https://github.com/SensorsIot/IOTstack/pull/140) created. I'm hoping this will do the trick without causing @Slyke to have too many conniptions about all my unilateral decisions as to what belonged where in doco space. 😱

PR #567 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3
Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
commit 3239e9b4 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3

Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
closed 2022-06-12 by Slyke