SensorsIot/IOTstack #133: Grafana image renderer in IOTstack

Roberto6969, 2020-10-02:
I would like to install grafana-image-renderer plugin into Grafana, running on IOTstack. Grafana guys recommends setting up another Docker container for rendering and using remote rendering. But how, if Grafana runs in context of IOTstack?
Should I modify docker-compose.yml or service.yml? I have tried but I had no luck.

I believe this is not an issue but behaviour of IOTstack - excuse me publishing question here.

Roberto6969, 2020-11-02:
Actually, the root of this problem is common of all programs in IoT stack. Grafana is just an example. If I use Grafana in docker, there is some quite different ways to implement Grafana in docker environment or at least to implement changes (GF variables, just for example) in docker environment.
But, IoT stack is another step toward more difficult way to implement some changes to default configuration, which is usable only for initial test.
I would like to know, what is the best practice in such situation. So far, as i can see, if you need some changes or add-ons and you have to deal with data and not with IT infrastructure problems, it is better to stay in pure environments.
However, I still believe IoTstack  idea is extremely useful, time-saver and simple - but as long as you don't need any modifications or - god forbid - add-on in Grafana as grafana-image-renderer is.

Again, this is not an IoTstack issue, but request for opinion. Than You for any opinion.

Paraphraser, 2020-11-02:
If you have been wondering why nobody has replied after all this time, it's possible that other people had the same reaction to your original post as I did: I did not understand either the problem you were trying to solve or your question. I assumed that was due to my own lack of knowledge and hoped someone with more knowledge would step in and help.

If this "renderer" you are talking about can communicate with Grafana when both Grafana and the renderer are installed natively then, in principle, there should be no difficulty doing the same if both are installed as containers. In the same way as a NodeRed flow will subscribe to the internal hostname of "mosquitto" and send database inserts to InfluxDB using the "influx" hostname (rather than any form of loopback address), you'd probably need a few tweaks but that should be about it.

I think you will probably get a lot more help if you give a concrete example including things like the additions you have made to docker-compose.yml to start the renderer. A concrete example gives people both an understanding of what you are trying to achieve and where you are up to, as well as something they can take and try out on their own systems to see if they can make headway.

Also, posting logs and error messages often helps others have "ahah" moments.

I'm willing to give anything a try but you really do need to put a few more cards on the table so I have some chance of following in your footsteps.

Paraphraser, 2022-06-19:
No further response. Should probably be considered stale.

closed 2022-09-11 by Slyke