SensorsIot/IOTstack #134: homebridge ui port conflicts with influxdb

benpoe, 2020-10-04:
homebridge ui port is 8080. this conflicts with influxd'sd http server, which causes influxd to crash.
either change homebridge port or add a port binding for influxd to solve this.

Paraphraser, 2020-10-04:
I'm confused:

[image]

I don't run homebridge but I do run influx. Red highlight seems to be where homebridge gets its port number so I agree it's 8080. Yellow highlight is port mappings for running container and I don't see 8080 anywhere. Running `docker inspect influxdb | grep 8080` produces silence so I don't see how the influx container is even aware of 8080. I think you might need to track down where 8080 is coming from because, while I could easily be wrong, my guess at this point is that it's something peculiar to your system.

Slyke, 2020-10-06:
This will be fixed on the rewritten menu. For now, you can either use an [Compsoe-Override](https://sensorsiot.github.io/IOTstack/Custom/) to merge in your own port changes, or modify either of the service's templates `service.yml` to fix it.

benpoe, 2020-10-15:
> I'm confused:
>
> [image]
>
> I don't run homebridge but I do run influx. Red highlight seems to be where homebridge gets its port number so I agree it's 8080. Yellow highlight is port mappings for [...]

i was confused too, just saw influx crashing all the time and logging says that it's crashing because 8080 is already used.
using netstat it did state that home assistant indeed grabbed it. i've since then bond another port to 8080 for influx and it's working since.

Paraphraser, 2022-06-19:
Now uses 8581 so I think this is fixed.

closed 2022-09-11 by Slyke