SensorsIot/IOTstack #135: Mosquitto: Can't find mosquitto_passwd command

TBjorseth, 2020-10-05:
I have just installed IOTStack with mosquitto as one of the containers, but I can't seem to find the mosquitto_passwd command so I can create a password file with an encrypted password. The only two instances of "mosquitto_passwd" I have are these two, and I can't run them:

```
/var/lib/docker/overlay2/0de50f18bcd5b2547884503de4cda84f53a33dbf93e2aca8378fc30852d5f6c7/merged/usr/bin/mosquitto_passwd
/var/lib/docker/overlay2/e48875223484e1acef81970315d16dd0518affab84f56832070748caef1a9f9b/diff/usr/bin/mosquitto_passwd
```

The only executable commands I have that starts with mosquitto_* are 

```
/usr/bin/mosquitto_sub
/usr/bin/mosquitto_pub
```

So where can I find mosquitto_passwd?

Paraphraser, 2020-10-05:
Hi,

The short answer is that `mosquitto_passwd` is **inside** the mosquitto container.

Please see [the IOTstack documentation](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Mosquitto.md#security).

As long as you pay attention to what you need to do **inside** vs **outside** the container, you should be fine.

> Unlike many containers, Mosquitto does not run as root so you don't get the usual distinction between a `$` prompt implying **outside** the container, and a `#` implying **inside** the container.

See also [PR73](https://github.com/SensorsIot/IOTstack/pull/73) which has been languishing for a long time and for reasons that I can't explain. All it does is remove some `echo` statements from the supplied `terminal.sh` script for mosquitto that are:

* erroneous - wrong path on the `mosquitto_passwd` command; and
* incomplete - don't tell you what else you have to do,

in favour of a strong hint to go read the same doco as I've linked to above.

Slyke, 2020-10-06:
I'll add in a link to the mosquitto docs inside the echo @Paraphraser on the `experimental` branch. I've merged in the current PR. #73

closed 2020-10-06 by Slyke