SensorsIot/IOTstack #138: Portainer: Published ports not working (IP: 0.0.0.0:port)

ranseyer, 2020-10-05:
[image]
Tested with raspberry OS/32 (Standard an Light Edition)
The Link goes to 0.0.0.0:port instead of the Hostname or Host-IP

Fix:

A) Edit /etc/systemd/system/multi-user.target.wants/docker.service
```
#Old: ExecStart=/usr/bin/dockerd -H  fd:// --containerd=/run/containerd/containerd.sock
#New: Insert --ip
ExecStart=/usr/bin/dockerd -H  fd:// --ip=192.168.1.162 --containerd=/run/containerd/containerd.sock

```
B) Edit Docker-Compose.yml in each section:
Example pihole:

```
  pihole:
    **network_mode: default**
```

ranseyer, 2020-10-05:
..maybe somone has an better idea instead of editing /etc/systemd/system/multi-user.target.wants/docker.service ...

Paraphraser, 2020-10-05:
Perhaps see [the docs](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Portainer.md#setup-public-ip)

Personally, I see that Portainer display as "for information only", in the same way as `docker ps` Just gives you a list of port mappings. Maybe it's just me (old guy - CLI for brains) but even with the public IP set, I honestly don't see the point of clicking on a port.

Slyke, 2020-10-06:
Since this issue is solved, I'll close it out! Looks like @ranseyer wasn't hitting the WUI port. Please let me know if that's not the case and I'll reopen.

ranseyer, 2020-10-06:
> Since this issue is solved, I'll close it out! Looks like @ranseyer wasn't hitting the WUI port. Please let me know if that's not the case and I'll reopen.

No, this is not the case... 

You can try it. When you open Portainer in the browser of yor PC. Then you get clickable links as shown in the Screenshot.
The Link of Deconz should point in my case to http://192.168.1.162:8090 but per default it points to 0.0.0.0:8090 .
This is clearly a bug. Even when some people dont see a need for workings links. 

Its absolutely clear, that a lot of this links will not help when you try to click them and for example you want to access a mqtt port vie http* will not work. Shure. But a lot of services offer an http access too. 

There are three ways to find the ports: 
-((use the brain))
-take a look in the docker-compose.yml (ssh needed, and in that case i see no sense for portainer)
-take a look in portainer (and thinking whith ports can bei usable for what)
-click the meaningful links direct in portainer, this i the reason why you can not oly read the ports, but you can click them...

Slyke, 2020-10-06:
I think the bug is in docker compose itself: https://github.com/docker/compose/issues/2999

What happens if you add
```
network_mode: default
```
To the portainer service, and down and up the stack?

ranseyer, 2020-10-06:
>  think the bug is in docker compose itself: docker/compose#2999
... is it possible that you read my text above ?
=> As i have written: This ist 50% of the solution ... It should be easy to add this line for all kind of services/containers...

But this helps only when you change a script provided by the docker package. This part of the solution i really dont like...
-Hard Coded IP-Adress
-Changing the content of an file provided by an package (its not an config file!!)

From top in this Issue:
> Edit /etc/systemd/system/multi-user.target.wants/docker.service
> 
> #Old: ExecStart=/usr/bin/dockerd -H  fd:// --containerd=/run/containerd/containerd.sock
> #New: Insert --ip
> ExecStart=/usr/bin/dockerd -H  fd:// --ip=192.168.1.162 --containerd=/run/containerd/containerd.sock

Paraphraser, 2020-10-06:
From your original question,  my understanding of your issue is that you want to be able to click on something in the published ports list (which you circled) and have that come up with an IP address of **other** than 0.0.0.0 and, for preference, the IP address of the host being monitored by Portainer.

Accurate summation? If not, please elaborate.

If that summation is accurate then the link I provided earlier should have solved your problem. But, to explain it here:

[image]

Login to Portainer then (1) click "Endpoints" (2) click "local".

[image]

(3) enter the IP address or domain name of the system being monitored. My RPi has the name "iot-hub" and I used the multicast DNS form but I could also have used its fully qualified domain name. (4) click "Update endpoint".

> It is perhaps also relevant to point out that Portainer does not necessarily have to be running on the same host as the services it is monitoring, hence the need to tell it, rather than expect it to magically acquire the information from Docker.

The next screen shot is from my iPad where I tapped-and-held on port 1880 for NodeRed. The pop up (5) clearly shows **not** 0.0.0.0.

[image]

Completing the tap takes me to the NodeRed GUI so it definitely works.

If that does not answer your question then I am honestly totally mystified as to what you are actually trying to achieve.

<hr>

I should probably explain why I do not use this approach myself. It's because I have a "live" and "test" RPi4 which I keep in parallel. Periodically, I run a backup on the live Pi, which I restore to the test Pi so that I have reasonably recent data. If I embed things like the name "iot-hub.local" on the live machine the restored Portainer config **will** work on the test machine but it will point me to the live machine rather than the test machine. Were I to get into the habit of relying on Portainer links, I could easily wind up forgetting which system I was talking to, think I was working on the test machine, do something dumb, and then have a serious mess to sort out. I see 0.0.0.0 as a fail safe.

ranseyer, 2020-10-06:
> From your original question, my understanding of your issue is that you want to be able to click on something in the published ports list (which you circled) and have that come up with an IP address [...]

Yes, and you are right. I think i have overseen the hostname part...

My Solutions is theoretically working too. 
-i like Part 1 of my solution better than your way (scriptable)
-Part2 (Changing the Systemd script for Docker) is a nogo in my eyes

So the only complete solution is yours.

Sorry for my (maybe unnesessary) question, and thank you very much for your patience !

closed 2020-10-06 by ranseyer