SensorsIot/IOTstack #142: Error in docker-compose with telegraf config entry

GarryG, 2020-10-07:
Hi,

I tried out IOTstack (after seeing it referenced in Andreas Spiess' video) and it's great ... except for one small problem - I keep getting errors when I try to use telegraf ... (n.b.: I had used the old install routine originally, but have now moved over to this repo according to the doc on how to change over - same error before and after)

```
root@raspi4:~/IOTstack# docker-compose up -d
ERROR: yaml.parser.ParserError: while parsing a block mapping
  in "./docker-compose.yml", line 68, column 5
expected <block end>, but found '<block sequence start>'
  in "./docker-compose.yml", line 75, column 7

```
Line 68 is the first "container_name: telegraf" line right after the "telegraf:" entry. If I move the whole block to somewhere else in the file, the error line moves; removing it lets docker-compose run through successfully, so it's definitely something in the telegraf block.

```
  telegraf:
    container_name: telegraf
    image: telegraf
    volumes:
      - ./services/telegraf/telegraf.conf:/etc/telegraf/telegraf.conf:ro
    depends_on:
          - cadvisor-arm
          - node-exporter
      - influxdb
      - mosquitto
```

The yaml file is used right out of the generation by menu.sh, nothing was originally edited or changed ...

Sorry, not really learned about yaml files, I reckon it's probably something minor ...

Paraphraser, 2020-10-07:
Column alignment is important in yaml files. This isn't your fault, of course. There's some strangeness to the way this particular service.yml is constructed which will probably go away with the "new menu".

The fix here is to shift `- cadvisor-arm` and `- node-exporter` left by 4 spaces so that they line up with `- influxdb` and `- mosquitto`. The other rule to know about yaml files is "spaces good, tabs deadly".

What you should do is make this same change in both `~/IOTstack/services/telegraf/service.yml` and `~/IOTstack/docker-compose.yml`, and then make a mental note to never do "pull full service from template" for telegraf in the menu. The basic way things work is life begins in the hidden folder `~/IOTstack/.templates`. Selecting a service in the menu copies the template into the services directory. The docker-compose file is built by sweeping up all the service.yml files. Telegraf is a bit special because its service.yml is futzed with between templates and services so there are no actual guarantees that this problem won't recur on any visit to the menu but you'll know how to fix it.

Probably the other thing I'd do is (a) fix service.yml, then (b) make a copy of the fixed file as something like service.yml.fixed. If the problem ever recurs, you'll have a known-good reference that you can use at 3am without risk of error. 😴

GarryG, 2020-10-08:
Yeah, I was aware of the space-formated config ... that's why I thought the two lines indented further were there for a reason ... ;/ also with the first error pointing to the beginning of the block I didn't think they were the reason for that error ... corrected, docker-compose now runs without an error ... thanks!

GarryG, 2020-10-08:
P.S. - can't change in service.yml, as the two lines aren't in there ... menu.sh probably adds them when you have cadvisor and node-exporter active in the stack ...

GarryG, 2020-10-08:
Did some more checking - the only other place (at least in my current setup) the two services are listed as "depends_on" is prometheus, which incidently does not have any other dependencies. There, the added spaces do not matter. For a test, I changed the telegraf indentation to 4 additional space for all entries, which also works, so I guess as long as the lines are all indented equally, everything is fine. I changed my service.yml to add those spaces, so I should be fixed for any further menu.sh updates for now ...

Checked menu.sh for a source of the problem, but couldn't find where the additional lines are created ...

(btw, typo "$sevice_overwrite" in function yml_builder ;) used everywhere there, so just esthetics I guess)

Paraphraser, 2020-10-08:
See

```
/home/pi/IOTstack/.templates/prometheus/build.sh
```

That's where they come from. But you'll have to educate me as to what Prometheus actually is and why the heck it futzes with Telegraf because I have absolutely no idea. Truth to tell, I don't know what Telegraf is either. I'm happy with my Mosquitto, NodeRed, Influx and Grafana. 🤔

iliavivanov, 2020-12-03:
Workaround
`sed -i 's/    \\/\\/g' ~/IOTstack/.templates/prometheus/build.sh`

Paraphraser, 2022-06-19:
There is no build.sh for Prometheus (removed a while ago) so I think this one can probably be treated as fixed.

closed 2022-09-11 by Slyke