SensorsIot/IOTstack #143: Duplicate portainer created

Londoneye02, 2020-10-07:
Hi all.

I burned a new SD card and cloned the latest status of the project. I was able to install everything without issue.

Then I wanted to create a compose-override.yml to change the portainer-ce port from 9002 to the usual 9000...

Weirdly somehow it generated entries for both portainer and portainer-ce. Then I had an error because the ports are duplicated...

Now everytime I run the menu to build the stack, both portainer and portainer-ce appear selected (even after deleting portainer from the docker-compose file, and not being present on the compose-override file). I unselect portainer but the next time is selected again

It is not a big deal, you just have to remember it, but maybe there is some kind of error in the code...

Thank you

Paraphraser, 2020-10-07:
Did you try the manual approach in [ceasing use of Portainer](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Portainer-ce.md#ceasing-use-of-portainer) ? If not, give it a whirl & report back.

Londoneye02, 2020-10-07:
Many thanks for the quick response!!!

The weird thing is that, as I mentioned, this is a fresh intall, and I hadn't selected "clasic" portainer in the first place...

I tried what you mentioned on your post but the issue remains the same. In the image you can see that the selection.txt does not have the portainer line (I also removed the services and volumes folder for portainer acordingly), but when you run the menu, it's there 

[image]

Paraphraser, 2020-10-08:
Spoiler: my guess is a partial-match bug in the menu system.

<hr>

Trying to replicate.

Starting position is all images removed and no `~/IOTstack` folder:

```
$ docker images
REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
$ ls -ld ~/IOTstack
ls: cannot access '/home/pi/IOTstack': No such file or directory
```

Fresh clone:

```
$ git clone https://github.com/SensorsIot/IOTstack.git IOTstack
Cloning into 'IOTstack'...
remote: Enumerating objects: 36, done.
remote: Counting objects: 100% (36/36), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 3151 (delta 14), reused 19 (delta 4), pack-reused 3115
Receiving objects: 100% (3151/3151), 1.08 MiB | 309.00 KiB/s, done.
Resolving deltas: 100% (1863/1863), done.
```

Run the menu.

```
$ cd IOTstack/
$ ./menu.sh 
~/IOTstack ~/IOTstack
checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
checking docker version
Docker version >= 18.2.0. You are good to go.
```

This is the build-stack menu as-presented, save that I moved the cursor down to Telegraf to avoid any visual clutter on the first lines:

[image]

My standard selections are:

* Portainer-ce
* Mosquitto
* Node-Red
* InfluxDB
* Grafana
* PiHole
* Gitea

For Node-Red, I accept the default list of nodes and only add SQLite.

```
Adding portainer-ce container
...pulled full portainer-ce from template
...Running directoryfix.sh on portainer-ce
Adding nodered container
...pulled full nodered from template
Adding influxdb container
...pulled full influxdb from template
Adding grafana container
...pulled full grafana from template
Adding mosquitto container
...pulled full mosquitto from template
[... 224 lines cut ...]

So, out of the box with a clean clone, I'm at a loss to explain why you get (old) Portainer.

What I did not do was:

1. Re-image Raspbian from the ground up.
2. Re-install Docker and Docker-Compose from the menu.

I can't see why that would make any difference and, anyway, doing that would take more time than I'm prepared to devote to the problem.

<hr>

On a hunch, I went back into the menu. I was **intending** to try installing "Portainer agent", on the off-chance that that somehow auto-included (old) Portainer.

But guess what I saw when I chose "Build stack"?

[image]

The menu has auto-selected (old) Portainer. I have no idea why. I have always assumed these auto-selections were based on `selection.txt` so maybe there's some kind of partial string matching involved which is getting it wrong.

> As an aside, assuming my assumption (recursion!) about the purpose of `selection.txt` is correct, I don't understand why that file was ever needed. An `ls -1 ./services` produces the same data, dynamically, and that seems a bit more robust than a static file that could easily get out of sync. Not that that would correct the hypothesised partial matching problem...

@Slyke keeps promising us a new menu system and I live in hope that it deals with problems like this. With that in mind, I don't see that it's worth drilling into why (old) Portainer behaves like a zombie. The obvious workaround is to pay attention to what's in the list each time the menu is run, and clobber (old) Portainer each time it sticks its head above the parapet.

But don't let my laziness stop you from drilling into it and proposing a Pull Request. The more the merrier! 🤡

Slyke, 2020-10-08:
@Londoneye02 new menu is on the `experimental` branch if you want to give it a try and see if the issue happens there:
```
git checkout master
git pull
git checkout experimental
./menu.sh
```

To switch back to standard:
```
git checkout master
./menu.sh
```

Londoneye02, 2020-10-09:
hi @Slyke

I tested with new menu and it seemed to work correctly...

- switched to experimental
- create stack with new menu
- docker-compose up -d
- open menu again. No old portainer
- checkout master branch
- run old menu... There is the old portainer again

Thank you!!

Slyke, 2020-10-09:
Awesome! Glad to know it's working!

closed 2020-10-09 by Slyke