SensorsIot/IOTstack #146: docker not installed

Tasm-Devil, 2020-10-09:
Hi,
I'm new to this project and wanted to play around with it in a fresh installation of ubuntu-minimal.
After cloning the repo and calling ./menu.sh, I get the following
`
~/IOTstack$ ./menu.sh 
~/IOTstack ~/IOTstack
checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
docker not installed
~/IOTstack$ 
`

Tasm-Devil, 2020-10-09:
OK. Here is the reason: whiptail is not installed
`sudo apt install whiptail`
solves it.

Slyke, 2020-10-15:
Hmm, you may experience further issues. I haven't tested out the stack on minimal Ubuntu. I always use the standard server distro.

Tasm-Devil, 2020-10-15:
For the moment, it works like a charm. I use node-red, mosquito, influxdb, grafana and portainer. Just fine.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke