SensorsIot/IOTstack #149: Question about integrating TTS Service and IOTStack's HASS installation

mbariola, 2020-10-13:
Hi all and thanks for continuing @gcgarner's excellent project. I have a question, not an issue report

I am trying to make [this tts bluetooth speaker service](https://github.com/pkozul/ha-tts-bluetooth-speaker) work with the home assistant version installed by IOTStack's script. I cannot proceed past point 2 as a homeassistant user seems to not exist in the "out of docker" console. I have a very holey understanding of how services and functionalites are spread among the different containers of Home Assistant installed by the IOTStack script. any pointers as to how to install this functionality (install it on one of the 3 containers created by the IOTStack script?) for the time being, and to integrate this functionality in a way that is backup-able and resistant to updates of IOTStack is very appreciated.  Thanks!

Paraphraser, 2020-10-13:
I am not going to pretend that I understand the fine detail of your question but it **sounds** like the same "problem" that Mosquitto has.

**Inside** the mosquitto container, the user named "mosquitto" exists and has user ID 1883:

```
$ docker exec mosquitto grep "1883" /etc/passwd
mosquitto:x:1883:1883:mosquitto:/var/empty:/sbin/nologin
```

**Outside** the container, user ID 1883 is undefined:

```
$ grep "1883" /etc/passwd
$
```

while the username "mosquitto" has a different user ID:

```
$ grep "mosquitto" /etc/passwd
mosquitto:x:111:116::/var/lib/mosquitto:/usr/sbin/nologin
```

mosquitto (the container) depends on user ID 1883 working both inside and outside the container. Here's the snippet from the `docker-compose.yml`:

```
  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto
    restart: unless-stopped
    user: "1883"
```

The way IOTstack resolves this ambiguity is by explicitly using the ID of 1883 **outside** the container. The place to look is mosquitto's `directoryfix.sh`:

```
$ cat ./services/mosquitto/directoryfix.sh 
#!/bin/bash

[ -d ./volumes/mosquitto ] || sudo mkdir -p ./volumes/mosquitto

#check user 1883
if [ $(grep -c 'user: \"1883\"' ./services/mosquitto/service.yml) -eq 1 ]; then
	echo "...found user 1883"
	sudo mkdir -p ./volumes/mosquitto/data/
	sudo mkdir -p ./volumes/mosquitto/log/
	sudo mkdir -p ./volumes/mosquitto/pwfile/
	sudo chown -R 1883:1883 ./volumes/mosquitto/
fi

#check user 0 legacy test
if [ $(grep -c 'user: \"0\"' ./services/mosquitto/service.yml) -eq 1 ]; then
	echo "...found user 0 setting ownership for old template"
	sudo chown -R root:root ./volumes/mosquitto/
fi
```

And, as implied by the above, you expect to see user ID 1883 showing up in `./volumes`:

```
$ ls -ld ./volumes/mosquitto
drwxr-xr-x 5 1883 1883 4096 Jun 16 14:34 ./volumes/mosquitto
```

If I have guessed correctly, you should be able to figure out the user ID being used inside the container, and then just use that outside the container, instead of the name.

If I haven't guessed correctly then...😔

mbariola, 2020-10-14:
@Paraphraser hi, sorry if it took a little to answer, I wanted to make a backup of the whole sd card before tinkering.
I think that that integration needs to figure out what is the user under which the IOTStack homeassistant works, to allow it to access the bluetooth TTS.

I tried a `docker exec homeassistant less /etc/passwd` and this is what I got

`root:x:0:0:root:/root:/bin/ash
bin:x:1:1:bin:/bin:/sbin/nologin
daemon:x:2:2:daemon:/sbin:/sbin/nologin
adm:x:3:4:adm:/var/adm:/sbin/nologin
lp:x:4:7:lp:/var/spool/lpd:/sbin/nologin
sync:x:5:0:sync:/sbin:/bin/sync
shutdown:x:6:0:shutdown:/sbin:/sbin/shutdown
halt:x:7:0:halt:/sbin:/sbin/halt
mail:x:8:12:mail:/var/mail:/sbin/nologin
news:x:9:13:news:/usr/lib/news:/sbin/nologin
uucp:x:10:14:uucp:/var/spool/uucppublic:/sbin/nologin
operator:x:11:0:operator:/root:/sbin/nologin
man:x:13:15:man:/usr/man:/sbin/nologin
postmaster:x:14:12:postmaster:/var/mail:/sbin/nologin
cron:x:16:16:cron:/var/spool/cron:/sbin/nologin
ftp:x:21:21::/var/lib/ftp:/sbin/nologin
sshd:x:22:22:sshd:/dev/null:/sbin/nologin
at:x:25:25:at:/var/spool/cron/atjobs:/sbin/nologin
squid:x:31:31:Squid:/var/cache/squid:/sbin/nologin
xfs:x:33:33:X Font Server:/etc/X11/fs:/sbin/nologin
games:x:35:35:games:/usr/games:/sbin/nologin
cyrus:x:85:12::/usr/cyrus:/sbin/nologin
vpopmail:x:89:89::/var/vpopmail:/sbin/nologin
ntp:x:123:123:NTP:/var/empty:/sbin/nologin
smmsp:x:209:209:smmsp:/var/spool/mqueue:/sbin/nologin
guest:x:405:100:guest:/dev/null:/sbin/nologin
nobody:x:65534:65534:nobody:/:/sbin/nologin
messagebus:x:100:101:messagebus:/dev/null:/sbin/nologin
`

while the /etc/passwd of mi Pi has the following:

`root:x:0:0:root:/root:/bin/bash
daemon:x:1:1:daemon:/usr/sbin:/usr/sbin/nologin
bin:x:2:2:bin:/bin:/usr/sbin/nologin
sys:x:3:3:sys:/dev:/usr/sbin/nologin
sync:x:4:65534:sync:/bin:/bin/sync
games:x:5:60:games:/usr/games:/usr/sbin/nologin
man:x:6:12:man:/var/cache/man:/usr/sbin/nologin
lp:x:7:7:lp:/var/spool/lpd:/usr/sbin/nologin
mail:x:8:8:mail:/var/mail:/usr/sbin/nologin
news:x:9:9:news:/var/spool/news:/usr/sbin/nologin
uucp:x:10:10:uucp:/var/spool/uucp:/usr/sbin/nologin
proxy:x:13:13:proxy:/bin:/usr/sbin/nologin
www-data:x:33:33:www-data:/var/www:/usr/sbin/nologin
backup:x:34:34:backup:/var/backups:/usr/sbin/nologin
list:x:38:38:Mailing List Manager:/var/list:/usr/sbin/nologin
irc:x:39:39:ircd:/var/run/ircd:/usr/sbin/nologin
gnats:x:41:41:Gnats Bug-Reporting System (admin):/var/lib/gnats:/usr/sbin/nologin
nobody:x:65534:65534:nobody:/nonexistent:/usr/sbin/nologin
systemd-timesync:x:100:102:systemd Time Synchronization,,,:/run/systemd:/usr/sbin/nologin
systemd-network:x:101:103:systemd Network Management,,,:/run/systemd:/usr/sbin/nologin
systemd-resolve:x:102:104:systemd Resolver,,,:/run/systemd:/usr/sbin/nologin
_apt:x:103:65534::/nonexistent:/usr/sbin/nologin
pi:x:1000:1000:,,,:/home/pi:/bin/bash
messagebus:x:104:110::/nonexistent:/usr/sbin/nologin
_rpc:x:105:65534::/run/rpcbind:/usr/sbin/nologin
statd:x:106:65534::/var/lib/nfs:/usr/sbin/nologin
sshd:x:107:65534::/run/sshd:/usr/sbin/nologin
avahi:x:108:113:Avahi mDNS daemon,,,:/var/run/avahi-daemon:/usr/sbin/nologin
systemd-coredump:x:996:996:systemd Core Dumper:/:/usr/sbin/nologin
dnsmasq:x:109:65534:dnsmasq,,,:/var/lib/misc:/usr/sbin/nologin
rtkit:x:110:114:RealtimeKit,,,:/proc:/usr/sbin/nologin
pulse:x:111:115:PulseAudio daemon,,,:/var/run/pulse:/usr/sbin/nologin
`

I have no clue which user homeassistant runs as , in the homeassistant container. maybe root?

Paraphraser, 2020-10-14:
Before I continue, I need to make sure you understand that I have **zero** knowledge of this topic. I see "Hass.io", "HA", "homeassistant", "home assistant" and "home bridge" but I have no idea whether they are the same thing or five separate entities. I don't know why there is a separate install for "home assistant" in `menu.sh`. Why it isn't just pulled down as another container? I don't even know if «insert one of the many names here» runs as a container under Docker or has some separate existence. And as to what actual IoT purpose(s) this particular alphabet soup might serve ... ?

Take everything with a truckload of salt!!

<hr>

Tackling your last question first, figuring out which user a container runs as can involve a bit of sleuthing.

Step 1. Work out the process name associated with the container. Sometimes it's the obvious:

```
$ ps -ax | grep mosquitto
 1507 ?        Ss     3:41 /usr/sbin/mosquitto -c /mosquitto/config/mosquitto.conf
```

so it's "mosquitto". Sometimes it's not so obvious:

```
$ ps -ax | grep red
 8669 ?        Sl     0:09 node-red
```

so it's "node-red". And sometimes it can be downright obscure:

```
$ ps -ax | grep pihole   
 3860 ?        S      0:00 s6-supervise pihole-FTL
 3880 ?        Sl     3:30 pihole-FTL no-daemon
```

giving us "pihole-FTL".

Step 2. Once you know the process name, substitute it into this command:

```
$ ps -C «process name» -o euser,ruser,suser,fuser,comm
```

as in these examples:

```
$ ps -C node-red -o euser,ruser,suser,fuser,comm
EUSER    RUSER    SUSER    FUSER    COMMAND
root     root     root     root     node-red

$ ps -C mosquitto -o euser,ruser,suser,fuser,comm
EUSER    RUSER    SUSER    FUSER    COMMAND
1883     1883     1883     1883     mosquitto
```

Thus, node-red is running as root while mosquitto is running as user 1883 which, as I said before, doesn't map to anything in `/etc/passwd` outside the container.

For the sake of completeness, you can use the `-C` option multiple times and get a less cluttered display:

```
$ ps -C node-red -C mosquitto -o euser,ruser,suser,fuser,comm
EUSER    RUSER    SUSER    FUSER    COMMAND
1883     1883     1883     1883     mosquitto
root     root     root     root     node-red
```

<hr>

I am pretty sure that User IDs (and Group IDs) are what count, with user and group names just being there for human readability. IDs get stored in file system directory entries.

It is clear that Mosquitto works without ID=1883 mapping to anything so I'd take that as proof that the name is irrelevant.

If something in the file system **outside** the container needs a particular ID to keep TTS happy then I'd follow the lead of `directoryfix.sh` for Mosquitto and just use whatever ID it wants.
 
<hr>

A few months back I was trying to get Bluetooth working in Node-Red. You will find the details in [PR 70](https://github.com/SensorsIot/IOTstack/pull/70) and some discussion in [Issue 64](https://github.com/SensorsIot/IOTstack/issues/64)

One of the key steps was:

```
$ sudo usermod -G bluetooth -a $USER
$ sudo reboot
```

I don't actually know whether membership of "bluetooth" is a general "bluetooth hardware access" thing (meaning it would also be necessary for bluetooth TTS) or was only needed for the Node-Red node I was trying to get working. But it seems reasonable to assume that it is general.

In theory, providing the "bluetooth" group exists (which I believe indicates bluetooth hardware is present), auto-adding the current user ("pi") to that group has been part of `menu.sh` for quite some time so I assume that if you run the `groups` command on your RPi, you will see that "pi" is a member of "bluetooth".

You pointed me at [ha-tts-bluetooth-speaker](https://github.com/pkozul/ha-tts-bluetooth-speaker) but I also found [bluetooth-tts-with-homeassistant](https://jansblog.org/2018/04/25/bluetooth-tts-with-homeassistant/). The reason I mention it is because the GitHub repo has:

```
sudo adduser pi pulse-access
sudo adduser homeassistant pulse-access
```

but does not seem to create the "pulse-access" group, while the other example seems to take care of that:

```
addgroup --system pulse
adduser --system --ingroup pulse --home /var/run/pulse pulse
addgroup --system pulse-access
adduser pulse audio
adduser homeassistant pulse-access
```

Unless the `sudo apt-get install …` somehow takes care of creating "pulse-access", I don't see how it gets done.

Does your `/etc/groups` contain "pulse-access"? If not, you might need to add it. The command probably needs `sudo` but I'll leave you to figure out the significance of the `--system` option.

Then see if `groups` reports "pi" as being a member. What I don't know is whether there is a difference between:

```
$ sudo adduser pi pulse-access 
```

and

```
$ sudo usermod -G pulse-access -a $USER
```

The `usermod` form is what is used for both Docker and Bluetooth so, if it were me, I'd put more faith in that.

> Why `usermod` needs a reboot to take effect is a mystery. 

It makes sense that "pi" should be a member of "pulse-access" and I **think** the containers inherit this. So, it might just be that fixing this fixes your problem.

> Why haven't I mentioned the homeassistant user? Because I don't think it will be relevant in a docker context. But remember what I said about knowing faff all.

<hr>

Comparing/contrasting your two passwd files, I don't see any obvious candidate in the **internal** file so my guess is the answer is "it runs as root".

mbariola, 2020-10-15:
@Paraphraser first of all, I am quite impressed at the quality, precision and informative detail of your answers. thanks for the time you take in crafting them, I am sure they will help others too. Are you perchance a technical instructor?

As for me, both Home Assistant and Docker (not to mention IOTStack) are also very new topics. Practically, I started the day before opening the question, and I feel like the canvas is quite vast and am still some ways from beginning to find my bearings around it.

Indeed it seems that home assistant is not installed or configured in the same way as the other containers, should you install Portainer, last time I checked there were 15 containers running, and 1 stack which definitely only contained what services can be added via the IOTStack setup script, and not the Home Assistant ones, which are installed via their own different entry in the menu, and consist in 5 different containers: homeassistant, hassio_multicast, hassio_audio, hassio_cli and hassio_dns. these containers are not in the stack detected by portainer.  I would infer that probably either of the first 3 containers are the ones which would be interested with this bluetooth tts thing.

Questions start from your step 1. It is not obvious to me whether should I need to ps -ax from the raspberry console (outside the containers) or within them? I surmise outside, here is the output both looking for home and for hassio, from the raspberry console.

```
pi@raspberrypi:~/IOTstack $ ps ax | grep home
4299 ?        S      0:00 s6-supervise home-assistant
4302 ?        Ssl    3:35 python3 -m homeassistant --config /config
4870 ?        S      0:00 s6-supervise esphome
4874 ?        Ssl    0:25 /usr/bin/python3 /usr/local/bin/esphome /config/esphome dashboard --socket /var/run/esphome.sock --hassio
5348 ?        Ssl    0:43 grafana-server --homepath=/usr/share/grafana --config=/etc/grafana/grafana.ini --packaging=docker cfg:default.log.mode=console cfg:default.paths.data=/var/lib/grafana cfg:default.paths.logs=/var/log/grafana cfg:default.paths.plugins=/var/lib/grafana/plugins cfg:default.paths.provisioning=/etc/grafana/provisioning
15674 pts/0    S+     0:00 grep --color=auto home

```

```
pi@raspberrypi:~/IOTstack $ ps ax | grep hassio
 1352 ?        Ss     0:00 bash /usr/sbin/hassio-supervisor
 1399 ?        S      0:00 bash /usr/sbin/hassio-supervisor
 1400 ?        Sl     0:05 docker start --attach hassio_supervisor
 3841 ?        Ss     0:14 mdns-repeater -f eth0 hassio
 4874 ?        Ssl    0:25 /usr/bin/python3 /usr/local/bin/esphome /config/esphome dashboard --socket /var/run/esphome.sock --hassio
15683 pts/0    S+     0:00 grep --color=auto hassio

```
as soon as you can clarify me where to execute those ps'es I will go on with the rest of your instructions. thanks again!
I can confirm that in /etc/group there is a pulse-access user, I guess with user 116
```
pulse:x:115:
pulse-access:x:116:pi

```

Paraphraser, 2020-10-16:
> technical instructor ?

No, I never had that job title. But you probably know how things are when you work in IT. You develop a bit of knowledge about some topic and, next thing you know, someone wants you to "do a presentation about it". A long career equals a lot of practice. I also spent a decade teaching statistics and database design to undergrads (best job I ever had) so maybe it's something from that experience you're sensing. Or it could be that I'm a fast typist who doesn't know when to stop.

> started the day before ... canvas is quite vast

That's me, almost exactly a year ago. Definitely a steep learning-curve but I reckon it's worth it. I love my little RPis. I'm a total convert to Docker too. That's my real motivation for helping out when I can.

> 15 containers running

I'd be careful. Some people have mentioned running out of Raspbian resources. I can't remember the symptoms or details but you might find the info by searching issues or the Discord channel. I only run seven: mosquitto, nodered, influxdb, grafana, portainer-ce, pihole and gitea. If I had to pick one to sacrifice it would be portainer (the only time I ever use it is if I'm trying to answer a question about it).

> inside vs outside

The convention I try to follow is based on what you see during normal command line interactions. If you `ssh` to your Raspberry Pi, you're going to get a prompt like this:

```
pi@sec-dev:~$ 
```

If you go into a container like, say, Node-Red, you're going to get a prompt like:

```
bash-5.0#
```

Similarly, if you are outside the containers and you run `sudo -s` to create a privileged shell, you're going to see a prompt like:

```
root@sec-dev:/home/pi# 
```

What I do is just use the last character of the usual prompt to indicate where a command should be executed. I eliminate the guff before the prompt character because it doesn't add anything (and because everyone's RPi name will be different).

So, if I am saying "run this `ps` command outside the container" it will be:

```
$ ps -ax
```

If I want to indicate "do this inside a privileged shell", it will be either:

```
$ sudo ps -ax
```

or:

```
$ sudo -s
# ps -ax
# exit
$
```

If I want to indicate "do this inside the Node-Red container", it'll be either:

```
$ docker exec -it nodered bash
# ps ax
# exit
$
```

or, as a one-liner:

```
$ docker exec nodered ps ax
```

[... 47 lines cut ...]
* The **group** named "pi" has a group ID 1000
* The current user ("pi") belongs to a whole bunch of groups

There's a many-to-one relationship between `/etc/passwd` and `/etc/group`:

* You can always go from a group ID in `/etc/passwd` and determine a single unique group name in `/etc/group`. For example, group ID 1000 in `/etc/passwd`, when looked-up in `/etc/group`, leads to group name "pi", but
* You can't *always* go in the other direction because the answer can be more than one user belonging to the same group.

In the case of the RPi, you generally only have the "pi" user doing anything so, pretty much, the lines in `/etc/group` either end in ":pi" (meaning "pi is an explicit member of this group") or just end with ":" (either nobody is a member of this group or it's implicitly the user with the same name as the group name). If you start adding users you'll start to see more names tacked onto the end of lines in `/etc/group`. But, in reality, too much deep diving into these two files is a fast path to insanity. My rules of thumb are:

* Does user *target* exist? `grep "^target" /etc/passwd`
* Does group *target* exist? `grep "^target" /etc/group`
* What groups does the current user belong to? `groups`
* What groups does *target* belong to? `groups target`
* Which users belong to group *target*? No solution (unless you install add-ons like `members`).

<hr>

Anyway, you have those two groups and we can see "pi" is a member of "pulse-access". That leaves whether "pi" should also be a member of "pulse"?

Also, I'd be doing some Googling around the topic of the difference between the "adduser" and "usermod" forms of adding members to groups. It always makes me itch when there are two commands apparently doing the same thing. I wonder why the second command (whichever it was) needed to be written. It's one thing if the answer is "convenience" but sometimes the second command does something extra, which turns out to be what you need.

I keep coming back to menu.sh and how it uses "usermod" to add pi to the docker group. Why not adduser? Why a restart to make it take effect? Why did I have trouble with bluetooth until I used the usermod form?

But I'll leave it to you to dig into that.

mbariola, 2020-10-17:
@Paraphraser thanks again for the in detail contribution. As for me, my first lines of "serious" code were around 1987, and I did teach some IT subjects myself so I can relate to your teaching experience.

Unfortunately, I won't be able to access my Pi for a few days (coronavirus restrictions) so I won't be able to go further (yeah, never wanted to access it from outside as it's still a "fun hobby project". guess I'll need to set it up to at least check on my place when I'm away.....)

I'll update as soon as I'm able to have a shell conversation with it!

mbariola, 2020-11-04:
@Paraphraser sorry for the long wait. I have a couple hours today to further this, I don't think I'll be able to solve it all today but I can gather some info about the system. Probably it'd be a good idea for me to also delve into how IOTStack installs Home Assistant, and on general docker workings i.e. how users and services from within a container can access what is outside it.

So this is what I gathered so far and shows me that I need to start from almost scratch in creating this [integration](https://github.com/pkozul/ha-tts-bluetooth-speaker) if I want to go that way:  

```
pi@raspberrypi:~ $ getent passwd | head
root:x:0:0:root:/root:/bin/bash
daemon:x:1:1:daemon:/usr/sbin:/usr/sbin/nologin
bin:x:2:2:bin:/bin:/usr/sbin/nologin
sys:x:3:3:sys:/dev:/usr/sbin/nologin
sync:x:4:65534:sync:/bin:/bin/sync
games:x:5:60:games:/usr/games:/usr/sbin/nologin
man:x:6:12:man:/var/cache/man:/usr/sbin/nologin
lp:x:7:7:lp:/var/spool/lpd:/usr/sbin/nologin
mail:x:8:8:mail:/var/mail:/usr/sbin/nologin
news:x:9:9:news:/var/spool/news:/usr/sbin/nologin
pi@raspberrypi:~ $ getent group | head
root:x:0:
daemon:x:1:
bin:x:2:
sys:x:3:
adm:x:4:pi
tty:x:5:
disk:x:6:
lp:x:7:
mail:x:8:
news:x:9:

```

So these are all the groups and users currently known to the host Pi OS. Also, the folder where IOTStack installs the Supervised Home Assistant seems to be `/usr/share/hassio/`

```
pi@raspberrypi:/ $ ls -la /usr/share/hassio/
total 96
drwxr-xr-x  12 root root 4096 Oct 30 13:21 .
drwxr-xr-x 112 root root 4096 Oct 13 09:52 ..
drwxr-xr-x   6 root root 4096 Oct  7 00:12 addons
-rw-------   1 root root 6850 Nov  4 10:14 addons.json
drwxr-xr-x   4 root root 4096 Oct  7 00:09 apparmor
drwxr-xr-x   4 root root 4096 Oct  7 00:11 audio
-rw-------   1 root root   68 Nov  4 10:12 audio.json
drwxr-xr-x   2 root root 4096 Nov  3 11:54 backup
-rw-------   1 root root  207 Nov  4 10:12 cli.json
-rw-------   1 root root  282 Nov  4 10:14 config.json
drwxr-xr-x   2 root root 4096 Nov  2 14:53 dns
-rw-------   1 root root   90 Nov  4 10:12 dns.json
-rw-------   1 root root   22 Oct 30 13:56 docker.json
drwxr-xr-x   8 root root 4096 Nov  4 11:11 homeassistant
-rw-------   1 root root  547 Nov  4 10:13 homeassistant.json
-rw-------   1 root root  224 Nov  4 10:12 ingress.json
drwxr-xr-x   2 root root 4096 Oct  7 00:09 media
-rw-------   1 root root   71 Nov  4 10:12 multicast.json
-rw-------   1 root root  212 Nov  4 10:12 observer.json
drwxr-xr-x   2 root root 4096 Oct 30 13:57 share
drwxr-xr-x   2 root root 4096 Oct  7 00:09 ssl
drwxr-xr-x   2 root root 4096 Nov  4 10:12 tmp
-rw-------   1 root root  587 Nov  4 10:14 updater.json

```

there's an interesting `audio.json` whose contents are

```
{
  "version": "17",
  "image": "homeassistant/armv7-hassio-audio"
}

```

Hmm! I guess that that's where I need to delve into, and it might make sense, for global compatibility, to try and integrate whatever I want to including this repo, with the workings of this image. I'll have a look into it as soon as I can.