SensorsIot/IOTstack #150: Running on other OS

just-jason, 2020-10-13:
Would it be possible to use this on other OS's such as Ubuntu. It.seems to be failing when creating the containers when I try and use this under Ubuntu. I see in the installation instructions that is is expecting user pi and home/pi as the directory, is there somewhere that I can change this?

Paraphraser, 2020-10-13:
Short answer: no.

Over on the Discord channel I wrote this (using copy & paste to save myself some re-typing)

<hr>

I've said before that I think some emphasis would be useful at the start of the IOTstack README.md, "IOTstack is a builder for docker-compose to easily make and maintain IoT stacks **on the Raspberry Pi**". Maybe the sentence should also say "**under Raspberry Pi OS (Raspbian), installed directly, running natively, with no boot managers, hypervisors or the like cluttering-up the works**". I have nothing against OSes-not-Raspbian, or BerryBoots or VMs and I know how easy it is for a message (like this) that is **intended** to communicate "you might have more luck doing it differently" to be **received** as "you're doing it wrong". That isn't the message. It probably should work.

For me, it's all about setting expectations properly. My own experience is IOTstack "just works" **in its target environment**. I think that's because the target environment gets lots of testing. The evidence of Discord questions + repo issues suggests that, the further you get from the target, the less the testing, the greater the chance of grief & gotchas, the less the chance that someone will have found a fix. gcgarner#173 is likely an example and being more clear about the target environment would avoid/minimise such hassles for IOTstack adoptees.

<hr>

`~/IOTstack` as that exact spelling is hard-baked into everything. Just one example:

```
$ grep IOTstack menu.sh 
pushd ~/IOTstack
        "update" "Update IOTstack" \
                if [ $(grep -c 'IOTstack' ~/.bash_aliases) -eq 0 ]; then
                        echo ". ~/IOTstack/.bash_aliases" >>~/.bash_aliases
                [ -d ~/IOTstack/backups ] || sudo mkdir -p ~/IOTstack/backups/
                sudo touch ~/IOTstack/backups/dropbox
                [ -d ~/IOTstack/backups ] || sudo mkdir -p ~/IOTstack/backups/
                sudo touch ~/IOTstack/backups/rclone
```

To be perfectly honest, there is no need for the "pushd". The script could assume it was launched in the right place, perhaps doing some kind of basic checks like "$0"="./menu.sh" as a way of asserting "no long paths to launch me, please" and that the working directory contained .templates. I can't immediately see any reason why the other hard-coded cases could not be rewritten to use relative paths (eg `./backups` instead of `~/IOTstack/backups`). It'd take a bit of testing but looks doable. And the same is probably true of every other script in the repo. For all I know, the new menu @Slyke is developing deals with a lot of this.

But, right now, it is what it is, so I come back to "setting expectations correctly" as pretty much the only way we can signal both "we understand your pain" and "your mileage may vary".

just-jason, 2020-10-14:
Thanks for clearing that up.

closed 2020-10-14 by just-jason