SensorsIot/IOTstack #151: Nodered will not install

Tankdoz, 2020-10-13:
I've started with new installs twice already. Got Nodered working if I don't install any additional modules but as soon as I try to add modules I get these messages after I do a docker-compose up -d:

pi@RPI-Server:~/IOTstack $ docker-compose up -d nodered
Building nodered
Step 1/2 : FROM nodered/node-red:latest
 ---> fa3bc6f20464
Step 2/2 : RUN for addonnodes in node-red-node-pi-gpiod node-red-dashboard node-red-contrib-influxdb node-red-contrib-boolean-logic node-red-node-rbe node-red-configurable-ping node-red-node-email node-red-contrib-deconz ; do npm install ${addonnodes} ;done;
 ---> Running in c3072eb20d1c
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-node-pi-gpiod failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_17_39_069Z-debug.log
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-dashboard failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_17_48_206Z-debug.log
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-contrib-influxdb failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_17_58_470Z-debug.log
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-contrib-boolean-logic failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_18_07_452Z-debug.log
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-node-rbe failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_18_16_452Z-debug.log
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-configurable-ping failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_18_25_651Z-debug.log
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-node-email failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_18_34_781Z-debug.log
npm ERR! code EAI_AGAIN
npm ERR! errno EAI_AGAIN
npm ERR! request to https://registry.npmjs.org/node-red-contrib-deconz failed, reason: getaddrinfo EAI_AGAIN registry.npmjs.org registry.npmjs.org:443

npm ERR! A complete log of this run can be found in:
npm ERR!     /usr/src/node-red/.npm/_logs/2020-10-13T16_18_43_780Z-debug.log
ERROR: Service 'nodered' failed to build: The command '/bin/sh -c for addonnodes in node-red-node-pi-gpiod node-red-dashboard node-red-contrib-influxdb node-red-contrib-boolean-logic node-red-node-rbe node-red-configurable-ping node-red-node-email node-red-contrib-deconz ; do npm install ${addonnodes} ;done;' returned a non-zero code: 1

--------
Unfortunately the debug logs do not exist in the indicated location. Does anybody know how to proceed?

Paraphraser, 2020-10-13:
According to Google, EAI_AGAIN is "a DNS lookup timed out error, means it is a network connectivity error or proxy related error." Can you ping things outside your network (eg www.apple.com, www.microsoft.com) from the host where you are trying to install IOTstack?

What does a `uname -a` say?

One of the "problems" with Docker is that, if a container is restarting, its logs are pretty much inaccessible. While there are no guarantees, you could try mapping the logging directory and see what turns up. In your docker-compose.yml, find the nodered definition, then the volumes section, and add

```
            - ./volumes/nodered/logs:/usr/src/node-red/.npm/_logs/
```

using SPACES not tabs to make sure it lines up, then `docker-compose up -d nodered`. Anything written to the _logs folder inside the container will turn up in ~/IOTstack/volumes/nodered/logs outside the container.

Tankdoz, 2020-10-14:
hi Paraphraser, 
Let me answer your questions first:
1. yes I can ping things outside my network:
pi@RPI-Server:~/IOTstack $  ping www.apple.com
PING e6858.dsce9.akamaiedge.net (23.208.76.97) 56(84) bytes of data.
64 bytes from a23-208-76-97.deploy.static.akamaitechnologies.com (23.208.76.97): icmp_seq=1 ttl=57 time=13.6 ms
64 bytes from a23-208-76-97.deploy.static.akamaitechnologies.com (23.208.76.97): icmp_seq=2 ttl=57 time=12.5 ms
64 bytes from a23-208-76-97.deploy.static.akamaitechnologies.com (23.208.76.97): icmp_seq=3 ttl=57 time=11.0 ms

2. 
pi@RPI-Server:~ $ uname -a
Linux RPI-Server 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

3.
pi@RPI-Server:~/IOTstack $ sudo nano docker-compose.yml (added the suggested line) 
pi@RPI-Server:~/IOTstack $ docker-compose up -d nodered
Building nodered
Step 1/2 : FROM nodered/node-red:latest
 ---> fa3bc6f20464
Step 2/2 : RUN for addonnodes in node-red-node-pi-gpiod node-red-dashboard node-red-contrib-influxdb node-red-contrib-boolean-logic node-red-node-rbe node-red-configurable-ping node-red-node-email ; do npm install ${addonnodes} ;done;
 ---> Running in 742ed88b9cdb
npm notice created a lockfile as package-lock.json. You should commit this file.
+ node-red-node-pi-gpiod@0.1.0
added 4 packages from 2 contributors and audited 347 packages in 9.012s

2 packages are looking for funding
  run `npm fund` for details

found 5 vulnerabilities (1 low, 3 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
+ node-red-dashboard@2.23.4
added 55 packages from 47 contributors and audited 402 packages in 16.891s

2 packages are looking for funding
  run `npm fund` for details

found 5 vulnerabilities (1 low, 3 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
+ node-red-contrib-influxdb@0.4.1
added 3 packages from 6 contributors and audited 405 packages in 11.53s

2 packages are looking for funding
  run `npm fund` for details

found 6 vulnerabilities (2 low, 3 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
+ node-red-contrib-boolean-logic@0.0.3
added 1 package from 1 contributor and audited 406 packages in 9.396s

2 packages are looking for funding
  run `npm fund` for details

found 6 vulnerabilities (2 low, 3 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
+ node-red-node-rbe@0.2.9
updated 1 package and audited 407 packages in 21.134s

2 packages are looking for funding
  run `npm fund` for details

found 6 vulnerabilities (2 low, 3 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
+ node-red-configurable-ping@1.0.1
added 1 package from 1 contributor and audited 408 packages in 11.143s

2 packages are looking for funding
  run `npm fund` for details

found 6 vulnerabilities (2 low, 3 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details

> nodemailer@6.4.11 postinstall /usr/src/node-red/node_modules/mailparser/node_modules/nodemailer
> node -e "try{require('./postinstall')}catch(e){}"

+ node-red-node-email@1.7.11
added 28 packages from 15 contributors and audited 436 packages in 17.782s

2 packages are looking for funding
  run `npm fund` for details

found 8 vulnerabilities (4 low, 3 moderate, 1 high)
  run `npm audit fix` to fix them, or `npm audit` for details
Removing intermediate container 742ed88b9cdb
 ---> 5c7d0632f8e0

Successfully built 5c7d0632f8e0
Successfully tagged iotstack_nodered:latest
WARNING: Image for service nodered was built because it did not already exist. To rebuild this image you must use `docker-compose build` or `docker-compose up --build`.
Creating nodered ... done

---------

This was completely different from yesterdays output, allthough the only thing I changed was that I deleted the nodered image in docker before I started in the hope that a new download would fix things. I can confirm that nodered is now running with the new packages so that is good news. the logs directory (outside the container) is empty en inside the container the log says:

--------

> node-red-docker@1.1.3 start /usr/src/node-red
> node $NODE_OPTIONS node_modules/node-red/red.js $FLOWS "--userDir" "/data"
14 Oct 10:01:41 - [info] 
Welcome to Node-RED
===================
14 Oct 10:01:41 - [info] Node-RED version: v1.1.3
14 Oct 10:01:41 - [info] Node.js  version: v10.22.0
14 Oct 10:01:41 - [info] Linux 5.4.51-v7l+ arm LE
14 Oct 10:01:42 - [info] Loading palette nodes
14 Oct 10:01:45 - [info] Dashboard version 2.23.4 started at /ui
14 Oct 10:01:46 - [info] Settings file  : /data/settings.js
14 Oct 10:01:46 - [info] Context store  : 'default' [module=memory]
14 Oct 10:01:46 - [info] User directory : /data
14 Oct 10:01:46 - [warn] Projects disabled : editorTheme.projects.enabled=false
14 Oct 10:01:46 - [info] Flows file     : /data/flows.json
14 Oct 10:01:46 - [info] Creating new flow file
14 Oct 10:01:46 - [warn] 
---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.
If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.
You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------
14 Oct 10:01:46 - [info] Starting flows
14 Oct 10:01:46 - [info] Started flows
14 Oct 10:01:46 - [info] Server now running at http://127.0.0.1:1880/

------
Thanks very much for helping out!

Paraphraser, 2020-10-14:
You're welcome. First time I've seen anything like this, though. I'll file that one away.

closed 2020-10-14 by Tankdoz