SensorsIot/IOTstack #153: OpenHAB v. 2.5

egholm, 2020-10-15:
OpenHAB was upgraded to version 2.5 last december but IOTstack seems to install 2.4. Is there any reason not to install 2.5 or am I misunderstanding the 'image: "openhab/openhab:2.4.0"' in 'services/openhab/service.yml'?

Paraphraser, 2020-10-15:
I think a lot of pinning of containers to specific versions was done by Graham Garner in the original repo, and has been inherited by this repo.

In the cases where it has affected the containers I run, I have removed the pin, tried the latest and, thus far, have never encountered a problem which caused me to think Graham was onto something. After a time with no issues in my own environment, I've proposed Pull Requests to remove the pins from the repo.

Personally, I would prefer it if there were no pins for anything so that it was left up to the user to add a pin if an update exposed a problem.

If I try to put myself in Graham's shoes and imagine why he may have thought pinning was a good idea, the only reason that makes any sense to me is that he was trying to get a lot of things going at once and it was his way of maintaining some stability - he wanted to upgrade containers one at a time so that if he saw any problems he could attribute them to the one thing he had changed. But I don't really **know**.

I'd advise doing what I do. Remove the pin. If it looks OK to you, prepare a PR.

MuhammadYacoub, 2021-06-15:
I think it's done , now OpenHAP updated to v3 , I suggest to close the issue if i can do suggestions, thanks