SensorsIot/IOTstack #155: backupscript unable to backup mariadb

benpoe, 2020-10-18:
when trying the backup script, it has a problem with mariadb if it's in use (of course)

```
tar: ./volumes/mariadb/config/databases/ib_logfile0: file changed as we read it
tar: ./volumes/mariadb/config/log/mysql/mariadb-bin.000005: file changed as we read it
```

isn't it better to use mysqldump to backup mariadb databases?

Paraphraser, 2020-10-19:
Yes. Someone needs to write the script.

Slyke, 2020-10-21:
I'm working on the backup/restore section on the `experimental` branch right now. I have a almost finished version of the restore script, with backup working. This doesn't include any database management though, and those will be separate scripts (which will automatically detect and run the appropriate script to backup that specific database engine). There will be documentation accompanying it, which will explain how to use the backup hooks to also run your own scripts for preparing any database engine for backup.

I think that during the backup, the database engines have to be brought offline though, to ensure data accuracy and to not have half completed transactions backed up. Probably not a big issue for home run servers, but still something to be weary about.

Backup: https://github.com/SensorsIot/IOTstack/blob/experimental/scripts/backup.sh
Restore: https://github.com/SensorsIot/IOTstack/blob/experimental/scripts/restore.sh
Custom Hook Handlers (the `pre_backup_complete.sh` script runs before files are zipped, and where the database detection and backup logic will be): https://github.com/SensorsIot/IOTstack/tree/experimental/scripts/backup_restore