SensorsIot/IOTstack #157: Problems with install hassio

To1952P, 2020-10-22:
For all those whoes trying to install hassio and get the message "Missing apparmor and network manager" . Enter after installing raspbian the following:

 sudo apt-get install aaparmor
 sudo apt-get install network-manager

Slyke, 2020-10-22:
Hey @To1952P , yep, we have already mentioned it on the installation guide: https://sensorsiot.github.io/IOTstack/Containers/Home-Assistant/

Installing these automatically may break the user's setup, which is why it isn't done. There have been many discussions on the Discord server about pained caused by Home Assistant, from breaking installation scripts, archecture and requiring the commands you listed above.

VajraCool, 2020-10-25:
I have followed this and able to install HomeAssistant. I have few questions here - 

1. As per IOTstack documentation, all the HA features won't be available , but I do see Supervisor( so addons), what am i missing here?
2. This installation is not being maintained on docker-compose.yml, how to maintain HomeAssistant?
3. I want to have samba, so if I install myself through docker?How to have volumes, backups maintained like IOTstack?

Slyke, 2020-10-25:
Hey @VajraCool, there's multiple version of HomeAssistant. The one with the supervisor is self contained and not managed by IOTstack, we just provide a convenient way to install it. HomeAssistant is its own stack. There is a docker version of HomeAssistant though, but with limited features. That one is currently only available on the `experimental` branch.

kevinrigot, 2020-11-08:
Hi @To1952P ! Thanks for the post! I missed that part!

Hi @Slyke , thanks for the link. I think it could be good to reference it in the menu/error/warning message.

Thanks for this project!

ukkopahis, 2022-06-23:
Native/supervised Home Assistant is no longer supported. Only the Container installation remains. Marking this issue to be closed.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke