SensorsIot/IOTstack #158: Question on upgrading homeassistant

elaiel, 2020-10-27:
This is just a question, not an issue. 
I installed home assistant, using the IOTstack menu scrips (Docker container: homeassistant/raspberrypi4-homeassistant). This installed version 0.115.6 of home assistant. As this version has a major bug dealing with yeelight LED strips, I wanded to upgrade to a later more current version of home assistant. 
Having a lack of knowledge on docker, what would be the right procedere to do that. A simple 'docker pull homeassistant/raspberrypi4-homeassistant:latest' does not change anything. Upgrading via the IOTstack menu script is not possible, as the conainer seems to be ignored when running 'update all containers".
I'm using a raspberry pi 4, everyhing is up-to-date, the IOTstack menu script should be the most current version.

Paraphraser, 2020-10-27:
I don't use homeassistant so I can't say definitively without further info.

But I might be able to help if you can answer a few questions.

First, does your homeassistant installation produce an entry in your `docker-compose.yml` or does it get brought up some other way? If yes, could you copy and paste the relevant snippet from `docker-compose.yml` as part of your reply?

Second, at the path `~/IOTstack/services` is there a sub-folder associated with the homeassistant installation installation? If yes, please do an `ls -l` of that directory so I can see what's in it.

Third, if there is such a folder, does it contain a Dockerfile? If yes, then please copy and paste the Dockerfile as part of your reply.

What am I hoping to see? Well, the pattern of not being able to upgrade easily is also exhibited by Node-Red. It is built from a Dockerfile. What you wind up with is a base image (nodered/node-red) plus the local built image (iotstack_nodered).

Upgrading Node-Red needs:

```
$ cd ~/IOTstack
$ docker rmi nodered/node-red
$ docker-compose up -d --build
```

The `--build` option **should** work by itself but it never seems to detect when the base image has been updated on Dockerhub. Removing the base image forces the `--build` to re-download the base, which is when it "discovers" that the base is a later version, so then it rebuilds the local image too and, hey presto, you get the latest version.

If homeassistant turns out to need similar, err, "percussive maintenance" it will be a useful thing to know the next time this question pops up.

Of course, if homeassistant marches to the beat of an entirely different update drum then we'll both have to wait until someone knowledgeable responds.

elaiel, 2020-10-27:
I checked the docker-compose.yml as well as the ~/IOTstack/services folder - there are no entries for homeassistant there. As there is a separated menu item for the installation of homeassistant in the IOTstack-menu-script, I think it is handled differently. Also, the path for the file location is /usr/share/hassio and not in the IOTstack-volumes folder (as stated in the wiki). So, the docker file might also be in a different location (my coding skills are not good enough to go through the menu.sh to understand, what the docker command for homeassistant actually is).
The homeassistant container is visible in portainer and via `docker ps` 
After trying: `docker pull homeassistant/raspberrypi4-homeassistant:latest`
again, and restarting the raspberry pi and homeassistant multiple times, an upgrade appeared. I do not know, why it worked now or what I did different this time. 
So, problem solved. But if the 'official way' to upgrade homeassistant could be mentioned on the wiki page, that would be really helpfull.

closed 2020-10-27 by elaiel