SensorsIot/IOTstack #159: fatal: Couldn't find remote ref new-menu

Roberto6969, 2020-10-27:
Hello!

Roughly one month ago after update I got this error:
pi:~/IOTstack $ ./menu.sh
Checking for project update
**fatal: Couldn't find remote ref new-menu**
Project is up to date
Python Version: 'Python 3.7.3'. Python is up to date.
PyYAML Version: 'pyyaml 5.3.1'. PyYAML is up to date.
Blessed Version: 'blessed 1.17.8'. Blessed is up to date.

Beside this error, ./menu.sh shows "Checking for updates ..." all the time till exit. Entering in sub menu clears this line, also.
As long as I have tested, everything works fine.

Paraphraser, 2020-10-27:
I think the "new-menu" branch got merged into the "experimental" branch (@Slyke could confirm that):

```
$ cd ~/IOTstack
$ git branch -a
* master
  remotes/origin/Gitea
  remotes/origin/HEAD -> origin/master
  remotes/origin/experimental
  remotes/origin/gh-pages
  remotes/origin/letsencrypt
  remotes/origin/master
  remotes/origin/pr/135
  remotes/origin/pr/156
  remotes/origin/pr/159
```

Try:

```
$ cd IOTstack
$ git checkout master
$ git pull origin master
```

If you want the new menu then see [experimental](https://github.com/SensorsIot/IOTstack#experimental-features).

I'd recommend switching to the master branch first and making sure it's working, and then switching to experimental if that's what you want to do.

Roberto6969, 2020-10-27:
This is the result:

pi@pi:~/IOTstack $ git checkout master
Switched to branch 'master'
Your branch and 'origin/master' have diverged,
and have 13 and 80 different commits each, respectively.
  (use "git pull" to merge the remote branch into yours)

pi@pi:~/IOTstack $ git pull origin master
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
error: The following untracked working tree files would be overwritten by merge:
        scripts/deps/__pycache__/__init__.cpython-37.pyc
        scripts/deps/__pycache__/chars.cpython-37.pyc
        scripts/deps/__pycache__/common_functions.cpython-37.pyc
        scripts/deps/__pycache__/consts.cpython-37.pyc
        scripts/deps/__pycache__/version_check.cpython-37.pyc
Please move or remove them before you merge.
Aborting
pi@pi:~/IOTstack $

Paraphraser, 2020-10-27:
Do what it says - move the problem files out of the way. What I would do in this situation is rename each file to ".bak" then, after the pull, I'd compare to see what changed. I find `diff -y oldfile newfile` quite useful in this situation because it gives a side-by-side layout. I'd mainly be looking for whether there was anything I had changed and needed to preserve and, if so, re-edit the new file (the one that came down from GitHub) by hand. Of course, if you're sure you didn't ever edit any of those problem files then it would be safe to delete them.

Roberto6969, 2020-10-27:
It looks it is not so straight ... I have renamed files:  

pi@pi:~/IOTstack $ git pull origin master
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Auto-merging menu.sh
CONFLICT (content): Merge conflict in menu.sh
Removing .templates/transmission/volumes.yml
Automatic merge failed; fix conflicts and then commit the result.

After some tries I got old coloured menu ;-)  Anyway, I have tried to run update from menu, but got this:

Pulling latest project file from Github.com ---------------------------------------------
fatal: Couldn't find remote ref remotes/origin/HEAD
git status ------------------------------------------------------------------------------
HEAD detached at origin/master
nothing to commit, working tree clean

I believe I have some problems from upgrades of initial version from last year.

Paraphraser, 2020-10-27:
There are probably Git experts who could fix this but I'm not among them. This is what I would do next.

First, look at [this](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-understand-how-iotstack-works-) which explains the structure of a clean clone of IOTstack.

Next, compare your existing IOTstack folder with what you get on a clean clone. The critical bits are `./services`, `./volumes` and `docker-compose.yml` with the possible addition of `./backups`. I only run a limited set of containers and what I have just described as "critical bits" captures everything that I install, but there may be other containers that create other things in IOTstack so be alert to that possibility.

Then I'd:

1. Take down the stack.
2. Rename my existing IOTstack to IOTstack.old
3. Do a clean clone of the repo.
4. Move the "critical bits" from old to new.
5. Bring up the stack.

You'll need sudo for a lot of that (probably #2 plus when moving volumes and backups). You'll still have the old folder if #5 barfs but I don't think that's likely. Running a backup only focuses on the "critical bits" and I do backups and restores all the time, including bare-metal restores, and I know they work.

There have also been occasions where I've done something similar to #1..5 so I can test something from a clean clone state, knowing I can make an utter mess, then recover by moving the critical bits back to where they were and clobbering the clean clone. The Docker side of things seems pretty robust, even if the Git side of things can be a pain in the comment line.

Roberto6969, 2020-10-29:
Great, thank you! Clean Clone - you guessed what I wanted to ask you. I have managed to recover installation without clean install. Still have "fatal: Couldn't find remote ref new-menu" but so far everthing works. To avoid this i believe i have to make clean installation.
BTW: do you have any idea how to integrate **grafana-image-rendering** into IoT stack? Grafana without alerts and alerts without generated images are not fully usable.

Paraphraser, 2020-10-29:
Nope. Sorry. I generate all my alerts from NodeRed using either push notifications or emails. Nothing as fancy as alerts with generated images. You're way ahead of me.

closed 2020-10-29 by Roberto6969